aircrack-ng/rtl8812au #1052: DKMS error compiling rtl8812au-5.6.4.2 from kernel 6.0 series in Debian Testing

Tnix76, 2023-03-10:
I'm trying to compile the driver rtl8812au-5.6.4.2~20220606 for kernel 6.1.0-5-amd64 in Debian Testing into the DKMS system and I get the next errors:

```
DKMS make.log for rtl8812au-5.6.4.2~20220606 for kernel 6.1.0-5-amd64 (x86_64)
mié 08 mar 2023 20:28:16 CET
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.1.0-5-amd64/build M=/var/lib/dkms/rtl8812au/5.6.4.2~20220606/build  modules
make[1]: se entra en el directorio '/usr/src/linux-headers-6.1.0-5-amd64'
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_mi.o
/var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_mlme_ext.c: In function ‘rtw_append_assoc_req_owe_ie’:
/var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_mlme_ext.c:8974:13: warning: the comparison will always evaluate as ‘true’ for the address of ‘owe_ie’ will never be NULL [-Waddress]
 8974 |         if (sec->owe_ie && sec->owe_ie_len > 0) {
      |             ^~~
In file included from /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/include/drv_types.h:70,
                 from /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_mlme_ext.c:17:
/var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/include/rtw_security.h:167:12: note: ‘owe_ie’ declared here
  167 |         u8 owe_ie[MAX_OWE_IE_LEN];/* added in assoc req */
      |            ^~~~~~
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_xmit.o
/var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_ap.c: In function ‘rtw_ap_update_chbw_by_ifbmp’:
/var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_ap.c:4445:42: warning: the comparison will always evaluate as ‘true’ for the address of ‘padapters’ will never be NULL [-Waddress]
 4445 |                 if (!(ifbmp & BIT(i)) || !dvobj->padapters)
      |                                          ^
In file included from /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_ap.c:17:
/var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/include/drv_types.h:1046:19: note: ‘padapters’ declared here
 1046 |         _adapter *padapters[CONFIG_IFACE_NUMBER];/*IFACE_ID_MAX*/
      |                   ^~~~~~~~~
/var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_ap.c:4465:42: warning: the comparison will always evaluate as ‘true’ for the address of ‘padapters’ will never be NULL [-Waddress]
 4465 |                 if (!(ifbmp & BIT(i)) || !dvobj->padapters)
      |                                          ^
/var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/include/drv_types.h:1046:19: note: ‘padapters’ declared here
 1046 |         _adapter *padapters[CONFIG_IFACE_NUMBER];/*IFACE_ID_MAX*/
      |                   ^~~~~~~~~
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/os_dep/linux/usb_intf.o
/var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/os_dep/osdep_service.c: In function ‘rtw_random32’:
/var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/os_dep/osdep_service.c:2587:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘prandom_u32_max’? [-Werror=implicit-function-declaration]
 2587 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                prandom_u32_max
/var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/os_dep/linux/os_intfs.c: In function ‘rtw_os_ndev_register’:
/var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/os_dep/linux/os_intfs.c:1740:9: error: too many arguments to function ‘netif_napi_add’
 1740 |         netif_napi_add(ndev, &adapter->napi, rtw_recv_napi_poll, RTL_NAPI_WEIGHT);
      |         ^~~~~~~~~~~~~~
In file included from /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/include/osdep_service_linux.h:34,
                 from /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/include/osdep_service.h:58,
                 from /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/include/drv_types.h:26,
                 from /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/os_dep/linux/os_intfs.c:17:
/usr/src/linux-headers-6.1.0-5-common/include/linux/netdevice.h:2569:1: note: declared here
 2569 | netif_napi_add(struct net_device *dev, struct napi_struct *napi,
      | ^~~~~~~~~~~~~~
make[2]: *** [/usr/src/linux-headers-6.1.0-5-common/scripts/Makefile.build:255: /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/os_dep/linux/os_intfs.o] Error 1
make[2]: *** Se espera a que terminen otras tareas....
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.1.0-5-common/scripts/Makefile.build:255: /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build/os_dep/osdep_service.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.1.0-5-common/Makefile:2030: /var/lib/dkms/rtl8812au/5.6.4.2~20220606/build] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-6.1.0-5-amd64'
make: *** [Makefile:2251: modules] Error 2

```
The most recent release at the moment rtl8812au-5.6.4.2~20221105 gives the same or similar errors.

The rtl8812au-5.6.4.2~20220606 are now running fine in a 5.18 kernel

I've also tried the morrownr drivers: they compile fine and running into 6.1.0-5 kernel but not working very well (connects to wifi but not works until reconnects).

Thanks.

KES777, 2023-03-13:
```
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.2.5-060205-generic/build M=/home/kes/INST/driver/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-6.2.5-060205-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.2.0-16ubuntu1) 12.2.0
  You are using:           gcc-12 (Ubuntu 12.1.0-2ubuntu1~22.04) 12.1.0
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_cmd.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_security.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_debug.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_io.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_ioctl_query.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_ioctl_set.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_ieee80211.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_mlme.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_mlme_ext.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_mi.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_wlan_util.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_vht.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_pwrctrl.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_rf.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_chplan.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_recv.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_sta_mgt.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_ap.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/mesh/rtw_mesh.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_xmit.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_p2p.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_rson.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_tdls.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_br_ext.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_iol.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_sreset.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_btcoex_wifionly.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_btcoex.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_beamforming.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_odm.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_rm.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_rm_fsm.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/efuse/rtw_efuse.o
  CC [M]  /home/kes/INST/driver/rtl8812au/os_dep/osdep_service.o
  CC [M]  /home/kes/INST/driver/rtl8812au/os_dep/linux/os_intfs.o
  CC [M]  /home/kes/INST/driver/rtl8812au/os_dep/linux/usb_intf.o
  CC [M]  /home/kes/INST/driver/rtl8812au/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/kes/INST/driver/rtl8812au/os_dep/linux/ioctl_linux.o
  CC [M]  /home/kes/INST/driver/rtl8812au/os_dep/linux/xmit_linux.o
  CC [M]  /home/kes/INST/driver/rtl8812au/os_dep/linux/mlme_linux.o
  CC [M]  /home/kes/INST/driver/rtl8812au/os_dep/linux/recv_linux.o
  CC [M]  /home/kes/INST/driver/rtl8812au/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /home/kes/INST/driver/rtl8812au/os_dep/linux/wifi_regd.o
  CC [M]  /home/kes/INST/driver/rtl8812au/os_dep/linux/rtw_android.o
  CC [M]  /home/kes/INST/driver/rtl8812au/os_dep/linux/rtw_proc.o
  CC [M]  /home/kes/INST/driver/rtl8812au/os_dep/linux/rtw_rhashtable.o
  CC [M]  /home/kes/INST/driver/rtl8812au/os_dep/linux/ioctl_mp.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/hal_intf.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/hal_com.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/hal_com_phycfg.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/hal_phy.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/hal_dm.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/hal_dm_acs.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/hal_btcoex_wifionly.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/hal_btcoex.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/hal_mp.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/hal_mcc.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/hal_hci/hal_usb.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/led/hal_led.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/led/hal_usb_led.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/HalPwrSeqCmd.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/rtl8812a/Hal8812PwrSeq.o
[... 89 lines cut ...]
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/phydm/rtl8821a/phydm_regconfig8821a.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/phydm/rtl8821a/phydm_rtl8821a.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/phydm/halrf/rtl8821a/halrf_iqk_8821a_ce.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/phydm/rtl8814a/halhwimg8814a_bb.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/phydm/rtl8814a/halhwimg8814a_mac.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/phydm/rtl8814a/halhwimg8814a_rf.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/phydm/halrf/rtl8814a/halrf_iqk_8814a.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/phydm/rtl8814a/phydm_regconfig8814a.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/phydm/halrf/rtl8814a/halrf_8814a_ce.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/phydm/rtl8814a/phydm_rtl8814a.o
  CC [M]  /home/kes/INST/driver/rtl8812au/hal/phydm/txbf/haltxbf8814a.o
  CC [M]  /home/kes/INST/driver/rtl8812au/platform/platform_ops.o
  CC [M]  /home/kes/INST/driver/rtl8812au/core/rtw_mp.o
  LD [M]  /home/kes/INST/driver/rtl8812au/88XXau.o
[... cut ...]
  BTF [M] /home/kes/INST/driver/rtl8812au/88XXau.ko
Skipping BTF generation for /home/kes/INST/driver/rtl8812au/88XXau.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-6.2.5-060205-generic'
---------------------------------------------------------------------------
Visit https://github.com/aircrack-ng/rtl8812au for support/reporting issues
or check for newer versions (branches) of these drivers.                   
---------------------------------------------------------------------------
```

Tnix76, 2023-03-20:
Hi,

I've tried and now compiles fine. But when I boot or restart with the 6.x kernel the wireless device in KDE Plasma stays in "Configuring device" and finally fails after about 1 minute. 
In 5.8 kernel it works fine without problems.

Thanks for your answer.