aircrack-ng/rtl8812au #1058: Wireless extension support

b4cker, 2023-03-17:
Hi guys, when I compile for x64 linux (WSL 2) the driver has support for NETLINK and WIRELESS EXTENSIONS, but when I compile it for a raspberry pi 4 (using CONFIG_PLATFORM_ARM64_RPI) it only has NETLINK support. I can verify this from hcxlabtool @ZerBea.

ZerBea, 2023-03-18:
Looks like your kernel is compiled without WEXT support.
Take a look at your kernel config:

```
$ cat /proc/config.gz | gunzip | grep CFG80211
CONFIG_CFG80211=m
# CONFIG_CFG80211_DEVELOPER_WARNINGS is not set
CONFIG_CFG80211_REQUIRE_SIGNED_REGDB=y
CONFIG_CFG80211_USE_KERNEL_REGDB_KEYS=y
CONFIG_CFG80211_DEFAULT_PS=y
CONFIG_CFG80211_DEBUGFS=y
CONFIG_CFG80211_CRDA_SUPPORT=y
CONFIG_CFG80211_WEXT=y
CONFIG_CFG80211_WEXT_EXPORT=y
```

It sholud look like this:
```
CONFIG_CFG80211_WEXT=y
CONFIG_CFG80211_WEXT_EXPORT=y
```
if WIRELESS EXTENSIONS (WEXT) are supported by kernel.

b4cker, 2023-03-18:
thanks for answering, I have the same configuration except for "CONFIG_CFG80211_WEXT_EXPORT", I'm having difficulties getting the .config to be generated with that prop. (I create a new defconfig but then this prop does not appear in the .config). I'm using this kernel [/raspberrypi/linux](https://github.com/raspberrypi/linux) (default branch).

b4cker, 2023-03-18:
thanks for answering, yes, the kernel has all WEXT settings enabled, this is the kernel src [raspberrypi/linux](https://github.com/raspberrypi/linux) and the [bcm2711_defconfig](https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm64/configs/bcm2711_defconfig#L450). (all those props appear in the .config). I recompile and uninstall/install this driver and it still only supports NETLINK (at least that's says hcxlabtool -I wlan1), and hcxdumptool says "warning this driver does not support WIRELESS EXTENSIONS".

b4cker, 2023-03-18:
update: even though it compiles with "CONFIG_CFG80211_WEXT_EXPORT" when I looking at /proc/config.gz it is not there (when the other properties yes). For now everything indicates that it is a kernel problem, I will notify you of news.

ZerBea, 2023-03-19:
This is a standard default kernel config:
https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm64/configs/bcm2711_defconfig#L450
This does not automatically mean that it will be used to compile your kernel.

If this entries
CONFIG_CFG80211_WEXT=y
CONFIG_CFG80211_WEXT_EXPORT=y
are not present in your /proc/config.gz the kernel does not use WEXT.
If they are present in your /proc/config.gz the compiler may ignore them when compiling the driver.
In both cases, it is a config problem.

ZerBea, 2023-03-19:
The suggested way to detect the presence of WIRELESS EXTENSIONS is to use an ioctl() system call:
```
#define SIOCGIWNAME	0x8B01		/* get name == wireless protocol */
/* SIOCGIWNAME is used to verify the presence of Wireless Extensions.
 * Common values : "IEEE 802.11-DS", "IEEE 802.11-FH", "IEEE 802.11b"...
```
https://github.com/torvalds/linux/blob/master/include/uapi/linux/wireless.h#L238

Exactly that it is, what hcxdumptool/hcxlabtool is doing:
```
static void ioctl_get_wireless_extensions(size_t i)
{
struct iwreq pwrq;
static const char *protocol80211 = "IEEE 802.11";

memset(&pwrq, 0, sizeof(pwrq));
memcpy(pwrq.ifr_name, (ifpresentlist + i)->name, IF_NAMESIZE);
if(ioctl(fd_socket_unix, SIOCGIWNAME, &pwrq) < 0) return;
if(strncmp(protocol80211, pwrq.u.name, IF_NAMESIZE) == 0) (ifpresentlist + i)->type |= IF_HAS_WEXT;
return;
}
If the kernel report something like this "IEEE 802.11", WIRELESS EXTENSIONS are enabled,

ZerBea, 2023-03-19:
BTW:
If someone is interested how to detect the presence of NL80211, this is the suggested way:
Instead of running an ioctl() system call, we use CTRL_CMD_GETFAMILY (via GENERIC NETLINK) to determine the presence of NL80211:
https://github.com/ZerBea/wifi_laboratory/blob/main/hcxlabtool.c#L3214

ZerBea, 2023-03-19:
Please notice that this driver is a very, "special" driver.
It doesn't support all WIRELESS EXTENSIONS. That means, controlling the interface via ioctl() may or may not work.
To get rid of this problem, I recommend to use hcxlabtool.

b4cker, 2023-03-19:
> Please notice that this driver is a very, "special" driver. It doesn't support all WIRELESS EXTENSIONS. That means, controlling the interface via ioctl() may or may not work. To get rid of this problem, I recommend to use hcxlabtool.

yes, I know, but confused me that this driver work perfectly in WSL, but not in raspberry pi (although now everything seems to indicate that it's a kernel problem)

ZerBea, 2023-03-19:
I have several Raspberry PI systems (all armv6). Unfortunately I have to say, that they are "very special", too.
```
$ uname -r
5.15.32+
```

```
$ cat /etc/debian_version
bookworm/sid
```

```
$ modprobe configs
$ cat /proc/config.gz | gunzip | grep CFG
CONFIG_CFG80211=m
# CONFIG_CFG80211_DEVELOPER_WARNINGS is not set
# CONFIG_CFG80211_CERTIFICATION_ONUS is not set
CONFIG_CFG80211_REQUIRE_SIGNED_REGDB=y
CONFIG_CFG80211_USE_KERNEL_REGDB_KEYS=y
CONFIG_CFG80211_DEFAULT_PS=y
# CONFIG_CFG80211_DEBUGFS is not set
CONFIG_CFG80211_CRDA_SUPPORT=y
CONFIG_CFG80211_WEXT=y
```

CRDA support is enabled:
`CONFIG_CFG80211_CRDA_SUPPORT=y
`
but it doesn't load the regulatory domain due to a race condition that caused that the signature of the data base acn not be validated on system start:
```
[   57.927112] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   58.313738] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   58.330331] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
```

Unfortunately Arch developer decided to remove armv6 support and I have to use Debian bookworm (modified Raspbian lite) instead.

b4cker, 2023-03-19:
I did the wext check manually and iwreq_data->name is "IEEE 802.11AC" (raspberry pi) and the verification checks that it is exactly "IEEE 802.11", [hcxdumptool/pull/258](https://github.com/ZerBea/hcxdumptool/pull/258) now hcxdumptool/hcxlabtool works correctly.

ZerBea, 2023-03-20:
Great investigation. Thanks.
I took a closer look at the driver code and noticed that there are several names:
IEEE 802.11bn, EEE 802.11b, IEEE 802.11AC, IEEE 802.11an, IEEE 802.11a, IEEE 802.11gn, IEEE 802.11g
```
	/* Add the protocol name */
	iwe->cmd = SIOCGIWNAME;
	if ((rtw_is_cckratesonly_included((u8 *)&pnetwork->network.SupportedRates)) == _TRUE) {
		if (ht_cap == _TRUE)
			snprintf(iwe->u.name, IFNAMSIZ, "IEEE 802.11bn");
		else
			snprintf(iwe->u.name, IFNAMSIZ, "IEEE 802.11b");
	} else if ((rtw_is_cckrates_included((u8 *)&pnetwork->network.SupportedRates)) == _TRUE) {
		if (ht_cap == _TRUE)
			snprintf(iwe->u.name, IFNAMSIZ, "IEEE 802.11bgn");
		else
			snprintf(iwe->u.name, IFNAMSIZ, "IEEE 802.11bg");
	} else {
		if (pnetwork->network.Configuration.DSConfig > 14) {
			#ifdef CONFIG_80211AC_VHT
			if (vht_cap == _TRUE)
				snprintf(iwe->u.name, IFNAMSIZ, "IEEE 802.11AC");
			else
			#endif
			{
				if (ht_cap == _TRUE)
					snprintf(iwe->u.name, IFNAMSIZ, "IEEE 802.11an");
				else
					snprintf(iwe->u.name, IFNAMSIZ, "IEEE 802.11a");
			}
		} else {
			if (ht_cap == _TRUE)
				snprintf(iwe->u.name, IFNAMSIZ, "IEEE 802.11gn");
			else
				snprintf(iwe->u.name, IFNAMSIZ, "IEEE 802.11g");
		}
	}
```
https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.2/os_dep/linux/ioctl_linux.c#L435

No hcxdumptool and hcxlabtool take the first 11 characters (IEEE 802.11) only, to detect the presence of WIRELESS EXTENSIONS.

ZerBea, 2023-04-07:
Starting with this commit:
https://github.com/ZerBea/hcxdumptool/commit/311e09325d70a0774d25f687b285c3af26ca7519
hcxdumptool is refactored and the entire code is replaced by hcxlabtool code.

WIRELESS EXTENSIONS are completely retired and replaced by NL80211 stack (which is used by state of the art drivers).
Procedure is still the same:
stop services that take access to the interface
run "hcxdumptool -L" ot get an overview about the interfaces
run "hcxdumptool -I INTERFACE" to get information about a specified interface
to collect hashes (PMKIDs or EAPOL MESSAGE PAIRs) run
"sudo hcxdumptool --beacontx=5 --attemptapmax=100 -attemptclientmax=100"
or
"sudo hcxdumptool i INTERFACE --beacontx=5 --attemptapmax=100 -attemptclientmax=100"