aircrack-ng/rtl8812au #1071: Kali 6.1.0 - Awus036ACH - rtl8812au - Deauth not working

mkarne1, 2023-04-25:
Hello, 

I can't deauthenticate any client from any Wifi. I installed the [latest driver](https://github.com/aircrack-ng/rtl8812au) with the following commands: 

```
sudo apt-get install bc mokutil build-essential libelf-dev linux-headers-`uname -r`
reboot
git clone -b v5.6.4.2 https://github.com/aircrack-ng/rtl8812au.git
cd rtl
make dkms_install  
```

Afterwards, I ran:

```
airmon-ng check kill
ip link set wlan0 down
iw dev wlan0 set type monitor
ip link set wlan0 up
airodump-ng --bssid D4:3D:F3:74:F3:91 --channel 5 --write /tmp/test wlan0
```

And in a new terminal:

```
aireplay-ng --deauth 1000 -a D4:3D:XX:XX:XX:91 -c 14:7D:XX:XX:XX:44 wlan0          
19:46:04  Waiting for beacon frame (BSSID: D4:3D:F3:74:F3:91) on channel 5
19:46:05  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [31|73 ACKs]
19:46:05  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [74|89 ACKs]
19:46:06  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [136|118 ACKs]
19:46:06  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [ 2|63 ACKs]
19:46:07  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [ 0|63 ACKs]
19:46:07  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [ 1|64 ACKs]
19:46:08  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [14|65 ACKs]
19:46:08  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [89|89 ACKs]
19:46:09  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [74|88 ACKs]
19:46:09  Sending 64 directed DeAuth (code 7). STMAC: [14:7D:XX:XX:XX:44] [128|119 ACKs]
```

It just keeps sending deauth packets but the client isn't deauthenticating. 

Any help would be really appreciated!

Thanks!

kimocoder, 2023-04-25:
Well, this "problem" is seen on any chipsets and I think it may be due to unsupported protocolls, AX routers, WPA3-SAE. I got the same "issue" on more or less all chipsets. 6E support is on the way to aircrack-ng, so I dont think this is a driver issue, rather tools issue/not-yet-supported protocolls.

Mister-X-, 2023-04-25:
One thing first: there is no need to deauth that much. Each instance already sends 64/128 deauth. You need to let the client reconnect.

However, as kimocoder pointed at, if the network has 802.11w (PMF), which is always the case of WPA3/OWE, this won't work. PMF can be used on WPA/WPA2 networks as well, but it's less common, and you'd have to see if 1. It is optional in the beacon and 2. The client negotiated it.

I'd recommend reviewing the capture file.

mkarne1, 2023-04-25:
Hi @Mister-X-, thanks for the quick reply!

It's a WPA/2 network. I checked the router settings but couldn't find anything about PMF. 

Regarding the capture file: Would you mind taking a quick look at the following screenshots? I can't find anything useful :(

[image]
[image]
[image]

I also uploaded the whole cap to Google Drive: 
https://drive.google.com/drive/folders/1TJx5ivobjiNvzJd1jeWUApmhGKLW6iE4?usp=sharing

Again, thanks a lot for you help, I really appreciate it!

mkarne1, 2023-04-25:
@kimocoder thank you too for the quick reply! 

it's a normal WPA/2 network. Does it still apply? I posted the .cap file in my previous response to @Mister-X-, maybe it helps. 

Thanks for your time & best regards!

alza36sc105, 2023-05-11:
I googled and got it that Channel Switch Announcement Attack is available on WPA/3 in addition to WPA/2.
I'm not so sure if its true...
I will write test code if i feel like it

AbhiJohnv, 2023-05-17:
aireplay-ng --deauth 1000 -a D4:3D:XX:XX:XX:91 -c 14:7D:XX:XX:XX:44 wlan0          
remove the D4:3D:XX:XX:XX:91 -c 14:7D:XX:XX:XX:44 and give the target bssid 
run it

ZerBea, 2023-06-08:
> Well, this "problem" is seen on any chipsets and I think it may be due to unsupported protocolls, AX routers, WPA3-SAE. I got the same "issue" on more or less all chipsets. 6E support is on the way [...]

@kimocoder 
Hi Christian.
WPA3-SAE or MFP shouldn't be a big problem.  REASSOCIATION attack should work in both cases (downgrade CLIENT to WPA2 in case of WPA3-SAE and force a RECONNECT of the CLIENT in case of MFP is activated).
Neiter an AUTHENTICATION frame, nor an ACCOCIATIONREQUEST/RESPONSE frame nor a REASSOCIATIONREQUEST/REASSOCIATIONRESPONSE frame is protected.