aircrack-ng/rtl8812au #1073: No longer detecting networks after updating the driver

i486, 2023-05-28:
I recently installed the latest driver commit (b3227c731ec957b706f8eb2ebcf42ba1c2f0d610) for my USB-AC56 (RTL8812AU) WiFi card on my Kali virtual machine. However, after installing this driver, my WiFi card stopped detecting any networks in both monitor and managed modes. Additionally, the "Available Networks" list in the Network Manager appeared greyed out and inaccessible.

To resolve this issue, I randomly chose an earlier commit (ccda185731471e385d4b2c3a7910cb967447483d) to rollback to, and fortunately, my WiFi card started working again.

I then reinstalled the latest driver, and the same problem occurred once more.

Throughout the installation and testing process, I followed these steps: "sudo make dkms_remove," checked the DKMS tree, restarted the system, and performed "make dkms_install." No errors were encountered during these steps.

Based on my observations, it seems like something may be incorrect in the commits between April and May.

I plan to roll back through the commits, one by one, starting from the latest version. I will update this thread later with my findings.

Various System/Wi-Fi information (when latest driver is installed and scan doesn't find anything)
[wireless-info.txt](https://github.com/aircrack-ng/rtl8812au/files/11584079/wireless-info.txt)

i486, 2023-05-28:
I have just installed the previous version ([commit 35308f4dd73e77fa572c48867cce737449dd8548](https://github.com/aircrack-ng/rtl8812au/commit/35308f4dd73e77fa572c48867cce737449dd8548)) and now my WiFi card is working again.

[**Some cleanup of unused chips**](https://github.com/aircrack-ng/rtl8812au/commit/b3227c731ec957b706f8eb2ebcf42ba1c2f0d610). 

Well, well, well, it seems the cleanup crew had a little too much enthusiasm and accidentally swept away the vital chips that powered our WiFi. Fear not, intrepid troubleshooter! It's time to bring back those wayward chips and restore WiFi harmony to our digital realm :P

kimocoder, 2023-05-28:
Yeah, I got aware of it. I will revert when I get home 👍

kimocoder, 2023-05-31:
Reverting this evening.

vishytk, 2023-06-01:
I am using Fedora 30, Kernel-6.3.4 and the issue still persists. The solution suggested below works but the main branch still has problem.

My device is TP-Link 802.11ac

> I have just installed the previous version ([commit 35308f4dd73e77fa572c48867cce737449dd8548](https://github.com/aircrack-ng/rtl8812au/commit/35308f4dd73e77fa572c48867cce737449dd8548)) and now my [...]

NayamAmarshe, 2023-06-03:
Same here. Reverting to https://github.com/aircrack-ng/rtl8812au/commit/ccda185731471e385d4b2c3a7910cb967447483d fixed it for TP-Link Archer T2U Plus.

MightiestGoat, 2023-06-04:
@Entropy512  well the root cause of this that types are mapped according to the array before the cleanup. After the cleanup the type for rtl8812 driver is 4 while update require it to be 3. Either way I reverted that change(check my pr) or I can add padding to fix this.

Yoyolick, 2023-06-06:
Bumping this, reverting to ccda185 fixed for the TP-Link Archer T2U Plus.

etlam262, 2023-06-06:
> Reverting this evening.

Hey @kimocoder just a small reminder, that this hasn't been fixed yet.
5ae69b685b54af6797dd7473917c22de81ded338 is working for me.

git4rker, 2024-08-20:
waiting patiently lol