aircrack-ng/rtl8812au #1075: Kernel crash - 6.1.28 Gentoo

Anoncheg1, 2023-05-31:
I was trying master version right now.

uname -a : Linux  6.1.28-gentoo #7 SMP x86_64 GenuineIntel GNU/Linux

rm -r /usr/src/linux/drivers/net/wireless/realtek/rtlwifi/rtl8812au
cp -r /usr/local/src/rtl8812au /usr/src/linux/drivers/net/wireless/realtek/rtlwifi/
cat /usr/src/linux/drivers/net/wireless/realtek/rtlwifi/rtl8812au/Makefile | sed 's/export CONFIG_88XXAU = m/export CONFIG_88XXAU = y/' > /tmp/Makefile
mv /tmp/Makefile /usr/src/linux/drivers/net/wireless/realtek/rtlwifi/rtl8812au/Makefile
```

echo 'obj-$(CONFIG_88XXAU)                 += rtl8812au/' >> /usr/src/linux/drivers/net/wireless/realtek/rtlwifi/Makefile
if [ -e /usr/src/linux/drivers/net/wireless/realtek/rtlwifi/Kconfig.back ] ; then
    mv /usr/src/linux/drivers/net/wireless/realtek/rtlwifi/Kconfig.back /usr/src/linux/drivers/net/wireless/realtek/rtlwifi/Kconfig
fi
cp /usr/src/linux/drivers/net/wireless/realtek/rtlwifi/Kconfig /usr/src/linux/drivers/net/wireless/realtek/rtlwifi/Kconfig.back
head -n -1 /usr/src/linux/drivers/net/wireless/realtek/rtlwifi/Kconfig > /tmp/Kconfig
# cat /usr/src/linux/drivers/net/wireless/realtek/rtlwifi/rtl8812au/Kconfig | sed -e 's/88XXAU/RTL8812AU/' -e 's/88XXau/rtl8812au/' >> /tmp/Kconfig
cat /usr/src/linux/drivers/net/wireless/realtek/rtlwifi/rtl8812au/Kconfig >> /tmp/Kconfig
echo endif >> /tmp/Kconfig

mv /tmp/Kconfig /usr/src/linux/drivers/net/wireless/realtek/rtlwifi/Kconfig
```

that is what I add to /usr/src/linux/drivers/net/wireless/realtek/rtlwifi/Kconfig

```
config 88XXAU
	tristate "Realtek 88XXau USB WiFi"
	depends on USB
	select RTLWIFI
	select RTLWIFI_USB
	help
	  Help message of 88XXau
```

make 

make install

Everything was perfect, but when I plug RTL8821AU kernel crash with this messages:
[image]

Anoncheg1, 2023-06-02:
This error with:

```
commit dc8dcb4fa2c4a6adde0d9c19cc417050d80d2a13
Merge: b3227c7 5ae69b6
Author: Christian Bremvåg <christian@aircrack-ng.org>
Date:   Wed May 31 16:01:58 2023 +0200

```
Now I this and it work.
```
commit 285989435a43bad0f6fa7e8d7b2886dca0ee5e46 (HEAD)
Author: Marcelina Hołub <mholub@tutanota.com>
Date:   Wed Mar 15 00:37:43 2023 +0100

```

asm64, 2023-06-02:
I confirm. Ubuntu 22.04.2. Kernel 5.19.0-43-generic.
Bus 002 Device 002: ID 2357:0120 TP-Link Archer T2U PLUS [RTL8821AU]

[8812au_crash.log](https://github.com/aircrack-ng/rtl8812au/files/11633119/8812au_crash.log)