aircrack-ng/rtl8812au #1081: Wlan/Wifi drivers (8812au) does not work with Fedora 38 and Kernel 6.3.5-200.fc38.x86_64

terohiekkalinna, 2023-06-05:
TP-LINK AC600 / 8812au driver does not work in Fedora 38 anymore. During dkms install in get following "grep: warning: stray \ before #" messages:

grep: warning: stray \ before #
grep: warning: stray \ before #
grep: warning: stray \ before #
grep: warning: stray \ before #
grep: warning: stray \ before #
mkdir: created directory '/usr/src/8812au-5.6.4.2_35491.20191025'
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025
Creating symlink /var/lib/dkms/8812au/5.6.4.2_35491.20191025/source -> /usr/src/8812au-5.6.4.2_35491.20191025
dkms build -m 8812au -v 5.6.4.2_35491.20191025
Sign command: /lib/modules/6.3.5-200.fc38.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' -j16 KVER=6.3.5-200.fc38.x86_64 KSRC=/lib/modules/6.3.5-200.fc38.x86_64/build........
Signing module /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/88XXau.ko
Cleaning build area...
dkms install -m 8812au -v 5.6.4.2_35491.20191025

88XXau.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.3.5-200.fc38.x86_64/extra/
depmod......
dkms status -m 8812au
8812au/5.6.4.2_35491.20191025, 6.3.5-200.fc38.x86_64, x86_64: installed

And Wifi cannot find any network and dmes and system log says about 88XXau driver:

dmesg:

[  355.799016] ------------[ cut here ]------------
[  355.799019] WARNING: CPU: 1 PID: 7692 at /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_odm.c:62 rtw_odm_init_ic_type+0x44/0x70 [88XXau]
[  355.799064] Modules linked in: 88XXau(OE+) cfg80211 snd_seq_dummy snd_hrtimer nvidia_drm(POE) nvidia_modeset(POE) nvidia_uvm(POE) nf_conntrack_netbios_ns nf_conntrack_broadcast nft_fib_inet nft_fib_ipv4 nft_fib_ipv6 nft_fib nvidia(POE) nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip_set nf_tables nfnetlink qrtr snd_hda_codec_hdmi sunrpc binfmt_misc vfat fat intel_rapl_msr intel_rapl_common snd_hda_intel intel_uncore_frequency intel_uncore_frequency_common snd_intel_dspcfg sb_edac snd_intel_sdw_acpi x86_pkg_temp_thermal intel_powerclamp snd_hda_codec coretemp snd_usb_audio rapl snd_hda_core uvcvideo snd_usbmidi_lib snd_rawmidi snd_hwdep uvc snd_seq videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 snd_seq_device videobuf2_common snd_pcm videodev mei_wdt iTCO_wdt hp_wmi snd_timer intel_pmc_bxt sparse_keymap platform_profile iTCO_vendor_support mei_me intel_cstate rfkill wmi_bmof snd i2c_i801 mc pcspkr mei video i2c_smbus intel_uncore
[  355.799102]  soundcore lpc_ich tpm_infineon loop zram crct10dif_pclmul crc32_pclmul crc32c_intel polyval_clmulni polyval_generic e1000e ghash_clmulni_intel sha512_ssse3 serio_raw wmi ip6_tables ip_tables fuse
[  355.799111] CPU: 1 PID: 7692 Comm: modprobe Tainted: P           OE      6.3.5-200.fc38.x86_64 #1
[  355.799113] Hardware name: Hewlett-Packard HP Z640 Workstation/212A, BIOS M60 v02.59 03/31/2022
[  355.799114] RIP: 0010:rtw_odm_init_ic_type+0x44/0x70 [88XXau]
[  355.799186] Code: 3e 00 00 80 fa 0e 76 17 8b 15 c4 17 0b 00 85 d2 74 18 be 04 00 00 00 48 89 c7 e9 07 af 07 00 8b 14 95 10 46 f0 c4 85 d2 75 e8 <0f> 0b be 04 00 00 00 48 89 c7 e9 ed ae 07 00 66 66 2e 0f 1f 84 00
[  355.799188] RSP: 0018:ffffb6870ca6fae0 EFLAGS: 00010246
[  355.799190] RAX: ffffb687036f4ed0 RBX: ffffb68700277000 RCX: 0000000000000002
[  355.799191] RDX: 0000000000000000 RSI: ffffffffc4f15fe4 RDI: ffffb68700277000
[  355.799193] RBP: ffff9c7661230000 R08: ffffffffc4f15fe3 R09: 0000000000ffff0a
[  355.799194] R10: 0000000000000000 R11: 0000000000000100 R12: ffff9c758fcde400
[  355.799195] R13: ffffffffc4f24300 R14: ffff9c758fcde400 R15: ffff9c758d021000
[  355.799196] FS:  00007f915689c740(0000) GS:ffff9c846f640000(0000) knlGS:0000000000000000
[  355.799197] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  355.799199] CR2: 0000559834d97fe0 CR3: 000000029b560006 CR4: 00000000003706e0
[  355.799200] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[  355.799201] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[  355.799202] Call Trace:
[  355.799204]  <TASK>
[  355.799205]  ? rtw_odm_init_ic_type+0x44/0x70 [88XXau]
[  355.799254]  ? __warn+0x81/0x130
[  355.799259]  ? rtw_odm_init_ic_type+0x44/0x70 [88XXau]
[  355.799310]  ? report_bug+0x171/0x1a0
[  355.799314]  ? handle_bug+0x3c/0x80
[  355.799317]  ? exc_invalid_op+0x17/0x70
[  355.799319]  ? asm_exc_invalid_op+0x1a/0x20
[  355.799323]  ? rtw_odm_init_ic_type+0x44/0x70 [88XXau]
[  355.799373]  ? rtw_usb_primary_adapter_init+0x10c/0x880 [88XXau]
[  355.799425]  ? rtw_usb_primary_adapter_init+0x3d6/0x880 [88XXau]
[  355.799491]  ? usb_probe_interface+0xe6/0x2f0
[  355.799495]  ? really_probe+0x19e/0x3e0
[... 364 lines cut ...]
kesä 05 08:06:51 fedora audit: BPF prog-id=100 op=UNLOAD
kesä 05 08:06:51 fedora audit: BPF prog-id=99 op=UNLOAD
kesä 05 08:06:51 fedora audit: BPF prog-id=98 op=UNLOAD
kesä 05 08:06:57 fedora wpa_supplicant[1845]: wlp0s20u13u1: CTRL-EVENT-SCAN-FAILED ret=-16
kesä 05 08:06:57 fedora geoclue[3360]: WiFi scan failed
kesä 05 08:07:12 fedora wpa_supplicant[1845]: wlp0s20u13u1: CTRL-EVENT-SCAN-FAILED ret=-16
kesä 05 08:07:12 fedora geoclue[3360]: WiFi scan failed
kesä 05 08:07:26 fedora wpa_supplicant[1845]: wlp0s20u13u1: CTRL-EVENT-SCAN-FAILED ret=-16
kesä 05 08:07:26 fedora geoclue[3360]: WiFi scan failed
kesä 05 08:07:27 fedora wpa_supplicant[1845]: wlp0s20u13u1: CTRL-EVENT-SCAN-FAILED ret=-16
kesä 05 08:07:27 fedora geoclue[3360]: WiFi scan failed
kesä 05 08:07:42 fedora wpa_supplicant[1845]: wlp0s20u13u1: CTRL-EVENT-SCAN-FAILED ret=-16
kesä 05 08:07:42 fedora geoclue[3360]: WiFi scan failed
kesä 05 08:07:57 fedora wpa_supplicant[1845]: wlp0s20u13u1: CTRL-EVENT-SCAN-FAILED ret=-16
kesä 05 08:07:57 fedora geoclue[3360]: WiFi scan failed
kesä 05 08:08:12 fedora wpa_supplicant[1845]: wlp0s20u13u1: CTRL-EVENT-SCAN-FAILED ret=-16
kesä 05 08:08:12 fedora geoclue[3360]: WiFi scan failed
kesä 05 08:08:27 fedora wpa_supplicant[1845]: wlp0s20u13u1: CTRL-EVENT-SCAN-FAILED ret=-16
kesä 05 08:08:27 fedora geoclue[3360]: WiFi scan failed
kesä 05 08:08:30 fedora wpa_supplicant[1845]: wlp0s20u13u1: CTRL-EVENT-SCAN-FAILED ret=-16
kesä 05 08:08:30 fedora geoclue[3360]: WiFi scan failed
kesä 05 08:08:42 fedora wpa_supplicant[1845]: wlp0s20u13u1: CTRL-EVENT-SCAN-FAILED ret=-16
kesä 05 08:08:42 fedora geoclue[3360]: WiFi scan failed

I hope this info helps to debug drivers!

juceda, 2023-06-09:
I confirm that this problem still exists, and the mentioned patch does not solve it, since I have expressly checked the hal/hal_intf.c file and I have the same content that the patch intends to do in the _chip_type_to_odm_ic_type array, but I still have kernel crashes like the one exposed above.

terohiekkalinna, 2023-06-19:
I'm just wondering is this rtl8812au driver project alive anymore?

CGarces, 2023-06-19:
> I'm just wondering is this rtl8812au driver project alive anymore?

Last commit is from 2 weeks ago.

arbonvata, 2023-06-24:
Not working for me either. Tested on two different computers with both fedora 38 and the kernel mentioned in the title, and also on ubuntu 23.04 and kernel 6.2.0-20-generic