aircrack-ng/rtl8812au #1083: Oops in cfg80211 with a 8814au adapter

sgaure, 2023-06-13:
I'm using a `Bus 002 Device 002: ID 2357:0106 TP-Link Archer T9UH v1 [Realtek RTL8814AU]` on Ubuntu 23.04, and get a kernel Oops when trying to connect to my wifi-network. I'm using v5.6.4.2.

[oops.txt](https://github.com/aircrack-ng/rtl8812au/files/11731827/oops.txt)

sgaure, 2023-06-15:
Here's some more info. There's a long time since I debugged kernel stuff, but some printk's here and there reveals the following. The oops, a zero pointer reference, occurs in `__cfg80211_connect_result` in net/wireless/sme.c at the line 
`ether_addr_copy(wdev->u.client.connected_addr, connected_addr)`.

It turns out that `connected_addr` is a zero pointer which has been set to `cr->ap_mld_addr` above, because `cr->valid_links` is 0.  And `cr->ap_mld_addr` is zero. 

There is a test at the beginning of the function for `ap_mld_addr` being zero, but it's only performed if `cr->valid_links` is non-zero.

The request presumably originated in the 88XXau module, but I do not know where, neither do I know who should have set `cr->ap_mld_addr`.

aneutron, 2023-08-13:
I have almost exactly the same thing, with the Archer T2U Nan. Kernel is 6.10 (Debian 12). I don't really have as much experience, but here's what I could gather:

The oops log file: [rt-wifi-oops.txt](https://github.com/aircrack-ng/rtl8812au/files/12329665/rt-wifi-oops.txt)

In installed gdb, the debug symbols and followed [this tutorial](https://www.kernel.org/doc/html/v5.15/admin-guide/bug-hunting.html), but this is all I could get as of now:

[image]

I can't however reliably reproduce this error, as it occurs after a long time of inactivity.