aircrack-ng/rtl8812au #1087: I have removed the driver to install the latest commit but I can't get it installed

konsbn, 2023-06-26:
I installed the mightiestgoat commit and it worked until my kernel updated. After update it stopped connecting to wireless networks so I uninstalled it and cloned the `aircrack/rtl8812au`  repository. However on running `sudo make dkms_install`  I get this error
```grep: warning: stray \ before #
grep: warning: stray \ before #
grep: warning: stray \ before #
grep: warning: stray \ before #
grep: warning: stray \ before #
mkdir: created directory '/usr/src/8812au-5.6.4.2_35491.20191025'
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025
Error! DKMS tree already contains: 8812au-5.6.4.2_35491.20191025
You cannot add the same module/version combo more than once.
make: *** [Makefile:1783: dkms_install] Error 3
```
I have tried everything to remove the driver however I am still unable to get it working. Please help. Thanks

kimocoder, 2023-06-26:
I can try reproduce shortly

konsbn, 2023-06-27:
I finally solved it using the command `locate 88XXau`  to find where all the files and stored and then removed them manually. Finally I did a clean reinstall.