aircrack-ng/rtl8812au #1093: Make error on Rocky Linux 9.2

Fidelis-Marcos, 2023-07-04:
Couldn't make, complains about redefinition of eth_he_addr_set.  I tried the solution of issue [1005](https://github.com/aircrack-ng/rtl8812au/issues/1005) but it didn't work. If anyone could help I'd greatly appreciate it, I feel like is simpler than I think, tried changing some other things in a similar way to 1005 but couldn't figure it out. Thank you for reading this at all.

Here is the contents of the logfile. 

```
DKMS make.log for 8812au-5.6.4.2_35491.20191025 for kernel 5.14.0-284.18.1.el9_2.x86_64 (x86_64)
Mon  3 Jul 21:54:53 EDT 2023
make[1]: Entering directory '/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build'
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.14.0-284.18.1.el9_2.x86_64/build M=/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build  modules
make[2]: Entering directory '/usr/src/kernels/5.14.0-284.18.1.el9_2.x86_64'
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_mlme.o
In file included from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types.h:37,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_debug.c:17:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types_linux.h:20:20: error: redefinition of ‘eth_hw_addr_set’
   20 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types.h:37,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_cmd.c:17:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types_linux.h:20:20: error: redefinition of ‘eth_hw_addr_set’
   20 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types.h:37,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/../hal/phydm/phydm_types.h:215,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/../hal/phydm/phydm_precomp.h:29,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/hal_data.h:20,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_mlme.c:17:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types_linux.h:20:20: error: redefinition of ‘eth_hw_addr_set’
   20 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types.h:37,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_security.c:17:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types_linux.h:20:20: error: redefinition of ‘eth_hw_addr_set’
   20 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types.h:37,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_ieee80211.c:17:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types_linux.h:20:20: error: redefinition of ‘eth_hw_addr_set’
   20 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types.h:37,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_ioctl_set.c:17:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types_linux.h:20:20: error: redefinition of ‘eth_hw_addr_set’
   20 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types.h:37,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_ioctl_query.c:17:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types_linux.h:20:20: error: redefinition of ‘eth_hw_addr_set’
   20 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/osdep_service_linux.h:49,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/osdep_service.h:58,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types.h:26,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_cmd.c:17:
./include/linux/etherdevice.h:314:20: note: previous definition of ‘eth_hw_addr_set’ with type ‘void(struct net_device *, const u8 *)’ {aka ‘void(struct net_device *, const unsigned char *)’}
  314 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/osdep_service_linux.h:49,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/osdep_service.h:58,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types.h:26,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_debug.c:17:
./include/linux/etherdevice.h:314:20: note: previous definition of ‘eth_hw_addr_set’ with type ‘void(struct net_device *, const u8 *)’ {aka ‘void(struct net_device *, const unsigned char *)’}
  314 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
[... 41 lines cut ...]
   20 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/osdep_service_linux.h:49,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/osdep_service.h:58,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types.h:26,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_io.c:47:
./include/linux/etherdevice.h:314:20: note: previous definition of ‘eth_hw_addr_set’ with type ‘void(struct net_device *, const u8 *)’ {aka ‘void(struct net_device *, const unsigned char *)’}
  314 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
make[3]: *** [scripts/Makefile.build:321: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_ioctl_query.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: *** [scripts/Makefile.build:321: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_security.o] Error 1
make[3]: *** [scripts/Makefile.build:321: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_ieee80211.o] Error 1
make[3]: *** [scripts/Makefile.build:321: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_io.o] Error 1
make[3]: *** [scripts/Makefile.build:321: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_ioctl_set.o] Error 1
make[3]: *** [scripts/Makefile.build:321: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_cmd.o] Error 1
make[3]: *** [scripts/Makefile.build:321: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_mlme.o] Error 1
make[3]: *** [scripts/Makefile.build:321: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_debug.o] Error 1
make[2]: *** [Makefile:1923: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build] Error 2
make[2]: Leaving directory '/usr/src/kernels/5.14.0-284.18.1.el9_2.x86_64'
make[1]: *** [Makefile:1724: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build'
```

My kernel is 5.14.0-284.18.1.el9_2.x86_64, other than activating things like EPEL and failing to install vmware I have made no significant changes to my machine

SolAZDev, 2023-08-19:
same here on debian 11 with kernel 5.10.10.0-25

virer, 2023-11-29:
I've manage to fix it for Rocky9.3 in my fork available here :
https://github.com/virer/rtl8812au/tree/v5.6.4.2_rocky93
hope this help