aircrack-ng/rtl8812au #1098: DKMS error compiling rtl8812au-5.6.4.2 from kernel 5.14.0-340.el9.x86_64

alexzh16, 2023-07-26:
Hi,

I'm trying to compile the driver rtl8812au-5.6.4.2 for kernel 5.14.0-340.el9.x86_64 Centos 9 into the DKMS system and I get the next errors:
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.14.0-340.el9.x86_64/build M=/opt/rtl8812au  modules
make[1]: Entering directory '/usr/src/kernels/5.14.0-340.el9.x86_64'
  CC [M]  /opt/rtl8812au/core/rtw_cmd.o
In file included from /opt/rtl8812au/include/drv_types.h:37,
                 from /opt/rtl8812au/core/rtw_cmd.c:17:
/opt/rtl8812au/include/drv_types_linux.h:20:20: error: redefinition of ‘eth_hw_addr_set’
   20 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /opt/rtl8812au/include/osdep_service_linux.h:49,
                 from /opt/rtl8812au/include/osdep_service.h:58,
                 from /opt/rtl8812au/include/drv_types.h:26,
                 from /opt/rtl8812au/core/rtw_cmd.c:17:
./include/linux/etherdevice.h:314:20: note: previous definition of ‘eth_hw_addr_set’ with type ‘void(struct net_device *, const u8 *)’ {aka ‘void(struct net_device *, const unsigned char *)’}
  314 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:298: /opt/rtl8812au/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1928: /opt/rtl8812au] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.14.0-340.el9.x86_64'
make: *** [Makefile:1726: modules] Error 2

can you please help?
Thanks

vi, 2023-08-24:
Workaround: use older commit like  ae0b0a5a04527e337bde9076843519f21bf205e0.

virer, 2023-11-29:
I've manage to fix it for Rocky9.3 in my fork available here :
https://github.com/virer/rtl8812au/tree/v5.6.4.2_rocky93
hope this help