aircrack-ng/rtl8812au #1102: 5G network is not visible on 6.2.5 kernels

KES777, 2023-08-22:
I tried to search 5G network, but only 2G available.

```
Linux work 6.2.5-060205-generic #202303110831 SMP PREEMPT_DYNAMIC Sat Mar 11 13:57:40 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```

Which and how I can provide more debug information to resolve this issue?
Thank you.

KES777, 2023-08-22:
```
$ sudo lshw -C network
[sudo] password for kes:         
  *-network                 
       description: Ethernet interface
       product: Ethernet Connection (17) I219-V
       vendor: Intel Corporation
       physical id: 1f.6
       bus info: pci@0000:00:1f.6
       logical name: eno1
       version: 11
       serial: 50:eb:f6:2a:f8:11
       capacity: 1Gbit/s
       width: 32 bits
       clock: 33MHz
       capabilities: pm msi bus_master cap_list ethernet physical tp 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=e1000e driverversion=6.2.5-060205-generic firmware=2.3-4 latency=0 link=no multicast=yes port=twisted pair
       resources: irq:145 memory:84b00000-84b1ffff
  *-network
       description: Wireless interface
       physical id: 8
       bus info: usb@1:1
       logical name: enx08beac187b82
       serial: 08:be:ac:18:7b:82
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtl8812au driverversion=6.2.5-060205-generic multicast=yes wireless=unassociated
```

2l47, 2023-08-22:
@KES777 I had this issue just a bit ago on 6.1.0; I resolved it by installing libelf-dev, recompiling, and reloading the 88XXau module.

KES777, 2023-09-13:
@2l47 I have installed `libelf-dev`. I tried recompiling and installing once more without success. I can see TP-Link-5G network, I can connect to it and work. I do not see 5G network only from DIR-822.

pedroapfilho, 2024-01-04:
I'm also having the same issue, and the weird part is that it seems that I can see them, but I can't get their SSID correctly, so I can't connect to them. When I run `nmcli dev wifi list`, this is the output (changed the BSSID just a bit):

```
IN-USE  BSSID              SSID                   MODE   CHAN  RATE        SIGNAL  BARS  SECURITY
        4A:22:54:XX:XX:FD  --                     Infra  6     270 Mbit/s  100     ▂▄▆█  WPA2
*       48:22:54:XX:XX:FD  TP-Link_37D8           Infra  6     270 Mbit/s  100     ▂▄▆█  WPA2
        EE:50:8B:XX:XX:D8  --                     Infra  6     270 Mbit/s  100     ▂▄▆█  WPA2
        5C:62:8B:XX:XX:D8  TP-Link_37D8           Infra  6     270 Mbit/s  100     ▂▄▆█  WPA2
        EE:40:8B:XX:XX:D8  --                     Infra  6     270 Mbit/s  97      ▂▄▆█  WPA2
        48:22:54:XX:XX:60  TP-Link_37D8_2.4G_EXT  Infra  6     270 Mbit/s  54      ▂▄__  WPA2
```

This is the `sudo lshw -C network`

```
  *-network
       description: Ethernet interface
       product: RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0000:02:00.0
       logical name: enp2s0f0
       version: 0e
       serial: 6c:4b:90:15:31:6a
       capacity: 1Gbit/s
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress msix vpd bus_master cap_list ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=r8169 driverversion=6.2.0-39-generic latency=0 link=no multicast=yes port=twisted pair
       resources: irq:32 ioport:ec00(size=256) memory:fea18000-fea18fff memory:fea10000-fea13fff
  *-network
       description: Wireless interface
       physical id: 4
       bus info: usb@2:3
       logical name: wlx30de4bc5e761
       serial: 30:de:4b:c5:e7:61
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtl88XXau driverversion=6.2.0-39-generic ip=192.168.2.157 multicast=yes wireless=IEEE 802.11bgn
```