aircrack-ng/rtl8812au #1103: Missing kernel headers for RaspberryPi4 (32 bit OS)

D-314, 2023-08-27:
Missing kernel headers for RaspberryPi4 (32 bit OS):
**Config:**
Hardware: RP4B, tp-link T2U plus;
OS: Clean Raspbian OS full (32bit);
**Followed commands:**
```
sudo apt-get update
sudo apt-get install raspberrypi-kernel-headers dkms -y
#sudo apt-get install bc mokutil build-essential libelf-dev linux-headers-`uname -r`
git clone -b v5.6.4.2 https://github.com/aircrack-ng/rtl8812au.git
cd rtl*
sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
sed -i 's/CONFIG_PLATFORM_ARM_RPI = n/CONFIG_PLATFORM_ARM_RPI = y/g' Makefile
export ARCH=arm
sed -i 's/^MAKE="/MAKE="ARCH=arm\ /' dkms.conf
sudo make dkms_install
```
also I tried:
```
sed -i 's/CONFIG_PLATFORM_ARM_RPI = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
sed -i 's/CONFIG_PLATFORM_ARM64_RPI = n/CONFIG_PLATFORM_ARM64_RPI = y/g' Makefile
export ARCH=arm64
sed -i 's/^MAKE="/MAKE="ARCH=arm64\ /' dkms.conf
sudo make dkms_install
```
**Error itself:**
```
mkdir: created directory '/usr/src/8812au-5.6.4.2_35491.20191025'
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025

Creating symlink /var/lib/dkms/8812au/5.6.4.2_35491.20191025/source ->
                 /usr/src/8812au-5.6.4.2_35491.20191025

DKMS: add completed.
dkms build -m 8812au -v 5.6.4.2_35491.20191025
Error! echo
Your kernel headers for kernel 6.1.21-v8+ cannot be found at
/lib/modules/6.1.21-v8+/build or /lib/modules/6.1.21-v8+/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
make: *** [Makefile:1786: dkms_install] Error 1
```
**Additional info:**
`lsusb`:
```
Bus 001 Device 003: ID 2357:0120 TP-Link Archer T2U PLUS [RTL8821AU]
```
`ls -la /lib/modules`:
```
total 24
drwxr-xr-x   6 root root 4096 May  3 03:06 .
drwxr-xr-x 107 root root 4096 Aug 28 00:16 ..
drwxr-xr-x   3 root root 4096 Aug 28 00:17 6.1.21+
drwxr-xr-x   3 root root 4096 Aug 28 00:17 6.1.21-v7+
drwxr-xr-x   3 root root 4096 Aug 28 00:17 6.1.21-v7l+
drwxr-xr-x   3 root root 4096 May  3 03:06 6.1.21-v8+
```
`ls -la /lib/modules/6.1.21-v7l+`:
```
total 2616
drwxr-xr-x  3 root root   4096 Aug 28 00:17 .
drwxr-xr-x  6 root root   4096 May  3 03:06 ..
lrwxrwxrwx  1 root root     34 Apr  5 17:01 build -> /usr/src/linux-headers-6.1.21-v7l+
drwxr-xr-x 11 root root   4096 May  3 03:06 kernel
-rw-r--r--  1 root root 617097 Apr  5 14:32 modules.alias
etc.
```
`ls -la /lib/modules/6.1.21-v8+`:
```
total 2652
drwxr-xr-x  3 root root   4096 May  3 03:06 .
drwxr-xr-x  6 root root   4096 May  3 03:06 ..
drwxr-xr-x 11 root root   4096 May  3 03:06 kernel
-rw-r--r--  1 root root 617502 Apr  5 14:32 modules.alias
etc.
```
`ls -la /usr/src`:
```
total 24
drwxr-xr-x  6 root root 4096 Aug 28 00:21 .
drwxr-xr-x 11 root root 4096 May  3 03:04 ..
drwxr-xr-x 25 root root 4096 Aug 28 00:17 linux-headers-6.1.21+
drwxr-xr-x 25 root root 4096 Aug 28 00:17 linux-headers-6.1.21-v7+
drwxr-xr-x 25 root root 4096 Aug 28 00:17 linux-headers-6.1.21-v7l+
drwxr-xr-x  3 root root 4096 May  3 03:14 sense-hat
```
**Temporary solutiuon:**
Just use (reinstall OS to) Raspbian OS **64bit**, that have /usr/src/linux-headers-6.1.21-v8+ folder and run following commands:
```
#sudo apt-get update
#sudo apt-get bc mokutil build-essential libelf-dev linux-headers-`uname -r`
sudo apt-get install raspberrypi-kernel-headers dkms -y
git clone -b v5.6.4.2 https://github.com/aircrack-ng/rtl8812au.git
cd rtl*
sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
sed -i 's/CONFIG_PLATFORM_ARM64_RPI = n/CONFIG_PLATFORM_ARM64_RPI = y/g' Makefile
export ARCH=arm64
sed -i 's/^MAKE="/MAKE="ARCH=arm64\ /' dkms.conf
sudo make dkms_install
cd ~
```
After running this commands you need to reattach the dongle (unplud and plug in again) or reboot your raspberry.

D-314, 2023-08-27:
[UPD:](https://forums.raspberrypi.com/viewtopic.php?t=350269#p2099206)
Add "arm_64bit=0" to config.txt

tarbit, 2024-03-07:
vvvvvvvvvvvvvvery thanks