aircrack-ng/rtl8812au #1108: RTL8812AU Schematics

maxlacourchristensen, 2023-09-11:
Does anyone have access to schematics? 

This chinese site appears to have something but I am unable to download it in a readable resolution. I think it requires wechat or something like that:

https://m.elecfans.com/article/1118504.html

Any chinese with access to elecfans.com who can help with downloading it?

spikeyfpv, 2024-01-24:
Hello, seeing this post a bit late, but I was wondering if you were able to download the schematic or found an alternative schematic. If so, I would be very grateful if you included a pdf or svg of it. Thanks 🙏🏻

bkaz4283, 2024-11-01:
> [RTL8812AU Schematics.pdf](https://github.com/aircrack-ng/rtl8812au/files/14206886/RTL8812AU.Schematics.pdf)

Many thanks for the link to that schematic!

Is there a BOM available too?  Because without a BOM it's actually impossible to reproduce the design.  The switching regulator inductor L1 is one good example.  It says it's 4.7uH, but for a switching regulator we need the current rating too (best of all would be to provide the exact part number).  There are a few other examples like the SP3T RF switches without part numbers...

I would like to design this RTL8812AU chip into a high-volume design (100k per month) but need more info than what's in that schematic...

Thanks!