aircrack-ng/rtl8812au #1111: BUG Compile

vincenzogianfelice, 2023-10-02:
Debian 11 (x64, kernel v5.10)

```
In file included from /opt/rtl8812au/include/drv_types.h:37,
                 from /opt/rtl8812au/core/rtw_io.c:47:
/opt/rtl8812au/include/drv_types_linux.h:20:20: error: redefinition of ‘eth_hw_addr_set’
   20 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /opt/rtl8812au/include/osdep_service_linux.h:49,
                 from /opt/rtl8812au/include/osdep_service.h:58,
                 from /opt/rtl8812au/include/drv_types.h:26,
                 from /opt/rtl8812au/core/rtw_io.c:47:
/usr/src/linux-headers-5.10.0-25-common/include/linux/etherdevice.h:309:20: note: previous definition of ‘eth_hw_addr_set’ was here
  309 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /opt/rtl8812au/include/drv_types.h:37,
                 from /opt/rtl8812au/core/rtw_debug.c:17:
/opt/rtl8812au/include/drv_types_linux.h:20:20: error: redefinition of ‘eth_hw_addr_set’
   20 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /opt/rtl8812au/include/osdep_service_linux.h:49,
                 from /opt/rtl8812au/include/osdep_service.h:58,
                 from /opt/rtl8812au/include/drv_types.h:26,
                 from /opt/rtl8812au/core/rtw_debug.c:17:
/usr/src/linux-headers-5.10.0-25-common/include/linux/etherdevice.h:309:20: note: previous definition of ‘eth_hw_addr_set’ was here
  309 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /opt/rtl8812au/include/drv_types.h:37,
                 from /opt/rtl8812au/core/rtw_cmd.c:17:
/opt/rtl8812au/include/drv_types_linux.h:20:20: error: redefinition of ‘eth_hw_addr_set’
   20 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /opt/rtl8812au/include/osdep_service_linux.h:49,
                 from /opt/rtl8812au/include/osdep_service.h:58,
                 from /opt/rtl8812au/include/drv_types.h:26,
                 from /opt/rtl8812au/core/rtw_cmd.c:17:
/usr/src/linux-headers-5.10.0-25-common/include/linux/etherdevice.h:309:20: note: previous definition of ‘eth_hw_addr_set’ was here
  309 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /opt/rtl8812au/include/drv_types.h:37,
                 from /opt/rtl8812au/core/rtw_ioctl_query.c:17:
/opt/rtl8812au/include/drv_types_linux.h:20:20: error: redefinition of ‘eth_hw_addr_set’
   20 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /opt/rtl8812au/include/osdep_service_linux.h:49,
                 from /opt/rtl8812au/include/osdep_service.h:58,
                 from /opt/rtl8812au/include/drv_types.h:26,
                 from /opt/rtl8812au/core/rtw_ioctl_query.c:17:
/usr/src/linux-headers-5.10.0-25-common/include/linux/etherdevice.h:309:20: note: previous definition of ‘eth_hw_addr_set’ was here
  309 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
make[3]: *** [/usr/src/linux-headers-5.10.0-25-common/scripts/Makefile.build:291: /opt/rtl8812au/core/rtw_io.o] Errore 1
make[3]: *** Attesa per i processi non terminati....
make[3]: *** [/usr/src/linux-headers-5.10.0-25-common/scripts/Makefile.build:291: /opt/rtl8812au/core/rtw_ioctl_query.o] Errore 1
In file included from /opt/rtl8812au/include/drv_types.h:37,
                 from /opt/rtl8812au/core/rtw_security.c:17:
/opt/rtl8812au/include/drv_types_linux.h:20:20: error: redefinition of ‘eth_hw_addr_set’
   20 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
In file included from /opt/rtl8812au/include/osdep_service_linux.h:49,
                 from /opt/rtl8812au/include/osdep_service.h:58,
                 from /opt/rtl8812au/include/drv_types.h:26,
                 from /opt/rtl8812au/core/rtw_security.c:17:
/usr/src/linux-headers-5.10.0-25-common/include/linux/etherdevice.h:309:20: note: previous definition of ‘eth_hw_addr_set’ was here
  309 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
      |                    ^~~~~~~~~~~~~~~
make[3]: *** [/usr/src/linux-headers-5.10.0-25-common/scripts/Makefile.build:291: /opt/rtl8812au/core/rtw_debug.o] Errore 1
make[3]: *** [/usr/src/linux-headers-5.10.0-25-common/scripts/Makefile.build:291: /opt/rtl8812au/core/rtw_cmd.o] Errore 1
make[3]: *** [/usr/src/linux-headers-5.10.0-25-common/scripts/Makefile.build:291: /opt/rtl8812au/core/rtw_security.o] Errore 1
make[2]: *** [/usr/src/linux-headers-5.10.0-25-common/Makefile:1856: /opt/rtl8812au] Errore 2
make[1]: *** [/usr/src/linux-headers-5.10.0-25-common/Makefile:192: __sub-make] Errore 2
make[1]: uscita dalla directory «/usr/src/linux-headers-5.10.0-25-amd64»
make: *** [Makefile:1726: modules] Errore 2
```

CGarces, 2023-10-11:
The package linux-headers-5.10.0-25-common match with  5.10.191, so is patched.

You must manually remove this lines 

https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.2/include/drv_types_linux.h#L18-L24

Until I figure out how patch it, the fast solution is set additional `#if` sentences for each LTS kernel version.

emilcarr, 2023-10-17:
This fix worked for me on Debian 11 / linux 5.10.0-26-amd64 with latest commit. Thanks.

sarrchri, 2023-11-06:
I think doing same type of thing I did with the kernel 6.x build fix could be enough. Although I'm not sure how to implement it, since Ubuntu uses different kernel versioning style from Debian. (Ubuntu seems to stick to 5.10.**0** which they then append their own patch version like **-123** to.)