aircrack-ng/rtl8812au #1120: Can't build for Raspberry Pi 4 running Ubuntu server 23.04

marek-guran, 2023-10-31:
Kernel:  6.2.0-1016-raspi
Using: linux-headers-6.2.0-1016-raspi

```CC [M]  /home/pi/rtl8812au/os_dep/linux/wifi_regd.o
/home/pi/rtl8812au/os_dep/linux/wifi_regd.c: In function ‘_rtw_regd_init_wiphy’:
/home/pi/rtl8812au/os_dep/linux/wifi_regd.c:400:36: error: ‘REGULATORY_IGNORE_STALE_KICKOFF’ undeclared (first use in this function)
  400 |       wiphy->regulatory_flags |= REGULATORY_IGNORE_STALE_KICKOFF;
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                     /home/pi/rtl8812au/os_dep/linux/wifi_regd.c:400:36: note: each undeclared identifier is reported only once for each function it appears in                     make[2]: *** [scripts/Makefile.build:260: /home/pi/rtl8812au/os_dep/linux/wifi_regd.o] Error 1
make[1]: *** [Makefile:2026: /home/pi/rtl8812au] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.2.0-1016-raspi'
make: *** [Makefile:1728: modules] Error 2```

sarrchri, 2023-11-01:
Yeah, this is due to changes backported from kernel 6.5, which Ubuntu decided to include in 6.2 while many other distros skipped 6.2 completely.

I have PR which I'll push the fix into. (#1119)

marek-guran, 2023-11-01:
> Yeah, this is due to changes backported from kernel 6.5, which Ubuntu decided to include in 6.2 while many other distros skipped 6.2 completely.
> 
> I have PR which I'll push the fix into. (#1119)

Thank you very much

sarrchri, 2023-11-01:
I've just now updated the PR with a more proper fix taking into account kernels 6.3, 6.4 and Ubuntu 23.04 with kernel 6.2.

closed 2023-11-01 by marek-guran