aircrack-ng/rtl8812au #1130: Compiling on newest Kali Linux

christoffhs, 2024-01-04:
Hi,
when following the Readme to install the drivers I'm getting the error:

1st: While installing the prerequisite for Kali:
Unable to locate package linux-headers-6.1.64-v8

2nd: While trying anyways:
sudo make dkms_install

In the make.log
cannot open file include/generated/asm-offset.h for reading: No such file or directory....

[image]

TrueSnakeboy, 2024-01-07:
Im having the same issue

Jezza12345, 2024-01-13:
Yep - I've got the same issue trying to configure a ALFA AWUS036ACH using Kali on a Raspberry Pi 5.

appelboom, 2024-01-21:
kali@kali:~$ cd /usr/src/kernel
kali@kali:/usr/src/kernel/$ sudo git clean -fdx && sudo make bcm2711_defconfig && sudo make modules_prepare

Jezza12345, 2024-01-21:
Thanks very much @appelboom.  That worked! Greatly appreciated!

notsapinho, 2024-01-31:
> kali@kali:~$ cd /usr/src/kernel kali@kali:/usr/src/kernel/$ sudo git clean -fdx && sudo make bcm2711_defconfig && sudo make modules_prepare

holy shit bro thank you very much

julien-stiker, 2024-02-28:
Doesn't work on RPI5 with Kali 2024.1. :
  HOSTCC  scripts/basic/fixdep
make[1]: *** No rule to make target 'bcm2711_deconfig'.  Stop.
make: *** [Makefile:704: bcm2711_deconfig] Error 2

And 

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package raspberrypi-kernel-headers

bob454522, 2024-03-02:
> Doesn't work on RPI5 with Kali 2024.1. : HOSTCC scripts/basic/fixdep make[1]: *** No rule to make target 'bcm2711_deconfig'. Stop. make: *** [Makefile:704: bcm2711_deconfig] Error 2
>
> And
>
> Reading [...]

I was able to get this to work on the latest Kali 2024.1 on a rasPI 5  (while at first having lots of difficulties) ;

I think the fix was to first do:
sudo apt-get update && sudo apt-get upgrade 
(was a long process from fresh 2024.1 iso -> sd card burn).

then i had to be sure to 
sudo make dkms_remove   (to remove any prior failures, apparently)
then;
cd ~/rtl8812au
sudo make dkms_install

(if you still have issues reply back, as i did a few other things that i dont think are related, so i left them out of the steps above,  but my history below is an exact, full dump, nothing left out -  the steps I left out in above are mainly following the steps here before dkms_install )  
https://github.com/aircrack-ng/rtl8812au?tab=readme-ov-file#for-raspberry-rpi
(arm64 - be sure to use those cmds if pi5 or 64bit kali)
then did the process , but first try wo these).

To be clear I have not tested monitor mode yet will in a few minutes, but this is progress as I was finally able to get a successful make and make install, and for the first time my RTL8812AU is showing up for cmd:  ``` ip a ```  as wlan1

fwiw, below is my full bash history (note there are a few reboots in-between some of these cmds):

```
sudo apt update && sudo apt dist-upgrade
ip a
sudo airmon-ng
lsusb
dkms
sudo apt-get install dkms
git clone -b v5.6.4.2 https://github.com/aircrack-ng/rtl8812au.git\ncd rtl*
sudo make dkms_install
lsusb
sudo apt install dkms\n\nsudo apt install build-essential libelf-dev\n\nsudo apt install raspberrypi-kernel-headers
sudo apt install dkms
sudo make dkms_install
ip a
cat /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/make.log
lsusb
sudo apt update
sudo apt install realtek-rtl88xxau-dkms
sudo shutdown -r now
top
top
sudo air
sudo airmon-ng
lsusb
apt-get update && apt-get upgrade
sudo apt-get update && sudo apt-get upgrade
lsusb
ip a
sudo shutdown -r now
lsusb
ip a
lsusb
aircrack-ng
sudo aircrack-ng
airmon-ng
sudo airmon-ng
lsusb
cd rtl*
sudo make dkms_install
sudo make dkms_remove
sudo make dkms_install
sudo make dkms_remove
make && make install
sudo make dkms_remove
sudo apt-get install raspberrypi-kernel-headers
$ sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile\n$ sed -i 's/CONFIG_PLATFORM_ARM64_RPI = n/CONFIG_PLATFORM_ARM64_RPI = y/g' Makefile
sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile\nsed -i 's/CONFIG_PLATFORM_ARM64_RPI = n/CONFIG_PLATFORM_ARM64_RPI = y/g' Makefile
sudo make dkms_install
sudo make dkms_remove
make && make install
export ARCH=arm64\nsed -i 's/^MAKE="/MAKE="ARCH=arm64\ /' dkms.conf
sudo make dkms_install
sudo make dkms_remove
make && make install
sudo make dkms_install
lsusb
airmon-ng
sudo airmon-ng
cd /usr/src/kernel
sudo git clean -fdx && sudo make bcm2711_defconfig && sudo make modules_prepare
lsusb
sudo airmon-ng
cd ~/rtl8812au
sudo make dkms_install
sudo make dkms_remove
sudo make dkms_install
ip a
history
sudo shutdown -r now
top
lsusb
ip a
```

aregshmavonyan, 2024-08-26:
Will this work for rtl8814au?

> > Doesn't work on RPI5 with Kali 2024.1. : HOSTCC scripts/basic/fixdep make[1]: *** No rule to make target 'bcm2711_deconfig'. Stop. make: *** [Makefile:704: bcm2711_deconfig] Error 2
> > 
> > And
> > 
> > Reading package lists... Done Building dependency tree... Done Reading state information... Done E: Unable to locate package raspberrypi-kernel-headers
> 
> I was able to get this to work on the latest Kali 2024.1 on a rasPI 5  (while at first having lots of difficulties) ;
> 
> I think the fix was to first do:
> `sudo apt-get update && sudo apt-get upgrade `
> (was a long process from fresh 2024.1 iso -> sd card burn).
> 
> then i had to be sure to 
> `sudo make dkms_remove`   (to remove any prior failures, apparently)
> then;
> `cd ~/rtl8812au`
> `sudo make dkms_install`
> 
> (if you still have issues reply back, as i did a few other things that i dont think are related, so i left them out of the steps above,  but my history below is an exact, full dump, nothing left out -  the steps I left out in above are mainly following the steps at this link-anchor,  before dkms_install )  
> 
> https://github.com/aircrack-ng/rtl8812au?tab=readme-ov-file#for-raspberry-rpi
> (arm64 - be sure to use those cmds if pi5 or 64bit kali)
> 
> To be clear I have not tested monitor mode yet, will in a few minutes, but this is progress as I was finally able to get a successful dkms_install, and for the first time my RTL8812AU is showing up for cmd:  ``` ip a ```  as wlan1
> 
> EDIT:  also fwiw - The Blue LED on my alfa AWUS036ACM is illuminating/on now for the first time, prior to this I could never get it to illuminate at all, even after trying various different Linux distros / drivers -  The only system I could get it to illuminate /work on was a Windows 11 laptop.
> 
> I am using a very good and powered, usb3 hub (w external 12v power into it).
> 
> EDIT2- monitor mode seems to be working,  in kismet im seeing several 5ghz APs / SSIDs -  will update if its not stable or any other issues. (but so far seems good for past 3 or 4 minutes!)
> 
> fwiw, below is my full bash history (note there are a few reboots in-between some of these cmds):
> 
> ```
> sudo apt update && sudo apt dist-upgrade
> ip a
> sudo airmon-ng
> lsusb
> dkms
> sudo apt-get install dkms
> git clone -b v5.6.4.2 https://github.com/aircrack-ng/rtl8812au.git\ncd rtl*
> sudo make dkms_install
> lsusb
> sudo apt install dkms\n\nsudo apt install build-essential libelf-dev\n\nsudo apt install raspberrypi-kernel-headers
> sudo apt install dkms
> sudo make dkms_install
> ip a
> cat /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/make.log
> lsusb
> sudo apt update
> sudo apt install realtek-rtl88xxau-dkms
> sudo shutdown -r now
> top
> top
> sudo air
> sudo airmon-ng
> lsusb
> apt-get update && apt-get upgrade
> sudo apt-get update && sudo apt-get upgrade
> lsusb
> ip a
> sudo shutdown -r now
> lsusb
> ip a
> lsusb
> aircrack-ng
> sudo aircrack-ng
> airmon-ng
> sudo airmon-ng
> lsusb
> cd rtl*
> sudo make dkms_install
> sudo make dkms_remove
> sudo make dkms_install
> sudo make dkms_remove
> make && make install
> sudo make dkms_remove
> sudo apt-get install raspberrypi-kernel-headers
> $ sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile\n$ sed -i 's/CONFIG_PLATFORM_ARM64_RPI = n/CONFIG_PLATFORM_ARM64_RPI = y/g' Makefile
> sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile\nsed -i 's/CONFIG_PLATFORM_ARM64_RPI = n/CONFIG_PLATFORM_ARM64_RPI = y/g' Makefile
> sudo make dkms_install
> sudo make dkms_remove
> make && make install
> export ARCH=arm64\nsed -i 's/^MAKE="/MAKE="ARCH=arm64\ /' dkms.conf
> sudo make dkms_install
> sudo make dkms_remove
> make && make install
> sudo make dkms_install
> lsusb
> airmon-ng
> sudo airmon-ng
> cd /usr/src/kernel
> sudo git clean -fdx && sudo make bcm2711_defconfig && sudo make modules_prepare
> lsusb
> sudo airmon-ng
> cd ~/rtl8812au
> sudo make dkms_install
> sudo make dkms_remove
> sudo make dkms_install
> ip a
> history
> sudo shutdown -r now
> top
> lsusb
> ip a
> ```