aircrack-ng/rtl8812au #1132: Broken after kernel update

pindjur, 2024-01-11:
After updating the kernel from version 6.2.0-26-generic to 6.5.0-14-generic using Ubuntu's Software Updater, my wifi adapter is no longer recognized. I attempted to reinstall the driver, reconnect the adapter, and reboot the system, but got the same rasult. Is anyone else experiencing this problem?

windler47, 2024-01-11:
Yes, `sudo dkms autoinstall` fixed it for me.

pindjur, 2024-01-12:
Thank you for the hint @windler47 . My problem was that I removed dkms package after initial installation , then after updating kernel, information regarding previous module build was lost. I solved it by manually removing driver from old kernel /usr/lib/modules/[old kernel]/kernel/drivers/net/wireless/[wifi driver] and reinstalling driver.

closed 2024-01-12 by pindjur