aircrack-ng/rtl8812au #1133: Aircrack with Ubuntu in Windows Subsystem for Window

Roadracerboy, 2024-01-14:
I been using Aircrack with VMware and Kali Linux with success, the last week I been working to get Aircrack work in Ubuntu and WSL inside Windows
(windows subsystem for Linux).

First it appeared impossible to get the wifi-device (rtl8812au) to show up in Ubuntu but then I find this fantastic "USBIPD-win" that connects the usb from windows into linux.

Installed aircrack-ng, the driver and NetworkManager

I get it to work like expected and in Ubuntu the device shows up with lusub

```
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
But with nmcli the device does not show up as wlan0 as it does fine in Kali/vmware

```
$ nmcli d
DEVICE  TYPE      STATE      CONNECTION
eth0    ethernet  unmanaged  --
lo      loopback  unmanaged  --
```
Airmon-ng doesnt either show my device (wlan0) as it does fine in Kali/vmware

```
$ sudo airmon-ng
PHY     Interface       Driver          Chipset
```
So now Im stucked here and need some advices to wich direction I must go, what am I missing?

Windows software for sharing locally connected USB devices to other machines, including Hyper-V guests and WSL 2.
https://github.com/dorssel/usbipd-win

WSL support
https://github.com/dorssel/usbipd-win/wiki/WSL-support

The installed driver: 
https://github.com/aircrack-ng/rtl8812au

My aircrack-installation:
https://github.com/aircrack-ng/rtl8812au

Linux DESKTOP-NGT 5.15.133.1-microsoft-standard-WSL2 #1 SMP Thu Oct 5 21:02:42 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

RazviAlex, 2024-02-17:
same problem here.. when trying to make dkms_install 

`mkdir: created directory '/usr/src/8812au-5.6.4.2_35491.20191025'
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025
Creating symlink /var/lib/dkms/8812au/5.6.4.2_35491.20191025/source -> /usr/src/8812au-5.6.4.2_35491.20191025
dkms build -m 8812au -v 5.6.4.2_35491.20191025
Error! Your kernel headers for kernel 5.15.133.1-microsoft-standard-WSL2 cannot be found at /lib/modules/5.15.133.1-microsoft-standard-WSL2/build or /lib/modules/5.15.133.1-microsoft-standard-WSL2/source.
Please install the linux-headers-5.15.133.1-microsoft-standard-WSL2 package or use the --kernelsourcedir option to tell DKMS where it's located.
make: *** [Makefile:1788: dkms_install] Error 1`

hwertz, 2024-03-04:
`Please install the linux-headers-5.15.133.1-microsoft-standard-WSL2 package or use the --kernelsourcedir option to tell DKMS where it's located.`

This looks like the relevant part.  What happens if from a WSL shell you run "sudo apt-get install linux-headers-5.15.133.1-microsoft-standard-WSL2"?  (I've read comments that with WSL, you may have to run "sudo apt update" first, I think Ubuntu and Debian ship with a list of available packages "out of the box" (which is updated when you check for package updates) but perhaps WSL doesn't, this gets that list.)

This sounds irregular, but I mean it should work.  VirtualBox on Linux people were using PCI passhtrough to pass through whole Nvidia or AMD video cards and stuff, like they'd have 1 head running Linux and a second monitor plugged into the card, Windows or macOS running in there.

Iyabah, 2025-05-12:
so there is no other my Pc's nerwork adapter can be recognised on WSL ??

hwertz, 2025-05-13:
> so there is no other my Pc's nerwork adapter can be recognised on WSL ??

Nope!  I mean, I assume it'll show up in WSL in terms of, if your wifi is connected to the internet, so will the stuff in WSL.  But Windows network APIs (NDIS and Winsock) do not support wifi packet injection and associated goodies required for tools like aircrack to work.

Get a wifi adapter that supports packet injection etc. (https://deviwiki.com/wiki/List_of_Wireless_Adapters_That_Support_Monitor_Mode_and_Packet_Injection ) and play around with USBPID-Win etc... or just make it easy and dual boot into Kali Linux (which has the aircrack drivers and tools etc. all installed.)