aircrack-ng/rtl8812au #1140: Driver limit set wireless regulatory domain

LLH-l, 2024-01-25:
Hello, 
In kali 2023.4  or wifislax64-3.4-final, Driver install success .  .  . 
But canno set "phy#0" to IN by using command method set wireless regulatory domain
"phy#0" , it is wireless interface network card ( RTL8812AU )

```
$ sudo iw reg set IN
$ iw reg get
global
country IN: DFS-UNSET
	(2402 - 2482 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 30), (N/A)
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS
	(5470 - 5725 @ 160), (N/A, 24), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 30), (N/A)

phy#0
country 99: DFS-UNSET
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 40), (N/A, 20), (N/A), PASSIVE-SCAN
        (2474 - 2494 @ 40), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5140 - 5860 @ 80), (N/A, 30), (N/A), PASSIVE-SCAN

$ iw dev    
phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 74:DA:38:ff:ff:ff
                type managed
                channel 2 (2417 MHz), width: 20 MHz, center1: 2417 MHz
                txpower 20.00 dBm
```

Seems to be driver restricted, canno by command method set wireless regulatory domain !
In normal circumstances, it not like this

dubhater, 2024-02-14:
Is the wireless-regdb package installed? If not, install it and reboot.

LLH-l, 2024-02-16:
> Is the wireless-regdb package installed? If not, install it and reboot.

Yeah, wireless-regdb package installed
This issue exists in the wifislax64-3.4 final
Your can try reproducing this issue on KALI LINUX and wifislax64-3.4 final 
If you don't have this issue, maybe my problem

closed 2024-03-26 by LLH-l

qh-code, 2025-04-12:
OK. So the issue has been closed. But what is the solution?

dubhater, 2025-04-12:
You could try the ``rtw_country_code`` module parameter.

Or you could try kernel 6.14. It has a driver built-in and it definitely supports ``iw reg set __``.