aircrack-ng/rtl8812au #1141: Why has this so widely used driver not yet been submitted to the Linux kernel?

manuelpaulo, 2024-01-30:
Any particular reason why not? How can the community help? Thanks.

dubhater, 2024-02-12:
The kernel maintainers don't like this code, for good reason:
* It has its own 802.11 stack. The kernel maintainers want drivers to use the kernel's own 802.11 stack (cfg80211, mac80211). You know, don't reinvent the wheel. It's a lot of code, many opportunities for bugs to appear, big maintenance burden.
* It doesn't comply with the kernel's coding style standards.
* It duplicates a lot of code found in other Realtek drivers. If they accept one driver, they should accept others, but there are so, so many duplicated files with smaller or bigger differences.

These are the reasons I remember right now. Even if someone were to send the driver to the Linux maintainers, they would not accept it.

The good news is the rtlwifi driver, which has been in the kernel for many years now, supports the PCI version of this chip. Making it support the USB version as well is not that hard. You can help me do that by sending/buying me a device with this chip in it.

LLH-l, 2024-02-16:
They in working support Linux kernel, 
https://github.com/morrownr/8812au-20210629/issues/114

dubhater, 2024-02-22:
@LLH-l What they're discussing in that issue is another version of the driver you see in this repository. It has all the same problems I mentioned above, it's just a little newer.

chewitt, 2024-02-25:
> The good news is the rtlwifi driver, which has been in the kernel for many years now, supports the PCI version of this chip. Making it support the USB version as well is not that hard. You can help me do that by sending/buying me a device with this chip in it.

@dubhater If you're serious about pushing USB support upstream feel free to acquire the missing device and expense the cost to (claim a remibursement from) the LibreELEC distro here https://opencollective.com/libreelec - Just link/ref. this conversation in the claim.

dubhater, 2024-02-25:
Oh, that's interesting. Hey, I noticed you have ``packages/linux-drivers/RTL8192DU``. I'm currently cleaning up a new rtl8192du driver for inclusion in rtlwifi. Can I claim a reimbursement for that device too? It was 7.44 USD.

chewitt, 2024-02-25:
> Can I claim a reimbursement for that device too? It was 7.44 USD.

Sure. Just submit the purchase receipt with the title "Purchase of RTL8192DU hardware to support upstream development" or similar and reference this thread somewhere.

dubhater, 2024-02-25:
Cool! I will do that.