aircrack-ng/rtl8812au #1153: Edimax AC600 (+potentially other devices) no longer working after c7f8f6e3633bd91a9f359eb3406beb972e58cd7b

olaf-otto, 2024-05-11:
I am compiling the drivers from source on RPI Raspian 32 bit on a Raspberry Pi Zero W following the procedure documented here: 
https://edimax.freshdesk.com/support/solutions/articles/14000032146-how-to-install-ew-7811-ac600-series-and-ew-7822uac-adapters-on-raspberry-pi

The resulting driver worked perfectly fine a week ago. Starting with commit  ("[Remove leftovers of Intel WIDI and take a general cleanup](https://github.com/aircrack-ng/rtl8812au/commit/c7f8f6e3633bd91a9f359eb3406beb972e58cd7b)"), the driver has stopped working. The USB device is recognized and the module gets loaded, but any interaction with the device results in timeouts, e.g. "Failed to read scan data : Resource temporarily unavailable". In addition, detaching the USB Edimax AC600 causes a kernel freeze.

Unfortunately I am unfamiliar with the code base and lack C expertise, thus I have no patch suggestion to contribute.

ObsidianMaximus, 2024-05-12:
Can confirm that the recent commits (as mentioned above) have caused my "TP Link Archer T2U Plus" to not work on both Debian and Ubuntu server.

I use nmcli to connect to wifi networks.

When I issued "nmcli d wifi list", it did not list any networks, even after running "nmcli r wifi on".

And getting the same null pointer error as above on boot.

Going back to 63cf0b4 commit fixed it.

kstepanovdev, 2024-05-12:
Can confirm. The recent confirm doesn't work with "TP Link Archer T2U Plus". 
That adapter is greyed out in nm-applet. 
`ip link` shows its `state DORMANT`. Turning it on via `ip link set "interface_name" up` gives nothing.
`nmcli device wifi list ifname "interface_name"` shows empty available options

Rolling back to https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d fixes the issue.

georgiyozhegov, 2024-05-21:
I have the same problem with TP Link Archer T2U Plus. I have Arch Installed and recently, after update to 6.9.1 linux kernel wifi just stopped work. In GNOME control center wifi section is available, but there are no networks listed (but they're exist). Maybe it's a problem with `wpa_supplicant`, but I'm not sure. 

Logs:
```
 4:05:53 PM NetworkManager: <info>  [1716293153.1723] device (p2p-dev-wlp0s29u1u8): supplicant management interface state: disconnected -> inactive
 4:05:53 PM wpa_supplicant: wlp0s29u1u8: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
 4:05:53 PM NetworkManager: <info>  [1716293153.1639] device (p2p-dev-wlp0s29u1u8): supplicant management interface state: inactive -> disconnected
 4:05:53 PM wpa_supplicant: wlp0s29u1u8: CTRL-EVENT-ASSOC-REJECT status_code=1
 4:05:53 PM NetworkManager: <info>  [1716293153.1625] device (p2p-dev-wlp0s29u1u8): supplicant management interface state: interface_disabled -> inactive
 4:05:53 PM wpa_supplicant: wlp0s29u1u8: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
 4:05:53 PM NetworkManager: <info>  [1716293153.1624] device (wlp0s29u1u8): supplicant interface state: interface_disabled -> inactive
 4:05:53 PM wpa_supplicant: wlp0s29u1u8: CTRL-EVENT-ASSOC-REJECT status_code=1
 4:05:53 PM NetworkManager: <info>  [1716293153.1171] device (p2p-dev-wlp0s29u1u8): supplicant management interface state: disconnected -> interface_disabled
 4:05:39 PM wpa_supplicant: wlp0s29u1u8: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
 4:05:39 PM NetworkManager: <warn>  [1716293139.6667] device (wlp0s29u1u8): Activation: failed for connection 'Hotspot'
 4:05:29 PM wpa_supplicant: wlp0s29u1u8: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
 4:05:29 PM NetworkManager: <warn>  [1716293129.2151] device (wlp0s29u1u8): set-hw-addr: failed to set MAC address to B2:0C:28:F3:7A:CA (scanning) (NME_UNSPEC)
 4:05:24 PM wpa_supplicant: wlp0s29u1u8: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
 4:05:24 PM NetworkManager: <info>  [1716293124.7318] device (wlp0s29u1u8): state change: prepare -> failed (reason 'config-failed', sys-iface-state: 'managed')
 4:05:24 PM wpa_supplicant: wlp0s29u1u8: CTRL-EVENT-ASSOC-REJECT status_code=1
 4:05:24 PM NetworkManager: <warn>  [1716293124.7317] device (wlp0s29u1u8): set-hw-addr: failed to reset MAC address to 6C:5A:B0:F0:92:2C (preserve) (NME_UNSPEC)
 4:05:10 PM wpa_supplicant: wlp0s29u1u8: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
 4:05:10 PM NetworkManager: <warn>  [1716293110.0117] device (wlp0s29u1u8): set-hw-addr: failed to set MAC address to 56:85:67:44:AE:38 (scanning) (NME_UNSPEC)
 4:04:49 PM wpa_supplicant: wlp0s29u1u8: CTRL-EVENT-SCAN-FAILED ret=-16
 4:00:04 PM NetworkManager: <info>  [1716292804.8243] device (p2p-dev-wlp0s29u1u8): state change: unavailable -> disconnected (reason 'none', sys-iface-state: 'managed')
 4:00:03 PM kernel: rtl88XXau 1-1.8:1.0 wlp0s29u1u8: renamed from wlan0
```

juceda, 2024-06-18:
Same for me... :/

gauner1986, 2024-06-27:
I'm using a TP Link Archer T4U v2 and also hit these problems. By accident i saw this issue and a revert to the already mentioned commit fixed it for me.
I would also think that this is a critical issue that needs to be fixed ASAP.

tomkcook, 2024-07-04:
Same issue here.

Platform is Raspberry Pi 4B aarch64.

OS is Ubuntu 22.04.

Kernel is 5.15.0-1055-raspi.

Hardware reports as:

```
Bus 001 Device 014: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
```

A scan times out with `v5.6.4.2` but stepping back to 63cf0b4584aa8878b0fe8ab38017f31c319bde3d resolves it.

aptitudex, 2024-10-28:
I'm using an Archer T2U Nano on a Cortex A8 device (BeagleBone Black) and had the same problem reported here. Rollback to aforementioned commit and reinstall via DKMS brought up the network scan, working perfectly now.