aircrack-ng/rtl8812au #1155: 6.6.31 on x86_64 , adapter comes up but doesnt transfer any frames

Simon566, 2024-05-19:
funny thing is , that 6.6.30 with the same kernel config was working 

something in between killed the functionality ( its a 8812au adapter )

shaked8634, 2024-05-25:
I may be affected from the same issue.
Managed mode works great, but monitor mode has stopped working recently (no packets are seen and no error in dmesg).

* I have 2 8821AU USB dongles (Alpha AWUS036ACS and TP-Link Archer T2U Nano).
* I tried 2 kernels: 6.6.31 and 6.9.1.
* I tried the new rtw88 drivers.

I suspect that it's something to do with [nl80211](https://github.com/morrownr/8821au-20210708/issues/133#issuecomment-2123079774). Though it only reported on 6.9.

koku17, 2024-05-25:
* It was working fine before update. However after the update to kernel 6.9.1 (Arch linux) the problem started.
* I was not able to connect to wifi from Archer T2U Nano, the device was 'unavailable' in `nmcli`.
* I used `rfkill` to remove soft lock on the device, the device was shown as 'disconnected' in `nmcli` and tried to search for wifi networks buy no success on that.

* Later I downgrade the kernel to version 6.6.3-2(lts) after removing the modules in version 6.9.1.
* After reinstalling the modules via dkms for lts I had similar problem,
this time device was 'disconnected' instead of 'unmanaged', another reboot I tested the wifi same status 'disconnect'.

* After I installed Debian (Kernel 6.1.0) and tested again same error as the long-term kernel 6.6.3 as in Arch linux

#### status of NetworkManager in Debian
```
● NetworkManager.service - Network Manager
Loaded: loaded (/lib/systemd/system/NetworkManager.service; enabled; preset: enabled)
Active: active (running) since Sat 2024-05-25 20:20:54 IST; 6min ago
Docs: man:NetworkManager(8)
Main PID: 850 (NetworkManager)
Tasks: 3 (limit: 33335)
Memory: 18.4M
CPU: 153ms
CGroup: /system.slice/NetworkManager.service
└─850 /usr/sbin/NetworkManager --no-daemon

May 25 20:24:24 myhost NetworkManager[850]: <info>  [1716648864.3519] device (enx5ed7df4dee2e): state change: ip-config -> ip-check (reason 'none', sys-iface-state: 'managed')
May 25 20:24:24 myhost NetworkManager[850]: <info>  [1716648864.3967] device (enx5ed7df4dee2e): state change: ip-check -> secondaries (reason 'none', sys-iface-state: 'managed')
May 25 20:24:24 myhost NetworkManager[850]: <info>  [1716648864.3968] device (enx5ed7df4dee2e): state change: secondaries -> activated (reason 'none', sys-iface-state: 'managed')
May 25 20:24:24 myhost NetworkManager[850]: <info>  [1716648864.3970] manager: NetworkManager state is now CONNECTED_SITE
May 25 20:24:24 myhost NetworkManager[850]: <info>  [1716648864.3972] device (enx5ed7df4dee2e): Activation: successful, device activated.
May 25 20:24:24 myhost NetworkManager[850]: <info>  [1716648864.3974] manager: NetworkManager state is now CONNECTED_GLOBAL
May 25 20:26:06 myhost NetworkManager[850]: <warn>  [1716648966.7345] platform-linux: do-change-link[3]: failure 1 (Operation not permitted)
May 25 20:26:06 myhost NetworkManager[850]: <warn>  [1716648966.7345] device (wlx3460f9103cfd): set-hw-addr: failedto set MAC address to xx:yy:zz:xx:yy:zz (scanning) (NME_UNSPEC)
May 25 20:26:06 myhost NetworkManager[850]: <info>  [1716648966.8424] device (wlx3460f9103cfd): supplicant interface state: disconnected -> inactive
May 25 20:26:06 myhost NetworkManager[850]: <info>  [1716648966.8425] device (p2p-dev-wlx3460f9103cfd): supplicant management interface state: disconnected -> inactive
```

### Kernel (6.1.0) dmesg
```
[    8.725309] ------------[ cut here ]------------
[    8.725311] WARNING: CPU: 1 PID: 463 at /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_odm.c:62 rtw_odm_init_ic_type+0x40/0x50 [88XXau]
[    8.725341] Modules linked in: irqbypass bluetooth(+) 88XXau(OE+) ghash_clmulni_intel sha256_ssse3 sha1_ssse3 snd_hda_codec_realtek jitterentropy_rng snd_hda_codec_generic snd_hda_codec_hdmi sha512_ssse3 uvcvideo(+) snd_usb_audio(+) sha512_generic snd_hda_intel videobuf2_vmalloc videobuf2_memops snd_intel_dspcfg videobuf2_v4l2 snd_intel_sdw_acpi videobuf2_common aesni_intel snd_hda_codec snd_usbmidi_lib snd_rawmidi crypto_simd videodev cryptd ctr snd_hda_core snd_seq_device cfg80211 snd_hwdep mc eeepc_wmi drbg asus_wmi snd_pcm rapl battery platform_profile ansi_cprng sparse_keymap sp5100_tco snd_timer ledtrig_audio ecdh_generic pcspkr wmi_bmof k10temp watchdog snd ccp rfkill soundcore ecc joydev acpi_cpufreq sg evdev msr parport_pc ppdev lp parport fuse loop dm_mod efi_pstore configfs efivarfs ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic hid_generic usbhid hid amdgpu gpu_sched drm_buddy i2c_algo_bit drm_display_helper cec sd_mod rc_core drm_ttm_helper ttm
[    8.725418]  drm_kms_helper ahci xhci_pci nvme libahci xhci_hcd nvme_core libata r8169 t10_pi drm realtek crc32_pclmul mdio_devres usbcore scsi_mod crc32c_intel crc64_rocksoft libphy crc64 crc_t10dif i2c_piix4 crct10dif_genericscsi_common crct10dif_pclmul usb_common video crct10dif_common wmi button
[    8.725445] CPU: 1 PID: 463 Comm: (udev-worker) Tainted: G           OE      6.1.0-21-amd64 #1  Debian 6.1.90-1
[    8.725449] Hardware name: System manufacturer System Product Name/TUF GAMING X570-PLUS, BIOS 4403 04/28/2022
[    8.725450] RIP: 0010:rtw_odm_init_ic_type+0x40/0x50 [88XXau]
[    8.725487] Code: 3e 00 00 80 fa 0e 76 17 8b 15 f8 3d 0a 00 85 d2 74 18 be 04 00 00 00 48 89 c7 e9 0b 08 07 00 8b 14 95 90 7a b2 c1 85 d2 75 e8 <0f> 0b be 04 00 00 00 48 89 c7 e9 f1 07 07 00 90 0f 1f 44 00 00 48
[    8.725489] RSP: 0018:ffffb98680a2ba70 EFLAGS: 00010246
[    8.725492] RAX: ffffb9868135eed0 RBX: ffffb98680757000 RCX: 0000000000000002
[    8.725494] RDX: 0000000000000000 RSI: ffffffffc1b379cd RDI: ffffb98680757000
[    8.725495] RBP: ffff9e36696b7000 R08: ffffffffc1b379cc R09: 0000000000000000
[    8.725497] R10: ffffffffffffffff R11: ffffb98680a2b9c9 R12: ffff9e3669769800
[    8.725498] R13: ffffffffc1b8a440 R14: ffff9e3669769800 R15: ffff9e36de79b000
[    8.725500] FS:  00007f66bae928c0(0000) GS:ffff9e3c51c40000(0000) knlGS:0000000000000000
[    8.725501] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[... cut ...]
[    8.802106] SVM: LBR virtualization supported
[    8.812223] MCE: In-kernel MCE decoding enabled.
[    8.902758] intel_rapl_common: Found RAPL domain package
[    8.902761] intel_rapl_common: Found RAPL domain core
[    9.010018] Bluetooth: hci0: RTL: fw version 0x09a98a6b
[    9.271300] usb 3-1: 88XXau 34:60:f9:10:3c:fd hw_info[107]
[    9.277163] ------------[ cut here ]------------
[    9.277164] WARNING: CPU: 13 PID: 463 at /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_odm.c:62 rtw_odm_init_ic_type+0x40/0x50 [88XXau]
```

koku17, 2024-05-27:
I have a got a temporary solution, but not good for bleeding edge kernels

I used
```
$ git log --oneline
b44d288 Add logfile script
475e609 Add Android 10 docs/scripts to 'android' dir/collection
9d6b253 Add Android 11 docs/scripts to 'android' dir/collection
c7f8f6e Remove leftovers of Intel WIDI and take a general cleanup
63cf0b4 Merge pull request #1149 from alium/v5.6.4.2
860ccf3 Update ioctl_cfg80211.c -  support for kernel 6.9
4f645ee Merge pull request #1147 from heitbaum/patch-2
123e373 Update Makefile to error on incompatible-pointer-types
f23979f Update Makefile
98895e9 Merge pull request #1134 from crivasr/v5.6.4.2
6b6f134 Update Makefile
135756a Merge pull request #1145 from alium/v5.6.4.2
88d0f5c Update usb_intf.c for kernel 6.8
d3feb6a Update os_intfs.c for kernel 6.8
c0d1681 Fix change_beacon for kernel 6.7
a3e0c0b Merge pull request #1122 from bwh-mind/rssi-notif
d07318c Merge pull request #1123 from bwh-mind/compat
b3f7e7a Fix eth_hw_addr_set() definition for compatibility with stable branches
8b485b7 Make use of MODULE_IMPORT_NS conditional
5b90ddb ioctl_cfg80211: Add support for standard RSSI notifications
4a983e4 Merge pull request #1119 from sarrchri/fix-build-6x
9d4d3f5 Fix build on Ubuntu 23.04 (kernel 6.2).
d8ad927 Fix build on 6.x kernels.
81be2f4 Add kernel v6.6 support
8954f2b Fix the VFS internal message
04f600e Merge pull request #1097 from seppzer0/v5.6.4.2
eff5b5a Add an extra definition for cfg80211_connect_bss() usage
a842611 Merge pull request #1092 from paralin/fix-kernel-4.9
d00db12 ioctl_cfg80211: fix build against kernels < 4.10.x
9dc4a4c Merge pull request #1091 from CGarces/patch-1
3bcbfe4 Fix compilation error on kernels < 5.15
60222a5 Merge pull request #1089 from CGarces/fix_1088
09ba98c Use eth_hw_addr_set instead of dev_addr_set
6f7a4e5 Address warning in createbss_hdl()
ca797e1 Revert "Some cleanup of unused chips"
dc8dcb4 Merge pull request #1072 from DrSchottky/v5.6.4.2
b3227c7 Some cleanup of unused chips
5ae69b6 Fixed job name in raspberry action
35308f4 Code optimization
357faf7 Minor fixes and cleanup
6282169 fix SUBARCH on riscv64
8cb23d8 Remove dead code associated with CONFIG_PLATFORM_INTEL_BYT
0c3d4b3 Remove useless warning
d4ad212 get_ra() exists only on PowerPC
1e2ca33 Fix clang warning
9600ea1 Update dkms.conf
96b013c Possible fix for iwd crash
ccda185 Merge pull request #1069 from gglluukk/v5.6.4.2
555241e Merge pull request #1068 from asukiaaa/patch-1
1b0472d Parallelize RPi driver builds
4a0ceab Pragma GCC < 11 warning fix for -Wstringop-overread
7d8f129 Add 056E:4010 for elecom device
fe71d83 Update GH actions workflow
9d5c4aa Merge pull request #1065 from gglluukk/v5.6.4.2
a429fe1 Revert "Update GH actions workflow"
f6184e3 Update GH actions workflow
91db7d2 Merge pull request #1064 from DrSchottky/v5.6.4.2
42a0c86 Adopt patch for 5.x kernel warning dump
0445b2b Fix compiler (rtw_br_ext.c:969:11) and kernel (net/wireless/sme.c:843) warnings
0417a2c Fix RPi build github action
eee82ac Merge pull request #1056 from 154pinkchairs/fix-bashism
2859894 fix(Makefile): remove `[[]]` bashism
20f240f Merge pull request #1050 from 154pinkchairs/akms-hooks-compat
e9699df Merge pull request #1053 from astsam2/v5.6.4.2
47edc1b Add kernel v6.3 support
[... 1214 lines cut ...]
a88ff7e Fix channel switch in the monitor mode if both CONFIG_WIRELESS_EXT and CONFIG_CFG80211_WEXT are enabled
a4265be Enable TX queue in the monitor mode
9c17ea0 Radiotap iterator from kernel 4.9 was added due to VHT field bug in kernels before 4.6
6eb40af More USB VID/PID
428c217 Do not place signal strength into radiotap header if this information is not available
f80e7b7 Place signal quality and per antenna rssi into radiotap header
97059f4 Disable power saving mode
d9c3a57 Clear NO_IR flag for all channels
16a1de0 Fix error switching to 80MHz channels under certain conditions
7e48715 Unmask all available channels
6813182 Enable the channel 14 and 80 MHz width channels in the regdom
4f88d95 Added get channel function
4e69a27 Fix interface rename error
5119865 Added .gitignore
46ea178 Add support for kernels >= 4.8
6470206 Use in_compat_syscall() instead of is_compat_task() if available
7cd3949 Add support for kernels >= 4.7.
cf7aa86 Driver v5.1.5 for RTL8812AU (Tenda U12)
d62c22e Add driver from found on aircrack-ng site

$ git reset d8ad927 --hard
HEAD is now at d8ad927 Fix build on 6.x kernels.
```

After that follow installation instructions.

shaked8634, 2024-05-29:
Can confirm on then compiling the driver with commit #d8ad927 on 6.6.32 solves the issue.

yoyo956, 2024-05-29:
> * It was working fine before update. However after the update to kernel 6.9.1 (Arch linux) the problem started.
> * I was not able to connect to wifi from Archer T2U Nano, the device was 'unavailable' in `nmcli`.
> * I used `rfkill` to remove soft lock on the device, the device was shown as 'disconnected' in `nmcli` and tried to search for wifi networks buy no success on that.
> 
> * Later I downgrade the kernel to version 6.6.3-2(lts) after removing the modules in version 6.9.1.
> * After reinstalling the modules via dkms for lts I had similar problem,
> this time device was 'disconnected' instead of 'unmanaged', another reboot I tested the wifi same status 'disconnect'.
> 
> * After I installed Debian (Kernel 6.1.0) and tested again same error as the long-term kernel 6.6.3 as in Arch linux
> 
> #### status of NetworkManager in Debian
> ```
> ● NetworkManager.service - Network Manager
> Loaded: loaded (/lib/systemd/system/NetworkManager.service; enabled; preset: enabled)
> Active: active (running) since Sat 2024-05-25 20:20:54 IST; 6min ago
> Docs: man:NetworkManager(8)
> Main PID: 850 (NetworkManager)
> Tasks: 3 (limit: 33335)
> Memory: 18.4M
> CPU: 153ms
> CGroup: /system.slice/NetworkManager.service
> └─850 /usr/sbin/NetworkManager --no-daemon
> 
> May 25 20:24:24 myhost NetworkManager[850]: <info>  [1716648864.3519] device (enx5ed7df4dee2e): state change: ip-config -> ip-check (reason 'none', sys-iface-state: 'managed')
> May 25 20:24:24 myhost NetworkManager[850]: <info>  [1716648864.3967] device (enx5ed7df4dee2e): state change: ip-check -> secondaries (reason 'none', sys-iface-state: 'managed')
> May 25 20:24:24 myhost NetworkManager[850]: <info>  [1716648864.3968] device (enx5ed7df4dee2e): state change: secondaries -> activated (reason 'none', sys-iface-state: 'managed')
> May 25 20:24:24 myhost NetworkManager[850]: <info>  [1716648864.3970] manager: NetworkManager state is now CONNECTED_SITE
> May 25 20:24:24 myhost NetworkManager[850]: <info>  [1716648864.3972] device (enx5ed7df4dee2e): Activation: successful, device activated.
> May 25 20:24:24 myhost NetworkManager[850]: <info>  [1716648864.3974] manager: NetworkManager state is now CONNECTED_GLOBAL
> May 25 20:26:06 myhost NetworkManager[850]: <warn>  [1716648966.7345] platform-linux: do-change-link[3]: failure 1 (Operation not permitted)
> May 25 20:26:06 myhost NetworkManager[850]: <warn>  [1716648966.7345] device (wlx3460f9103cfd): set-hw-addr: failedto set MAC address to xx:yy:zz:xx:yy:zz (scanning) (NME_UNSPEC)
> May 25 20:26:06 myhost NetworkManager[850]: <info>  [1716648966.8424] device (wlx3460f9103cfd): supplicant interface state: disconnected -> inactive
> May 25 20:26:06 myhost NetworkManager[850]: <info>  [1716648966.8425] device (p2p-dev-wlx3460f9103cfd): supplicant management interface state: disconnected -> inactive
> ```
> 
> ### Kernel (6.1.0) dmesg
> ```
> [    8.725309] ------------[ cut here ]------------
> [    8.725311] WARNING: CPU: 1 PID: 463 at /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_odm.c:62 rtw_odm_init_ic_type+0x40/0x50 [88XXau]
> [    8.725341] Modules linked in: irqbypass bluetooth(+) 88XXau(OE+) ghash_clmulni_intel sha256_ssse3 sha1_ssse3 snd_hda_codec_realtek jitterentropy_rng snd_hda_codec_generic snd_hda_codec_hdmi sha512_ssse3 uvcvideo(+) snd_usb_audio(+) sha512_generic snd_hda_intel videobuf2_vmalloc videobuf2_memops snd_intel_dspcfg videobuf2_v4l2 snd_intel_sdw_acpi videobuf2_common aesni_intel snd_hda_codec snd_usbmidi_lib snd_rawmidi crypto_simd videodev cryptd ctr snd_hda_core snd_seq_device cfg80211 snd_hwdep mc eeepc_wmi drbg asus_wmi snd_pcm rapl battery platform_profile ansi_cprng sparse_keymap sp5100_tco snd_timer ledtrig_audio ecdh_generic pcspkr wmi_bmof k10temp watchdog snd ccp rfkill soundcore ecc joydev acpi_cpufreq sg evdev msr parport_pc ppdev lp parport fuse loop dm_mod efi_pstore configfs efivarfs ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic hid_generic usbhid hid amdgpu gpu_sched drm_buddy i2c_algo_bit drm_display_helper cec sd_mod rc_core drm_ttm_helper ttm
> [    8.725418]  drm_kms_helper ahci xhci_pci nvme libahci xhci_hcd nvme_core libata r8169 t10_pi drm realtek crc32_pclmul mdio_devres usbcore scsi_mod crc32c_intel crc64_rocksoft libphy crc64 crc_t10dif i2c_piix4 crct10dif_genericscsi_common crct10dif_pclmul usb_common video crct10dif_common wmi button
> [    8.725445] CPU: 1 PID: 463 Comm: (udev-worker) Tainted: G           OE      6.1.0-21-amd64 #1  Debian 6.1.90-1
> [    8.725449] Hardware name: System manufacturer System Product Name/TUF GAMING X570-PLUS, BIOS 4403 04/28/2022
> [    8.725450] RIP: 0010:rtw_odm_init_ic_type+0x40/0x50 [88XXau]
> [    8.725487] Code: 3e 00 00 80 fa 0e 76 17 8b 15 f8 3d 0a 00 85 d2 74 18 be 04 00 00 00 48 89 c7 e9 0b 08 07 00 8b 14 95 90 7a b2 c1 85 d2 75 e8 <0f> 0b be 04 00 00 00 48 89 c7 e9 f1 07 07 00 90 0f 1f 44 00 00 48
> [    8.725489] RSP: 0018:ffffb98680a2ba70 EFLAGS: 00010246
> [    8.725492] RAX: ffffb9868135eed0 RBX: ffffb98680757000 RCX: 0000000000000002
> [    8.725494] RDX: 0000000000000000 RSI: ffffffffc1b379cd RDI: ffffb98680757000
> [    8.725495] RBP: ffff9e36696b7000 R08: ffffffffc1b379cc R09: 0000000000000000
> [    8.725497] R10: ffffffffffffffff R11: ffffb98680a2b9c9 R12: ffff9e3669769800
> [    8.725498] R13: ffffffffc1b8a440 R14: ffff9e3669769800 R15: ffff9e36de79b000
> [    8.725500] FS:  00007f66bae928c0(0000) GS:ffff9e3c51c40000(0000) knlGS:0000000000000000
[... cut ...]
> [   32.662087] R10: 0000000000000000 R11: 0000000000000001 R12: 0000000000000000
> [   32.662087] R13: ffffb98681defd68 R14: 000000000000001e R15: ffff9e365308cf40
> [   32.662087] FS:  0000000000000000(0000) GS:ffff9e3c51c00000(0000) knlGS:0000000000000000
> [   32.662087] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
> [   32.662087] CR2: ffffffffffffffd6 CR3: 0000000111df4000 CR4: 0000000000750ef0
> [   32.662087] PKRU: 55555554
> [   32.662087] note: RTW_CMD_THREAD[961] exited with irqs disabled
> [  216.866391] usb 3-2: USB disconnect, device number 3
> ```

Scott31393, 2024-05-31:
Can confirm too! Thanks @koku17 
```
git clone https://github.com/aircrack-ng/rtl8812au.git
cd rtl8812au
git reset d8ad927 --hard
```
Then build:
```
sudo make install
```
I test that using the following dongle into imx8mm board (arm64)
```
Bus 001 Device 003: ID 2357:011e TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano]
```