aircrack-ng/rtl8812au #1157: Adapter is available but not working

sec0ps, 2024-05-22:
This was working, I wiped the laptop and reinstalled now I cannot get this working again. The adapter is visible, but not able to use it at all.

$ uname -av
Linux sec0ps 6.5.0-35-generic #35~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue May  7 09:00:52 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.4 LTS
Release:        22.04
Codename:       jammy

$ iwconfig wlx00c0cab48cfb
wlx00c0cab48cfb  IEEE 802.11b  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Monitor  Frequency:2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

$ sudo dkms status
8812au/5.6.4.2_35491.20191025, 6.5.0-35-generic, x86_64: installed

Is this a known issue, any idea on how to fix this?

victor4pinheiro, 2024-05-24:
I'm having this issue also on Linux arch 6.9.1-zen1-2-zen #1 ZEN SMP PREEMPT_DYNAMIC Wed, 22 May 2024 13:47:12 +0000 x86_64 GNU/Linux.

darash, 2024-05-25:
Me too..

uname -a

> Linux 6.9.1-zen1-2-zen #1 ZEN SMP PREEMPT_DYNAMIC Wed, 22 May 2024 13:47:12 +0000 x86_64 GNU/Linux

sudo dkms status
> rtl88xxau/r1298.b44d288, 6.9.1-arch1-2, x86_64: installed

darash, 2024-05-27:
hostnamectl

> Operating System: Arch Linux                        
>           Kernel: Linux 6.9.2-zen1-1-zen
>     Architecture: x86-64

dkms status 

> rtl88xxau/r1298.b44d288, 6.9.2-arch1-1, x86_64: installed

journalctl -b -p 4 --no-pager

> kernel: 88XXau: loading out-of-tree module taints kernel.
> kernel: ------------[ cut here ]------------
> kernel: Unpatched return thunk in use. This should not happen!
> kernel: WARNING: CPU: 1 PID: 367 at arch/x86/kernel/cpu/bugs.c:3023 __warn_thunk+0x2a/0x40
> kernel: Modules linked in: 88XXau(OE+) cfg80211 rfkill dm_mod loop nfnetlink ip_tables x_tables tcp_bbr sg vfat fat ext4 hid_generic crc32c_generic crc16 uas mbcache usb_storage usbhid jbd2 serio_raw amdgpu atkbd r8169 libps2 vivaldi_fmap nvme realtek msi_ec mdio_devres nvme_core crc32c_intel xhci_pci libphy nvme_auth xhci_pci_renesas i8042 serio msi_wmi sparse_keymap video wmi amdxcp i2c_algo_bit drm_ttm_helper ttm drm_exec gpu_sched drm_suballoc_helper drm_buddy drm_display_helper cec
> kernel: CPU: 1 PID: 367 Comm: systemd-modules Tainted: G           OE      6.9.2-zen1-1-zen #1 14404c7ad8f24e086d7631b72c6d01926c31bde3
> kernel: Hardware name: Micro-Star International Co.
> kernel: RIP: 0010:__warn_thunk+0x2a/0x40
> kernel: Code: 66 0f 1f 00 0f 1f 44 00 00 80 3d d0 4d 37 02 00 74 05 e9 69 35 03 01 48 c7 c7 10 cc 44 b7 c6 05 bb 4d 37 02 01 e8 c6 92 06 00 <0f> 0b e9 4f 35 03 01 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00
> kernel: RSP: 0018:ffffae8281ca7ab8 EFLAGS: 00010282
> kernel: RAX: 0000000000000000 RBX: 00000000ffffffff RCX: 0000000000000027
> kernel: RDX: ffff8d4b9f2a19c8 RSI: 0000000000000001 RDI: ffff8d4b9f2a19c0
> kernel: RBP: ffffae8281ca7b08 R08: ffffffffb7cb2648 R09: 00000000ffffefff
> kernel: R10: 0000000000000003 R11: 0000000000000002 R12: 0000000000000000
> kernel: R13: ffffae8281ca7b28 R14: ffff8d48d016a300 R15: ffff8d48cb128f20
> kernel: FS:  00007d3486cfb680(0000) GS:ffff8d4b9f280000(0000) knlGS:0000000000000000
> kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
> kernel: CR2: 00006421a147c6e0 CR3: 0000000105e48000 CR4: 0000000000350ef0
> kernel: Call Trace:
> kernel:  <TASK>
> kernel:  ? __warn_thunk+0x2a/0x40
> kernel:  ? __warn.cold+0x8e/0xf3
> kernel:  ? __warn_thunk+0x2a/0x40
> kernel:  ? report_bug+0xe7/0x200
> kernel:  ? srso_return_thunk+0x5/0x5f
> kernel:  ? handle_bug+0x3c/0x80
> kernel:  ? exc_invalid_op+0x19/0xc0
> kernel:  ? asm_exc_invalid_op+0x1a/0x20
> kernel:  ? __warn_thunk+0x2a/0x40
> kernel:  ? __warn_thunk+0x2a/0x40
> kernel:  warn_thunk_thunk+0x1a/0x30
> kernel:  ? __UNIQUE_ID_vermagic434+0x517d3afd3954/0x517d3afd3954 [88XXau 6fcb75a1792fe41d2130d50380c86b9d3b4e8061]
> kernel:  init_module+0x14/0xff0 [88XXau 6fcb75a1792fe41d2130d50380c86b9d3b4e8061]
> kernel:  ? __UNIQUE_ID_vermagic434+0x517d3afd3954/0x517d3afd3954 [88XXau 6fcb75a1792fe41d2130d50380c86b9d3b4e8061]
> kernel:  ? do_one_initcall+0x5b/0x310
> kernel:  ? do_init_module+0x60/0x220
> kernel:  ? idempotent_init_module+0x182/0x340
> kernel:  ? __x64_sys_finit_module+0x5e/0xb0
> kernel:  ? do_syscall_64+0x83/0x190
> kernel:  ? srso_return_thunk+0x5/0x5f
> kernel:  ? srso_return_thunk+0x5/0x5f
> kernel:  ? syscall_exit_to_user_mode+0x75/0x1f0
> kernel:  ? srso_return_thunk+0x5/0x5f
> kernel:  ? do_syscall_64+0x8f/0x190
> kernel:  ? srso_return_thunk+0x5/0x5f
> kernel:  ? syscall_exit_to_user_mode+0x75/0x1f0
> kernel:  ? srso_return_thunk+0x5/0x5f
> kernel:  ? do_syscall_64+0x8f/0x190
> kernel:  ? __x64_sys_openat+0x1f5/0x230
> kernel:  ? srso_return_thunk+0x5/0x5f
> kernel:  ? syscall_exit_to_user_mode+0x75/0x1f0
> kernel:  ? srso_return_thunk+0x5/0x5f
> kernel:  ? do_syscall_64+0x8f/0x190
> kernel:  ? srso_return_thunk+0x5/0x5f
> kernel:  ? srso_return_thunk+0x5/0x5f
> kernel:  ? entry_SYSCALL_64_after_hwframe+0x76/0x7e
> kernel:  </TASK>
> kernel: ---[ end trace 0000000000000000 ]---

sagypt, 2024-05-31:
Same for me on Debian 12

`Linux redacted 6.1.0-21-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.90-1 (2024-05-03) x86_64 GNU/Linux`

The driver installs but then the NIC does not detect any WIFI network.

zartaz, 2024-06-16:
i think is incompatibiliti issue of the driver with some newer kernels , on my kali with 6.8.11-amd64 everything went fine with installation but airodump doesnt' show any ssid after my kali clean install

alb3rt-on, 2024-06-18:
I had the same issue for days and finally solve it. I followed a lot of youtube videos and online forums and installed both Kali and aircrack-ng drivers but still wasn't working. I even ended up returning my old Alfa wifi adapter and getting a new Alfa AWUS036ACH adapter and it still didn't work after set up.

**Issue:**
Wifi adapter shows up after driver install but not showing available networks (Available networks also may be disabled)

**My system info:**
VM: VirtualBox v7
Linux OS: Kali Purple 2024
Kernel version: 6.8.11 

The issue is not the drivers or the updated kernel but instead the Debian version in Virtualbox (I'm sure it'll be the same issue for VMWare or other VM's)

**To fix this issue:**

Open Virtualbox 
Select your **VM** (in my case Kali Purple)
Go to **Settings**
In **General** > make sure **Type** is **Linux** 
Change **Version** from Debian 12 (or current version) to **Debian 10**
Start your VM and it.
It should fix the wifi adapter issue now and should be able to see wifi available wifi networks.

Hope this helps. Let me know if it works. If it does please share it and pass it on because I read a lot of forums with same issues but I'm not on any of them.

2smooth4u, 2024-06-22:
> I had the same issue for days and finally solve it. I followed a lot of youtube videos and online forums and installed both Kali and aircrack-ng drivers but still wasn't working. I even ended up returning my old Alfa wifi adapter and getting a new Alfa AWUS036ACH adapter and it still didn't work after set up.
> 
> **Issue:** Wifi adapter shows up after driver install but not showing available networks (Available networks also may be disabled)
> 
> **My system info:** VM: VirtualBox v7 Linux OS: Kali Purple 2024 Kernel version: 6.8.11
> 
> The issue is not the drivers or the updated kernel but instead the Debian version in Virtualbox (I'm sure it'll be the same issue for VMWare or other VM's)
> 
> **To fix this issue:**
> 
> Open Virtualbox Select your **VM** (in my case Kali Purple) Go to **Settings** In **General** > make sure **Type** is **Linux** Change **Version** from Debian 12 (or current version) to **Debian 10** **Start** the **VM** It should fix the wifi adapter issue now and should be able to see wifi available wifi networks.
> 
> Hope this helps. Let me know if it works.

For Vmware this is not working

amnuhs, 2024-06-27:
Same problem here!
My system info:
VM: Vmware workstation pro 17.5.2 build-23775571
Linux OS: Kali 2024
Kernel version: 6.8.11

asasasas88, 2024-06-29:
Same problem:

ALFA AWUS036ACH
Kali: 2024.2
Kernel: 6.8.11-amd64
VMWare Fusion 13.5.1

# dkms status
realtek-rtl8814au/5.8.5.1~git20240527.d8208c8, 6.8.11-amd64, x86_64: installed
realtek-rtl88xxau/5.6.4.2~git20240510.b44d288, 6.8.11-amd64, x86_64: installed

#lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter

CodeTrainerMan, 2024-07-11:
same problem：
realtek-rtl88xxau/5.6.4.2~20240509: added
rtl8812au/4.3.8.12175.20140902+dfsg, 5.15.0-116-generic, x86_64: installed

wlx08107bb3f9cd  unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

[image]
cannt find any network

R0VAD, 2024-07-12:
Same problem, tried on dedicated laptop, VMware & MacBook Pro using UTM.

ka3boosh, 2024-07-18:
here is the fix
https://www.youtube.com/watch?v=Yr-8RmoNi70&t=40s

raenk, 2024-07-25:
Add me to the list. Not working on EndeavourOS (Arch). This machine dual boots with Fedora and it's working over there.
TP-Link Archer T2U PLUS

Device does not detect any connections and adding a connection manually is useless.

ChayimEliazer, 2024-07-29:
> Same for me on Debian 12 The driver installs but then the NIC does not detect any WIFI network.
> 
> ```
> uname -a
> 
> Linux <REDACTED> 6.1.0-21-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.90-1 (2024-05-03) x86_64 GNU/Linux
> 
> 
> lsusb
> 
> Bus 001 Device 016: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
> 
> 
> sudo dkms status
> 
> 8812au/5.6.4.2_35491.20191025, 6.1.0-21-amd64, x86_64: installed
> 
> 
> sudo dmesg -W
> 
> [ 7606.269729] usb 1-10: new high-speed USB device number 17 using xhci_hcd
> [ 7606.418693] usb 1-10: New USB device found, idVendor=0bda, idProduct=8812, bcdDevice= 0.00
> [ 7606.418707] usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [ 7606.418713] usb 1-10: Product: 802.11n NIC
> [ 7606.418718] usb 1-10: Manufacturer: Realtek
> [ 7606.418721] usb 1-10: SerialNumber: 123456
> [ 7606.526073] usb 1-10: 88XXau <REDACTED> hw_info[d7]
> [ 7606.947245] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
> 
> 
> sudo lshw -c network
> 
>   *-network
>        description: Wireless interface
>        physical id: c
>        bus info: usb@1:10
>        logical name: wlan1
>        serial: <REDACTED>
>        capabilities: ethernet physical wireless
>        configuration: broadcast=yes driver=rtl88XXau driverversion=6.1.0-21-amd64 multicast=yes wireless=unassociated
> 
> 
> sudo rfkill list all
> 
> 7: phy6: Wireless LAN
> 	Soft blocked: no
> 	Hard blocked: no
> 
> 
> ip l
> 
> 7: wlan1: <NO-CARRIER,BROADCAST,MULTICAST,UP,LOWER_UP> mtu 2312 qdisc mq state DORMANT mode DORMANT group default qlen 1000
>     link/ether <REDACTED> brd ff:ff:ff:ff:ff:ff permaddr <REDACTED>
> 
> 
> nmcli device wifi list ifname wlan1
> 
> IN-USE  BSSID  SSID  MODE  CHAN  RATE  SIGNAL  BARS  SECURITY
> 
> (nothing)
> ```
> 
> [EDIT]
> 
> From [USB Wifi TP-Link AC600 cannot work with newest version](https://github.com/aircrack-ng/rtl8812au/issues/1159)
> 
> These instructions work for me
> 
> ```
> cd /usr/src
> sudo git clone https://github.com/aircrack-ng/rtl8812au.git
> cd /usr/src/rtl8812au
> sudo git checkout 63cf0b4
> sudo make dkms_install
> ```

Solved it for me

chiefMarlin, 2024-09-06:
Same issue on latest kernel in kali and debian 12, mentioned solution works! 👍
```
cd /usr/src
sudo git clone https://github.com/aircrack-ng/rtl8812au.git
cd /usr/src/rtl8812au
sudo git checkout 63cf0b4
sudo make dkms_install
```