aircrack-ng/rtl8812au #1159: USB Wifi TP-Link AC600 cannot work with newest version

kienngd, 2024-06-01:
I use **Ubuntu 24.04**
Kernel: **6.8.0-31**
Wifi: TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano] ID **2357:011e**
I install by command **sudo make dkms_install** with newest version, wifi will disconnect after init. So I cannot list wifi list.
But if I **reset code** to commit **63cf0b4584aa8878b0fe8ab38017f31c319bde3d**. Wifi works fine.

zCreeperYT, 2024-06-01:
Same for me!
THANK YOU VERY MUCH
Greetings from Germany <3
But the problem is really existing so please fix it!

cblknittights, 2024-06-02:
Same here, just got this device to replace an internal card that is not working with Kubuntu 24.04

Get a green blinking LED, no wireless networks listed in the network manager

I will gladly provide anything more:

I tried the repo rtl8812au via synpatic and refuses to build has all kinds of errors..... purged that out and did this ... and no joy...

Details:
uname:
Linux X 6.8.0-31-generic #31-Ubuntu SMP PREEMPT_DYNAMIC Sat Apr 20 00:40:06 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

lsusb
Bus 007 Device 002: ID 2357:011e TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano]

iwconfig:

wlxf0a731dc05c0  unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

dmesg output below:
```
[ 1767.784649] usb 2-3: new high-speed USB device number 4 using ehci-pci
[ 1767.917670] usb 2-3: New USB device found, idVendor=2357, idProduct=011e, bcdDevice= 2.00
[ 1767.917682] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1767.917687] usb 2-3: Product: 802.11ac WLAN Adapter 
[ 1767.917691] usb 2-3: Manufacturer: Realtek 
[ 1767.917695] usb 2-3: SerialNumber: 00e04c000001
[ 1768.614733] ------------[ cut here ]------------
[ 1768.614742] WARNING: CPU: 1 PID: 9988 at /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_odm.c:62 rtw_odm_init_ic_type+0x52/0x70 [88XXau]
[ 1768.615073] Modules linked in: 88XXau(OE+) snd_seq_dummy snd_hrtimer xt_CHECKSUM xt_MASQUERADE xt_conntrack ipt_REJECT nf_reject_ipv4 xt_tcpudp nft_compat nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables bridge stp llc qrtr binfmt_misc snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic snd_hda_intel snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec wl(POE) snd_hda_core uvcvideo snd_hwdep i915 videobuf2_vmalloc uvc videobuf2_memops snd_pcm videobuf2_v4l2 snd_seq_midi videodev snd_seq_midi_event snd_rawmidi videobuf2_common drm_buddy snd_seq ttm mc drm_display_helper snd_seq_device snd_timer dell_laptop cec dell_wmi cfg80211 dell_smbios snd rc_core dcdbas ledtrig_audio sparse_keymap dell_smm_hwmon dell_wmi_descriptor i2c_algo_bit wmi_bmof i2c_i801 coretemp i2c_smbus soundcore lpc_ich joydev input_leds mac_hid serio_raw msr parport_pc ppdev lp parport efi_pstore nfnetlink dmi_sysfs ip_tables x_tables autofs4 btrfs blake2b_generic xor raid6_pq libcrc32c dm_crypt dm_mirror
[ 1768.615248]  dm_region_hash dm_log sha256_ssse3 sha1_ssse3 r8169 ahci psmouse libahci realtek ssb video wmi ums_realtek uas usb_storage crypto_simd cryptd
[ 1768.615285] CPU: 1 PID: 9988 Comm: (udev-worker) Tainted: P           OE      6.8.0-31-generic #31-Ubuntu
[ 1768.615291] Hardware name: Dell Inc. Inspiron 1470/0M7CYJ, BIOS A03 10/19/2009
[ 1768.615295] RIP: 0010:rtw_odm_init_ic_type+0x52/0x70 [88XXau]
[ 1768.615624] Code: 85 d2 74 26 be 04 00 00 00 48 89 c7 e8 97 2c 09 00 5d 31 c0 31 d2 31 f6 31 ff c3 cc cc cc cc 8b 14 95 f0 4a c1 c1 85 d2 75 da <0f> 0b be 04 00 00 00 48 89 c7 e8 6f 2c 09 00 5d 31 c0 31 d2 31 f6
[ 1768.615631] RSP: 0018:ffffba49c5f078c0 EFLAGS: 00010246
[ 1768.615637] RAX: ffffba49c086ced0 RBX: ffffba49c005b000 RCX: 0000000000000000
[ 1768.615642] RDX: 0000000000000000 RSI: 0000000000000000 RDI: ffffba49c005b000
[ 1768.615645] RBP: ffffba49c5f078c0 R08: 0000000000000000 R09: 0000000000000000
[ 1768.615649] R10: 0000000000000000 R11: 0000000000000000 R12: ffff9b8886548000
[ 1768.615652] R13: ffff9b88a45c8c00 R14: 0000000000000000 R15: ffffffffc1ae0a40
[ 1768.615656] FS:  000070f68a27b8c0(0000) GS:ffff9b88bbf00000(0000) knlGS:0000000000000000
[ 1768.615661] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1768.615665] CR2: 0000784c00000020 CR3: 000000010b8ca000 CR4: 00000000000406f0
[ 1768.615670] Call Trace:
[ 1768.615675]  <TASK>
[ 1768.615681]  ? show_regs+0x6d/0x80
[ 1768.615692]  ? __warn+0x89/0x160
[ 1768.615701]  ? rtw_odm_init_ic_type+0x52/0x70 [88XXau]
[ 1768.615982]  ? report_bug+0x17e/0x1b0
[ 1768.615992]  ? handle_bug+0x51/0xa0
[ 1768.615998]  ? exc_invalid_op+0x18/0x80
[ 1768.616003]  ? asm_exc_invalid_op+0x1b/0x20
[ 1768.616011]  ? rtw_odm_init_ic_type+0x52/0x70 [88XXau]
[ 1768.616285]  rtw_hal_read_chip_version+0x21/0x30 [88XXau]
[ 1768.616594]  rtw_usb_primary_adapter_init+0x128/0xa50 [88XXau]
[ 1768.616880]  rtw_usb_primary_adapter_init+0x41c/0xa50 [88XXau]
[ 1768.617164]  ? pm_runtime_enable+0x51/0xd0
[ 1768.617171]  ? __pm_runtime_set_status+0x1b7/0x360
[ 1768.617179]  usb_probe_interface+0xee/0x300
[ 1768.617188]  really_probe+0x1c7/0x410
[ 1768.617195]  __driver_probe_device+0x8c/0x180
[ 1768.617201]  driver_probe_device+0x24/0xd0
[... 153 lines cut ...]
[ 1791.904041]  ? __pfx_rtw_cmd_thread+0x10/0x10 [88XXau]
[ 1791.904041]  kthread+0xf2/0x120
[ 1791.904041]  ? __pfx_kthread+0x10/0x10
[ 1791.904041]  ret_from_fork+0x47/0x70
[ 1791.904041]  ? __pfx_kthread+0x10/0x10
[ 1791.904041]  ret_from_fork_asm+0x1b/0x30
[ 1791.904041]  </TASK>
[ 1791.904041] Modules linked in: 88XXau(OE) snd_seq_dummy snd_hrtimer xt_CHECKSUM xt_MASQUERADE xt_conntrack ipt_REJECT nf_reject_ipv4 xt_tcpudp nft_compat nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables bridge stp llc qrtr binfmt_misc snd_hda_codec_ [... cut ...]
[ 1791.904041] RBP: ffffba49c08ffcc0 R08: ffffba49c08ffc10 R09: 0000000000000000
[ 1791.904041] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000000000
[ 1791.904041] R13: 000000000000001d R14: 000000000000001d R15: ffffba49c08ffc48
[ 1791.904041] FS:  0000000000000000(0000) GS:ffff9b88bbf00000(0000) knlGS:0000000000000000
[ 1791.904041] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1791.904041] CR2: ffffffffffffffd6 CR3: 000000009783c000 CR4: 00000000000406f0
[ 1791.904041] note: RTW_CMD_THREAD[10013] exited with irqs disabled

```

cblknittights, 2024-06-02:
> But if I **reset code** to commit **[63cf0b4](https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d)**. Wifi works fine.

How did you "reset" to this, I tried git reset 63cf0b4 and compiled, and no joy... still gives the same issues I posted.

kienngd, 2024-06-02:
> How did you "reset" to this, I tried git reset [63cf0b4](https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d) and compiled, and no joy... still gives the same [...]

I use command `git reset` to reset code. Then `git status` to show changed files. I remove all new files and checkout all changed files. Finally, I use command `sudo make dkms_install`. It works fine for me.

rajatkashyap47586, 2024-06-02:
Hi every1
I also facing the same issue but in my case the problem is I cloned this repo yesterday and get the same error. I cannot simply reset the repo as it contains only the latest clone.

Any solution on how get the working version ?

Thanks in advance.

dubhater, 2024-06-02:
``git checkout 63cf0b4`` should do it.

rajatkashyap47586, 2024-06-02:
> `git checkout 63cf0b4` should do it.

post this do I need to install rtl8812au from the repo again like sudo make dkms_install ?
Or only checkout is enough ?

dubhater, 2024-06-02:
Oh, you're using dkms. Yes, you probably need to install again.

rajatkashyap47586, 2024-06-02:
> Oh, you're using dkms. Yes, you probably need to install again.

I tried to do installation again but I'm gettting error 
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025
Error! DKMS tree already contains: 8812au-5.6.4.2_35491.20191025
You cannot add the same module/version combo more than once.
make: *** [Makefile:1789: dkms_install] Error 3

It looks like it is trying to install the same version even though in repo the git head is on  63cf0b4

Please let me know if I'm doing something wrong here.

dubhater, 2024-06-02:
Uninstall it then install again.

rajatkashyap47586, 2024-06-02:
> Uninstall it then install again.

I tried but getting same error 

usbcore: registered new interface driver rtl88XXau
[   58.325787] BUG: kernel NULL pointer dereference, address: 0000000000000000
[   58.325793] #PF: supervisor read access in kernel mode
[   58.325795] #PF: error_code(0x0000) - not-present page

cblknittights, 2024-06-02:
> I use command `git reset` to reset code. Then `git status` to show changed files. I remove all new files and checkout all changed files. Finally, I use command `sudo make dkms_install`. It works [...]

I am **not a git guy**.. so can you please detail the steps...

1) Pull git repo

git clone http...... 

2) reset to this commit

git reset 63cf0b4

3) ????????

4) ???????

5) ?????? 

6) sudo make dkms_install

As above, I am not git person, so more details are needed, please. **_Thank you._**

cblknittights, 2024-06-02:
Ok... based on some other replies this WORKS...

I tested in a LIVE USB setup and the device now works, I will wipe and reinstall... I think things were buggered up enough from playing with trying to get the internal card working to do this any way...

so for other NON git people like me:

git clone...... (from the readme)

git checkout 63cf0b4

sudo make dkms_install

Plug in/or replug device.. and I clicked on the network manager and it showed wireless networks!

Only glitch is the ONE 5Ghz network that I would prefer to use is not shown????? Weird.... 

But I an use the other AP for this... 

BUT WORKS!

Whereas the internal broadcom one wont. $13 well spent.

Hope this post helps others... and the debug info gets the bug swatted.... out.

rajatkashyap47586, 2024-06-02:
> 6\. sudo make dkms_install

what I did :-

step1) Go to the cloned git repo  -> cd rtl8812au
step2) Uninstall the package -> make dkms_remove
step3) Removed the entire package -> rm -rf rtl8812au
step4) Clone again the git repo -> git clone https://github.com/aircrack-ng/rtl8812au.git

step5) Go to the cloned git repo -> cd rtl8812au
step6) Checkout to the stable version -> git checkout 63cf0b4 
step7) Install again -> make dkms_install
step8) reboot -> reboot
step9) tested again -> selecting wifi adaptor from usb in virtual box post this I can wifi in kali and then I tried to connect to any network then I can only see loading bar, if in meanwhile I used "dmesg" then I can see the error :-

[   87.433972] BUG: kernel NULL pointer dereference, address: 0000000000000000
[   87.433993] #PF: supervisor read access in kernel mode
[   87.433997] #PF: error_code(0x0000) - not-present page

I'm not sure this error is corresponding to the wifi or not.

I have whole dmesg output if you want I can share that.

rajatkashyap47586, 2024-06-02:
> Ok... based on some other replies this WORKS...
>
> I tested in a LIVE USB setup and the device now works, I will wipe and reinstall... I think things were buggered up enough from playing with trying [...]

I tried the same but no luck yet. Waiting for some help on this...

rajatkashyap47586, 2024-06-02:
dmesg output I see when wifi is trying to connect to any network

[    0.000000] Linux version 6.6.15-amd64 (devel@kali.org) (gcc-13 (Debian 13.2.0-24) 13.2.0, GNU ld (GNU Binutils for Debian) 2.42) #1 SMP PREEMPT_DYNAMIC Kali 6.6.15-2kali1 (2024-05-17)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.6.15-amd64 root=UUID=d797ba13-5681-47ea-aae3-fb372a64bef8 ro quiet splash
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000dffeffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dfff0000-0x00000000dfffffff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000001d9afffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] APIC: Static calls initialized
[    0.000000] SMBIOS 2.5 present.
[    0.000000] DMI: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000003] kvm-clock: using sched offset of 843425196149 cycles
[    0.000006] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000009] tsc: Detected 2688.010 MHz processor
[    0.002455] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.002460] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.002468] last_pfn = 0x1d9b00 max_arch_pfn = 0x400000000
[    0.002494] MTRRs disabled by BIOS
[    0.002497] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.002511] last_pfn = 0xdfff0 max_arch_pfn = 0x400000000
[    0.002592] found SMP MP-table at [mem 0x0009fff0-0x0009ffff]
[    0.002615] Incomplete global flushes, disabling PCID
[    0.002879] RAMDISK: [mem 0x2e9ed000-0x334edfff]
[    0.002886] ACPI: Early table checksum verification disabled
[    0.002890] ACPI: RSDP 0x00000000000E0000 000024 (v02 VBOX  )
[    0.002896] ACPI: XSDT 0x00000000DFFF0030 00003C (v01 VBOX   VBOXXSDT 00000001 ASL  00000061)
[    0.002903] ACPI: FACP 0x00000000DFFF00F0 0000F4 (v04 VBOX   VBOXFACP 00000001 ASL  00000061)
[    0.002910] ACPI: DSDT 0x00000000DFFF0640 002353 (v02 VBOX   VBOXBIOS 00000002 INTL 20100528)
[    0.002914] ACPI: FACS 0x00000000DFFF0200 000040
[    0.002918] ACPI: FACS 0x00000000DFFF0200 000040
[    0.002922] ACPI: APIC 0x00000000DFFF0240 00008C (v02 VBOX   VBOXAPIC 00000001 ASL  00000061)
[    0.002926] ACPI: SSDT 0x00000000DFFF02D0 00036C (v01 VBOX   VBOXCPUT 00000002 INTL 20100528)
[    0.002929] ACPI: Reserving FACP table memory at [mem 0xdfff00f0-0xdfff01e3]
[    0.002930] ACPI: Reserving DSDT table memory at [mem 0xdfff0640-0xdfff2992]
[    0.002931] ACPI: Reserving FACS table memory at [mem 0xdfff0200-0xdfff023f]
[    0.002932] ACPI: Reserving FACS table memory at [mem 0xdfff0200-0xdfff023f]
[    0.002933] ACPI: Reserving APIC table memory at [mem 0xdfff0240-0xdfff02cb]
[    0.002934] ACPI: Reserving SSDT table memory at [mem 0xdfff02d0-0xdfff063b]
[    0.003985] No NUMA configuration found
[    0.003986] Faking a node at [mem 0x0000000000000000-0x00000001d9afffff]
[    0.003995] NODE_DATA(0) allocated [mem 0x1d9ad5000-0x1d9afffff]
[    0.004242] Zone ranges:
[    0.004247]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.004249]   DMA32    [mem 0x0000000001000000-0x00000000ffffffff]
[    0.004251]   Normal   [mem 0x0000000100000000-0x00000001d9afffff]
[    0.004252]   Device   empty
[    0.004253] Movable zone start for each node
[    0.004255] Early memory node ranges
[    0.004255]   node   0: [mem 0x0000000000001000-0x000000000009efff]
[    0.004257]   node   0: [mem 0x0000000000100000-0x00000000dffeffff]
[    0.004259]   node   0: [mem 0x0000000100000000-0x00000001d9afffff]
[    0.004260] Initmem setup node 0 [mem 0x0000000000001000-0x00000001d9afffff]
[    0.004266] On node 0, zone DMA: 1 pages in unavailable ranges
[    0.004304] On node 0, zone DMA: 97 pages in unavailable ranges
[    0.014038] On node 0, zone Normal: 16 pages in unavailable ranges
[    0.014495] On node 0, zone Normal: 25856 pages in unavailable ranges
[    0.016492] ACPI: PM-Timer IO Port: 0x4008
[    0.016832] IOAPIC[0]: apic_id 8, version 32, address 0xfec00000, GSI 0-23
[    0.016837] ACPI: INT_SRC_OVR (bus 0 bus_irq 0 global_irq 2 dfl dfl)
[    0.016839] ACPI: INT_SRC_OVR (bus 0 bus_irq 9 global_irq 9 low level)
[    0.016845] ACPI: Using ACPI (MADT) for SMP configuration information
[... 715 lines cut ...]
[   87.434508]  cfg80211_event_work+0x29/0x40 [cfg80211]
[   87.434551]  process_one_work+0x171/0x340
[   87.434558]  worker_thread+0x27b/0x3a0
[   87.434562]  ? __pfx_worker_thread+0x10/0x10
[   87.434565]  kthread+0xe5/0x120
[   87.434574]  ? __pfx_kthread+0x10/0x10
[   87.434578]  ret_from_fork+0x31/0x50
[   87.434585]  ? __pfx_kthread+0x10/0x10
[   87.434589]  ret_from_fork_asm+0x1b/0x30
[   87.434598]  </TASK>
[... cut ...]
[   87.434787] RBP: ffffa136c070fde0 R08: 0000000000000000 R09: 0000000000000000
[   87.434788] R10: 0000000000000002 R11: 0000000000000001 R12: ffff89ea88f97800
[   87.434789] R13: ffff89ea88f97c30 R14: 0000000000000000 R15: 0000000000000001
[   87.434791] FS:  0000000000000000(0000) GS:ffff89ea92880000(0000) knlGS:0000000000000000
[   87.434793] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   87.434795] CR2: 0000000000000000 CR3: 000000010e2d2000 CR4: 00000000000106e0
[   87.434797] note: kworker/u16:1[483] exited with irqs disabled

echo-lalia, 2024-06-02:
I was also having this issue. I originally installed the most recent version using dkms, but it didn't work and I tried a few times with no success. But I got it working. 

Ill share what I did to hopefully help others, but I'm not totally sure exactly what steps were important to fixing the issue:

- Unplugged the USB adapter
- Ran `dkms status` to find the name of the dkms module I had installed.
- Used `sudo dkms remove <module name> --all` to delete the old driver (replace module name with the name from the prev step).
- Restarted computer.
- Fully deleted the git folder (rtl8812au) so that I could re-clone it with no changes.
- Used `git clone https://github.com/aircrack-ng/rtl8812au.git` to redownload the files.
- `cd rt*`
- Then I decided to update everything just to be sure:
  - `sudo apt-get update`
  - ``sudo apt-get install bc mokutil build-essential libelf-dev linux-headers-`uname -r` ``
  - `sudo apt-get upgrade`
- Then, I did `git checkout 63cf0b4` to set it to the known working version, and a `git status` just to be sure there were no changed files.
- Ran `sudo make`, and then `sudo make install`.
- Finally, `sudo service NetworkManager restart`, and then I plugged in the USB adapter and it worked.

kienngd, 2024-06-03:
I add more detail what did I do:
- `git clone ....` to get newest source code from git.
- `cd ....` Move to the source code folder.
- `sudo make dkms_remove` to remove old install.
- `git reset 63cf0b4584aa8878b0fe8ab38017f31c319bde3d` to reset code to older commit.
- `git status` to view all modified files.
- I see:
```
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   Makefile
	modified:   android/android_ref_codes_10.x/Realtek_Wi-Fi_SDK_for_Android_10.pdf
	modified:   core/rtw_cmd.c
	modified:   core/rtw_mlme.c
	modified:   core/rtw_mlme_ext.c
	modified:   core/rtw_p2p.c
	modified:   core/rtw_rson.c
	modified:   core/rtw_wlan_util.c
	modified:   dkms.conf
	modified:   hal/hal_com.c
	modified:   hal/hal_halmac.h
	modified:   hal/hal_intf.c
	modified:   hal/phydm/halrf/rtl8814a/halrf_8814a_ap.c
	modified:   hal/phydm/rtl8812a/halhwimg8812a_bb.c
	modified:   hal/phydm/rtl8812a/halhwimg8812a_mac.c
	modified:   hal/phydm/rtl8812a/halhwimg8812a_rf.c
	modified:   hal/phydm/rtl8814a/halhwimg8814a_bb.c
	modified:   hal/phydm/rtl8814a/halhwimg8814a_mac.c
	modified:   hal/phydm/rtl8814a/halhwimg8814a_rf.c
	modified:   hal/phydm/rtl8821a/halhwimg8821a_bb.c
	modified:   hal/phydm/rtl8821a/halhwimg8821a_mac.c
	modified:   hal/phydm/rtl8821a/halhwimg8821a_rf.c
	modified:   hal/rtl8814a/rtl8814a_dm.c
	modified:   hal/rtl8814a/usb/usb_halinit.c
	modified:   include/hal_data.h
	modified:   include/rtw_mlme.h
	modified:   os_dep/linux/ioctl_cfg80211.c
	modified:   os_dep/linux/ioctl_linux.c
	modified:   os_dep/linux/os_intfs.c

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	android/android_ref_codes_10.x/realtek_wifi_SDK_for_android_10_x_20200505.tar.gz
	android/android_ref_codes_10.x/supplicant_overlay_config.tar.gz
	android/android_ref_codes_10.x/wpa_supplicant_8_10.x_rtw_29226.20200409.tar.gz
	android/android_ref_codes_11.x/
	save-log.sh
```
- `rm -rf android/android_ref_codes_10.x/realtek_wifi_SDK_for_android_10_x_20200505.tar.gz android/android_ref_codes_10.x/supplicant_overlay_config.tar.gz android/android_ref_codes_10.x/wpa_supplicant_8_10.x_rtw_29226.20200409.tar.gz android/android_ref_codes_11.x/ save-log.sh` to remove all new files.
- `git checkout Makefile android/android_ref_codes_10.x/Realtek_Wi-Fi_SDK_for_Android_10.pdf core/rtw_cmd.c ....` to reset all changed files.
- `sudo make dkms_install` to install it again using dkms.
- Don't need to restart computer. It work for me.

rajatkashyap47586, 2024-06-03:
> I was also having this issue. I originally installed the most recent version using dkms, but it didn't work and I tried a few times with no success. But I got it working.
>
> Ill share what I did to [...]

Hi
I tried the steps which you mentioned exactly in the same way still it is not working

Here are the part of the output of dmesg ->

usb 1-1: new high-speed USB device number 2 using ehci-pci
[  124.485127] usb 1-1: New USB device found, idVendor=2357, idProduct=0120, bcdDevice= 2.00
[  124.485131] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  124.485132] usb 1-1: Product: 802.11ac WLAN Adapter 
[  124.485132] usb 1-1: Manufacturer: Realtek 
[  124.485133] usb 1-1: SerialNumber: 00e04c000001
[  125.061910] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[  125.068207] Loaded X.509 cert 'benh@debian.org: 577e021cb980e0e820821ba7b54b4961b8b4fadf'
[  125.068296] Loaded X.509 cert 'romain.perier@gmail.com: 3abbc6ec146e09d1b6016ab9d6cf71dd233f0328'
[  125.068376] Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[  125.068455] Loaded X.509 cert 'wens: 61c038651aabdcf94bd0ac7ff06c7248db18c600'
[  125.074458] platform regulatory.0: firmware: direct-loading firmware regulatory.db
[  125.075875] platform regulatory.0: firmware: direct-loading firmware regulatory.db.p7s
[  125.080921] 88XXau: loading out-of-tree module taints kernel.
[  125.080950] 88XXau: module verification failed: signature and/or required key missing - tainting kernel
[  130.148611] usb 1-1: 88XXau 5c:62:8b:76:4d:ea hw_info[107]
[  130.202117] usbcore: registered new interface driver rtl88XXau
[  162.008579] BUG: kernel NULL pointer dereference, address: 0000000000000000
[  162.008584] #PF: supervisor read access in kernel mode
[  162.008586] #PF: error_code(0x0000) - not-present page
[  162.008587] PGD 0 P4D 0 
[  162.008588] Oops: 0000 [#1] PREEMPT SMP NOPTI
[  162.008591] CPU: 6 PID: 11 Comm: kworker/u16:0 Tainted: G        W  OE      6.6.15-amd64 #1  Kali 6.6.15-2kali1
[  162.008593] Hardware name: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[  162.008613] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[  162.008658] RIP: 0010:__cfg80211_connect_result+0x325/0x920 [cfg80211]
[  162.008689] Code: f6 74 1e 83 f9 0e 0f b7 d0 8d 79 01 40 0f 95 c6 48 0f a3 ca 0f 82 af 01 00 00 89 f9 66 85 c0 75 dd 41 80 8c 24 cc 00 00 00 01 <41> 8b 06 4c 89 e9 31 d2 41 89 84 24 c0 03 00 00 41 0f b7 46 04 66
[  162.008690] RSP: 0018:ffffa57f00063d90 EFLAGS: 00010202
[  162.008691] RAX: 0000000000000000 RBX: ffff8c7502742018 RCX: 0000000000000001
[  162.008693] RDX: 0000000000000000 RSI: 0000000000000001 RDI: ffff8c750db98f00
[  162.008693] RBP: ffffa57f00063de0 R08: 0000000000000000 R09: 0000000000000000
[  162.008694] R10: 0000000000000002 R11: 0000000000000000 R12: ffff8c750db98800
[  162.008695] R13: ffff8c750db98c30 R14: 0000000000000000 R15: 0000000000000001
[  162.008696] FS:  0000000000000000(0000) GS:ffff8c75d2780000(0000) knlGS:0000000000000000
[  162.008697] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  162.008698] CR2: 0000000000000000 CR3: 000000010d550000 CR4: 00000000000106e0
[  162.008699] Call Trace:
[  162.008702]  <TASK>
[  162.008703]  ? __die+0x23/0x70
[  162.008709]  ? page_fault_oops+0x171/0x4e0
[  162.008713]  ? exc_page_fault+0x7f/0x180
[  162.008716]  ? asm_exc_page_fault+0x26/0x30
[  162.008719]  ? __cfg80211_connect_result+0x325/0x920 [cfg80211]
[  162.008747]  ? cfg80211_process_wdev_events+0x84/0x1c0 [cfg80211]
[  162.008768]  cfg80211_process_wdev_events+0x84/0x1c0 [cfg80211]
[  162.008790]  cfg80211_process_rdev_events+0x2a/0x40 [cfg80211]
[  162.008811]  cfg80211_event_work+0x29/0x40 [cfg80211]
[  162.008829]  process_one_work+0x171/0x340
[  162.008832]  worker_thread+0x27b/0x3a0
[  162.008834]  ? __pfx_worker_thread+0x10/0x10
[  162.008835]  kthread+0xe5/0x120
[  162.008839]  ? __pfx_kthread+0x10/0x10
[  162.008841]  ret_from_fork+0x31/0x50
[  162.008844]  ? __pfx_kthread+0x10/0x10
[  162.008846]  ret_from_fork_asm+0x1b/0x30
[  162.008850]  </TASK>
[  162.008850] Modules linked in: 88XXau(OE) cfg80211 snd_seq_dummy snd_hrtimer snd_seq snd_seq_device rfkill qrtr vboxsf snd_intel8x0 snd_ac97_codec intel_rapl_msr ac97_bus intel_rapl_common snd_pcm intel_uncore_frequency_common snd_timer intel_pmc_core rapl snd soundcore joydev vboxguest pcspkr ac sunrpc sg serio_raw evdev binfmt_misc loop dm_mod configfs efi_pstore nfnetlink ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic hid_generic usbhid hid sd_mod t10_pi crc32_pclmul crc64_rocksoft_generic crc32c_intel crc64_rocksoft crc_t10dif sr_mod crct10dif_generic cdrom crct10dif_pclmul ghash_clmulni_intel crc64 crct10dif_common sha512_ssse3 ata_generic ahci sha256_ssse3 libahci sha1_ssse3 ata_piix ohci_pci vboxvideo libata drm_vram_helper ohci_hcd drm_ttm_helper ehci_pci ttm aesni_intel ehci_hcd scsi_mod drm_kms_helper crypto_simd usbcore video cryptd drm psmouse e1000 scsi_common i2c_piix4 usb_common battery wmi button
[  162.008891] CR2: 0000000000000000

rajatkashyap47586, 2024-06-03:
> * make dkms_install

Hey
I tried this as well but when I done with the steps you mentioned it works well but when I shutdown and open the system again I get the same error 

[   56.644359] usbcore: registered new interface driver rtl88XXau
[   87.985508] BUG: kernel NULL pointer dereference, address: 0000000000000000
[   87.985514] #PF: supervisor read access in kernel mode
[   87.985515] #PF: error_code(0x0000) - not-present page

This causing my whole kali system freeze after some couple of minutes.

cblknittights, 2024-06-03:
This was a "rebuild" of an old laptop for me... so **I just wiped it clean** since the install worked in the live USB...and did the following

git clone (repo listed in readme.md)

git checkout 63cf0b4

sudo make dkms_install

Plugged in the device, picked the AP.. done... No issues since. 

This was on Kubuntu 24.04 ESR using 6.8.0 kernel.

Mefist0, 2024-06-04:
Same here.
Huge thanks everybody who suggested the solution with `git checkout 63cf0b4` !
`make dkms_install` takes too long, so I used `make && sudo make install` approach.

cblknittights, 2024-06-05:
Here is some more possible info the issue(s) with latest v.  63cf0b4 branch 

Wireless DOES WORK using this device on 63cf0b4, but I get this in dmesg on bootup:
```

[   94.785789] ------------[ cut here ]------------
[   94.785800] UBSAN: array-index-out-of-bounds in /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_wlan_util.c:1905:48
[   94.785812] index 1 is out of range for type 'u8 [1]'
[   94.785818] CPU: 0 PID: 0 Comm: swapper/0 Tainted: P           OE      6.8.0-31-generic #31-Ubuntu
[   94.785825] Hardware name: Dell Inc. Inspiron 1470/0M7CYJ, BIOS A03 10/19/2009
[   94.785828] Call Trace:
[   94.785833]  <IRQ>
[   94.785839]  dump_stack_lvl+0x48/0x70
[   94.785856]  dump_stack+0x10/0x20
[   94.785863]  __ubsan_handle_out_of_bounds+0xc6/0x110
[   94.785872]  HT_caps_handler+0x339/0x350 [88XXau]
[   94.786133]  OnAssocRsp+0x222/0x290 [88XXau]
[   94.786349]  mgt_dispatcher+0x181/0x250 [88XXau]
[   94.786558]  validate_recv_mgnt_frame+0x203/0x3f0 [88XXau]
[   94.786725]  validate_recv_frame+0x146/0x1e0 [88XXau]
[   94.786725]  recv_func+0x50/0x2e0 [88XXau]
[   94.786725]  pre_recv_entry+0x59/0x140 [88XXau]
[   94.786725]  recvbuf2recvframe+0x4b/0x1e0 [88XXau]
[   94.786725]  usb_recv_tasklet+0x54/0x100 [88XXau]
[   94.786725]  tasklet_action_common.isra.0+0xfd/0x1e0
[   94.786725]  tasklet_action+0x22/0x30
[   94.786725]  __do_softirq+0xe1/0x363
[   94.786725]  ? ioapic_ack_level+0x57/0x140
[   94.786725]  __irq_exit_rcu+0xd7/0x100
[   94.786725]  irq_exit_rcu+0xe/0x20
[   94.786725]  common_interrupt+0xa4/0xb0
[   94.786725]  </IRQ>
[   94.786725]  <TASK>
[   94.786725]  asm_common_interrupt+0x27/0x40
[   94.786725] RIP: 0010:cpuidle_enter_state+0xda/0x730
[   94.786725] Code: d2 f6 fe e8 38 ee ff ff 8b 53 04 49 89 c7 0f 1f 44 00 00 31 ff e8 f6 69 f5 fe 80 7d d0 00 0f 85 61 02 00 00 fb 0f 1f 44 00 00 <45> 85 f6 0f 88 f7 01 00 00 4d 63 ee 49 83 fd 0a 0f 83 17 05 00 00
[   94.786725] RSP: 0000:ffffffffb2e03db8 EFLAGS: 00000246
[   94.786725] RAX: 0000000000000000 RBX: ffff9524f7e40810 RCX: 0000000000000000
[   94.786725] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
[   94.786725] RBP: ffffffffb2e03e08 R08: 0000000000000000 R09: 0000000000000000
[   94.786725] R10: 0000000000000000 R11: 0000000000000000 R12: ffffffffb30e5ba0
[   94.786725] R13: 0000000000000003 R14: 0000000000000003 R15: 0000001611ab5026
[   94.786725]  ? cpuidle_enter_state+0x33c/0x730
[   94.786725]  ? finish_task_switch.isra.0+0x93/0x300
[   94.786725]  cpuidle_enter+0x2e/0x50
[   94.786725]  call_cpuidle+0x23/0x60
[   94.786725]  cpuidle_idle_call+0x11d/0x190
[   94.786725]  do_idle+0x87/0xf0
[   94.786725]  cpu_startup_entry+0x2a/0x30
[   94.786725]  rest_init+0xde/0x100
[   94.786725]  ? acpi_enable_subsystem+0xe6/0x2a0
[   94.786725]  ? static_key_disable+0x1f/0x30
[   94.786725]  arch_call_rest_init+0xe/0x30
[   94.786725]  start_kernel+0x354/0x450
[   94.786725]  x86_64_start_reservations+0x18/0x30
[   94.786725]  x86_64_start_kernel+0xbf/0x110
[   94.786725]  secondary_startup_64_no_verify+0x184/0x18b
[   94.786725]  </TASK>
[   94.788414] ---[ end trace ]---
[   94.788419] ------------[ cut here ]------------
[   94.788421] UBSAN: array-index-out-of-bounds in /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_wlan_util.c:1910:75
[   94.788428] index 2 is out of range for type 'u8 [1]'
[   94.788433] CPU: 0 PID: 0 Comm: swapper/0 Tainted: P           OE      6.8.0-31-generic #31-Ubuntu
[   94.788439] Hardware name: Dell Inc. Inspiron 1470/0M7CYJ, BIOS A03 10/19/2009
[   94.788442] Call Trace:
[   94.788445]  <IRQ>
[   94.788448]  dump_stack_lvl+0x48/0x70
[   94.788457]  dump_stack+0x10/0x20
[   94.788463]  __ubsan_handle_out_of_bounds+0xc6/0x110
[   94.788471]  HT_caps_handler+0x205/0x350 [88XXau]
[... 129 lines cut ...]
[   94.794488] RAX: 0000000000000000 RBX: ffff9524f7e40810 RCX: 0000000000000000
[   94.794488] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
[   94.794488] RBP: ffffffffb2e03e08 R08: 0000000000000000 R09: 0000000000000000
[   94.794488] R10: 0000000000000000 R11: 0000000000000000 R12: ffffffffb30e5ba0
[   94.794488] R13: 0000000000000003 R14: 0000000000000003 R15: 0000001611ab5026
[   94.794488]  ? cpuidle_enter_state+0x33c/0x730
[   94.794488]  ? finish_task_switch.isra.0+0x93/0x300
[   94.794488]  cpuidle_enter+0x2e/0x50
[   94.794488]  call_cpuidle+0x23/0x60
[   94.794488]  cpuidle_idle_call+0x11d/0x190
[   94.794488]  do_idle+0x87/0xf0
[   94.794488]  cpu_startup_entry+0x2a/0x30
[   94.794488]  rest_init+0xde/0x100
[   94.794488]  ? acpi_enable_subsystem+0xe6/0x2a0
[   94.794488]  ? static_key_disable+0x1f/0x30
[   94.794488]  arch_call_rest_init+0xe/0x30
[   94.794488]  start_kernel+0x354/0x450
[   94.794488]  x86_64_start_reservations+0x18/0x30
[   94.794488]  x86_64_start_kernel+0xbf/0x110
[   94.794488]  secondary_startup_64_no_verify+0x184/0x18b
[   94.794488]  </TASK>
[   94.796032] ---[ end trace ]---
[  106.642651] warning: `kded5' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
[  376.699403] perf: interrupt took too long (2514 > 2500), lowering kernel.perf_event_max_sample_rate to 79000
```

koku17, 2024-06-05:
I had same issue in Archer T2U Nano
```
$ lsusb | cut -d ":" -f 3 | grep Archer
0xxx TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano]
```

I have partial solution discussed in issue #1155
> I have a got a temporary solution, but not good for bleeding edge kernels
> 
> Get all the commits done before
> 
> ```
> $ git log --oneline > ../rtl8812au.commits
> ```
> 
> Grab a cofee and read them and select what you want :
> 
> ```
> $ less git ../rtl*.commits
> ```
> 
> Replace `<commit>` with what you selected,
> 
> ```
> $ git reset <commit> --hard
> HEAD is now at d8ad927 Fix build on 6.x kernels.
> ```
> 
> In my case I have selected `d8ad927`
> 
> ```
> $ git reset d8ad927 --hard
> ```
> 
> Remove the module if exist (for more info see installation instructions)
> 
> ```
> $ cd rtl* ; sudo make dkms_remove
> ```
> 
> After that follow installation instructions.
> 
> ```
>  0 0      ---- I didn't wonder git had this feature ----
> \___/
> ```

It just works fine for 6.x kernels as confirmed by others in issue #1157

achille54, 2024-06-07:
Hi, i've tried to use `git checkout 63cf0b4` but is not working.

the steps i've done:
1) `git clone https://github.com/aircrack-ng/rtl8812au.git`
2) `cd rtl8812au/`
3) `git checkout 63cf0b4`
4) `make`
5) `sudo make install`

6.5.0-35-generic #35~22.04.1-Ubuntu

running the command to check if it is present:
`find /lib/modules/`uname -r`/ -name "88XXau.ko"`
`/lib/modules/6.5.0-35-generic/kernel/drivers/net/wireless/88XXau.ko`

can someone help me?

koku17, 2024-06-07:
you can try this

> I had same issue in Archer T2U Nano
> 
> ```
> $ lsusb | cut -d ":" -f 3 | grep Archer
> 0xxx TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano]
> ```
> 
> I have partial solution discussed in issue #1155 After changing to commit `d8ad927` I haven't got any `---[ cut here ]---` in `dmesg`
> 
> > I have a got a temporary solution, but not good for bleeding edge kernels
> > Get all the commits done before
> > ```
> > $ git log --oneline > ../rtl8812au.commits
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Grab a cofee and read them and select what you want :
> > ```
> > $ less git ../rtl*.commits
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Replace `<commit>` with what you selected,
> > ```
> > $ git reset <commit> --hard
> > HEAD is now at d8ad927 Fix build on 6.x kernels.
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > In my case I have selected `d8ad927`
> > ```
> > $ git reset d8ad927 --hard
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Remove the module if exist (for more info see installation instructions)
> > ```
> > $ cd rtl* ; sudo make dkms_remove
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > After that follow installation instructions.
> > ```
> >  0 0      ---- I didn't wonder git had this feature ----
> > \___/
> > ```
> 
> It just works fine for 6.x kernels as confirmed by others in issue #1155

Chandan-thakur, 2024-06-15:
The trick is to download the driver into /usr/src cause when dkms tries to build the firmware it looks for the source code in that folder.
Here is the complete command:

cd /usr/src
sudo git clone https://github.com/aircrack-ng/rtl8812au.git
cd /usr/src/rtl8812au
sudo git checkout 63cf0b4
sudo make dkms_install

alb3rt-on, 2024-06-18:
If that doesn't work and you're using Virtual Machine, check out this fix that worked for me.

https://github.com/aircrack-ng/rtl8812au/issues/1157#issuecomment-2176550393

rajatkashyap47586, 2024-06-20:
I tried all mentioned solution but nothing works for me.

cblknittights, 2024-06-21:
> I tried all mentioned solution but nothing works for me.

Try  this:

Get a LIVE ISO of your current distro, burn to USB.. Boot to the LIVE USB.

Then do the install process:
```
sudo apt-get install git build-essential

git clone (repo listed in readme.md)

cd rtl***

git checkout 63cf0b4

sudo make dkms_install
```
Plug in the device, or unplug, replug if you left it in...

IF it WORKS IN THE LIVE USB test.. Then this is a issue with YOUR INSTALL. Likely leftover earlier versions of the code or something..

IF it does NOT WORK in the live USB, then its likely a hardware issue with the device you have.

mazzi, 2024-06-24:
With [63cf0b4](https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d) in `Ubuntu 24.04 LTS (GNU/Linux 6.8.0-35-generic x86_64)` worked for me. Some `array-index-out-of-bounds` errors in `dmesg` but the board and wifi configuration worked. I have some doubts about the ping times but I need to test.
Thank you very much for this driver and the tips.

mazzi, 2024-06-25:
Follow up to my comment, debug info on `dmesg`

```
[    8.003646] ------------[ cut here ]------------
[    8.003650] UBSAN: array-index-out-of-bounds in /root/rtl8812au/core/rtw_wlan_util.c:1905:48
[    8.003654] index 1 is out of range for type 'u8 [1]'
[    8.003656] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G           OE      6.8.0-35-generic #35-Ubuntu
[    8.003659] Hardware name: Apple Inc. Macmini7,1/Mac-35C5E08120C7EEAF, BIOS 431.140.6.0.0 06/13/2021
[    8.003661] Call Trace:
[    8.003666]  <IRQ>
[    8.003669]  dump_stack_lvl+0x48/0x70
[    8.003684]  dump_stack+0x10/0x20
[    8.003687]  __ubsan_handle_out_of_bounds+0xc6/0x110
[    8.003693]  HT_caps_handler+0x339/0x350 [88XXau]
[    8.003822]  OnAssocRsp+0x222/0x290 [88XXau]
[    8.003924]  mgt_dispatcher+0x181/0x250 [88XXau]
[    8.004013]  validate_recv_mgnt_frame+0x203/0x3f0 [88XXau]
[    8.006420]  validate_recv_frame+0x146/0x1e0 [88XXau]
[    8.006537]  recv_func+0x50/0x2e0 [88XXau]
[    8.006638]  pre_recv_entry+0x59/0x140 [88XXau]
[    8.006737]  recvbuf2recvframe+0x4b/0x1e0 [88XXau]
[    8.006847]  usb_recv_tasklet+0x54/0x100 [88XXau]
[    8.006967]  tasklet_action_common.isra.0+0xfd/0x1e0
[    8.006973]  tasklet_action+0x22/0x30
[    8.006976]  __do_softirq+0xe1/0x363
[    8.006981]  __irq_exit_rcu+0xd7/0x100
[    8.006984]  irq_exit_rcu+0xe/0x20
[    8.006987]  common_interrupt+0xa4/0xb0
[    8.006992]  </IRQ>
[    8.006994]  <TASK>
[    8.006996]  asm_common_interrupt+0x27/0x40
[    8.007001] RIP: 0010:cpuidle_enter_state+0xda/0x730
[    8.007006] Code: b2 f6 fe e8 38 ee ff ff 8b 53 04 49 89 c7 0f 1f 44 00 00 31 ff e8 c6 4a f5 fe 80 7d d0 00 0f 85 61 02 00 00 fb 0f 1f 44 00 00 <45> 85 f6 0f 88 f7 01 00 00 4d 63 ee 49 83 fd 0a 0f 83 17 05 00 00
[    8.007009] RSP: 0018:ffffb0ca000dfe18 EFLAGS: 00000246
[    8.007013] RAX: 0000000000000000 RBX: ffffd0c9ffd1bdf0 RCX: 0000000000000000
[    8.007015] RDX: 0000000000000002 RSI: 0000000000000000 RDI: 0000000000000000
[    8.007017] RBP: ffffb0ca000dfe68 R08: 0000000000000000 R09: 0000000000000000
[    8.007019] R10: 0000000000000000 R11: 0000000000000000 R12: ffffffff9c0e5bc0
[    8.007021] R13: 0000000000000003 R14: 0000000000000003 R15: 00000001dd0d23fb
[    8.007026]  ? cpuidle_enter_state+0xca/0x730
[    8.007029]  ? finish_task_switch.isra.0+0x93/0x300
[    8.007034]  cpuidle_enter+0x2e/0x50
[    8.007040]  call_cpuidle+0x23/0x60
[    8.007044]  cpuidle_idle_call+0x11d/0x190
[    8.007048]  do_idle+0x87/0xf0
[    8.007052]  cpu_startup_entry+0x2a/0x30
[    8.007056]  start_secondary+0x129/0x160
[    8.007063]  secondary_startup_64_no_verify+0x184/0x18b
[    8.007070]  </TASK>
[    8.007089] ---[ end trace ]---
[    8.007092] ------------[ cut here ]------------
[    8.007093] UBSAN: array-index-out-of-bounds in /root/rtl8812au/core/rtw_wlan_util.c:1910:75
[    8.008260] index 2 is out of range for type 'u8 [1]'
[    8.008264] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G           OE      6.8.0-35-generic #35-Ubuntu
[    8.008267] Hardware name: Apple Inc. Macmini7,1/Mac-35C5E08120C7EEAF, BIOS 431.140.6.0.0 06/13/2021
[    8.008268] Call Trace:
[    8.008270]  <IRQ>
[    8.008273]  dump_stack_lvl+0x48/0x70
[    8.008281]  dump_stack+0x10/0x20
[    8.008285]  __ubsan_handle_out_of_bounds+0xc6/0x110
[    8.008290]  HT_caps_handler+0x205/0x350 [88XXau]
[    8.009458]  OnAssocRsp+0x222/0x290 [88XXau]
[    8.009529]  mgt_dispatcher+0x181/0x250 [88XXau]
[    8.009604]  validate_recv_mgnt_frame+0x203/0x3f0 [88XXau]
[    8.009708]  validate_recv_frame+0x146/0x1e0 [88XXau]
[    8.009811]  recv_func+0x50/0x2e0 [88XXau]
[    8.009903]  pre_recv_entry+0x59/0x140 [88XXau]
[    8.010004]  recvbuf2recvframe+0x4b/0x1e0 [88XXau]
[    8.010115]  usb_recv_tasklet+0x54/0x100 [88XXau]
[    8.010229]  tasklet_action_common.isra.0+0xfd/0x1e0
[... 146 lines cut ...]
[    8.118455]  __sys_sendto+0x21e/0x230
[    8.118460]  __x64_sys_sendto+0x24/0x40
[    8.118462]  x64_sys_call+0x1c33/0x25c0
[    8.118465]  do_syscall_64+0x7f/0x180
[    8.118469]  ? __x64_sys_epoll_wait+0x6f/0x110
[    8.118473]  ? syscall_exit_to_user_mode+0x86/0x260
[    8.118476]  ? do_syscall_64+0x8c/0x180
[    8.118479]  ? __fput_sync+0x1c/0x30
[    8.118481]  ? syscall_exit_to_user_mode+0x86/0x260
[    8.118483]  ? do_syscall_64+0x8c/0x180
[    8.118486]  ? do_syscall_64+0x8c/0x180
[    8.118488]  ? do_syscall_64+0x8c/0x180
[    8.118491]  ? common_interrupt+0x54/0xb0
[    8.118494]  entry_SYSCALL_64_after_hwframe+0x78/0x80
[    8.118497] RIP: 0033:0x7607d032bead
[    8.118506] Code: c3 ff ff ff ff 64 89 02 eb b9 0f 1f 00 f3 0f 1e fa 80 3d a5 f1 0d 00 00 41 89 ca 74 20 45 31 c9 45 31 c0 b8 2c 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 6b c3 66 2e 0f 1f 84 00 00 00 00 00 55 48 89
[    8.118508] RSP: 002b:00007ffd097ae568 EFLAGS: 00000246 ORIG_RAX: 000000000000002c
[    8.118511] RAX: ffffffffffffffda RBX: 000062fa9680ca40 RCX: 00007607d032bead
[    8.118512] RDX: 0000000000000060 RSI: 000062fa9682d840 RDI: 0000000000000004
[    8.118514] RBP: 00007ffd097ae580 R08: 0000000000000000 R09: 0000000000000000
[    8.118515] R10: 0000000000000000 R11: 0000000000000246 R12: 000062fa9682e740
[    8.118516] R13: 00007ffd097ae5dc R14: 000062fa96816620 R15: 000062fa9680c7a0
[    8.118518]  </TASK>
[    8.118519] ---[ end trace ]---
```

mazzi, 2024-06-25:
> Here is some more possible info the issue(s) with latest v. [63cf0b4](https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d) branch
>
> Wireless DOES WORK using this [...]

Do you get `(DUP!)` packets when you ping another host inside your WLAN?

rajatkashyap47586, 2024-06-27:
I tried all solution mentioned above nothing works yet.

Can someone please help

I'm using :-
OS - Kali 6.8.11-1kali2 (2024-05-30) x86_64 GNU/Linux
Container - Virtual Box (Version 7.0.18 r162988 (Qt5.15.2))
Wifi adaptor - TP-Link AC600

Also just an observation - When I delete the kali machine from virtual box and I reinstalled it again that for the first time wifi adaptor works perfectly fine post that when I turned the kali machine off and switch on again the issue again came.

The error message I can see from dmesg output

88XXau: loading out-of-tree module taints kernel.
[  369.641833] 88XXau: module verification failed: signature and/or required key missing - tainting kernel
[  375.488887] usb 1-1: 88XXau 5c:62:8b:76:4d:ea hw_info[107]
[  375.536102] usbcore: registered new interface driver rtl88XXau
[  408.101264] BUG: kernel NULL pointer dereference, address: 0000000000000000
[  408.101272] #PF: supervisor read access in kernel mode
[  408.101275] #PF: error_code(0x0000) - not-present page

nuketech9, 2024-06-29:
Just want to thank everyone for their efforts in figuring this out and taking the time to put all these steps up. Even after changing to the latest v. [63cf0b4] branch it wouldn't work until I cloned to and installed from the /src/rtl8812au folder. That finally fixed it for me and continues to work after a reboot. I know just enough to be dangerous, I would have never figured this all out on my own. Back up and running again. Thanks all.

mazzi, 2024-06-29:
> Just want to thank everyone for their efforts in figuring this out and taking the time to put all these steps up. Even after changing to the latest v. [...]

Thanks for sharing!

Distro? Kernel?

Just being curious. Already returned the device and plugged an Ethernet cable...

nuketech9, 2024-06-29:
> Thanks for sharing!
>
> Distro? Kernel?
>
> Just being curious. Already returned the device and plugged an Ethernet cable...

Kubuntu 24.04
6.8.0-36-generic 

Had just done an upgrade from Kubuntu 22 where it was working fine.  Could see it in lsusb and iwconfig  but would not show up in NetworkManager. Was using dkms. uninstall/removed/reinstalled several times but wouldn't work till I moved the clone to \src

mazzi, 2024-06-29:
Try to check if you have `DUP!` when you do a ping. Was my case despite was detected and configured with an IP.

nuketech9, 2024-06-29:
No DUP!s 
Doing a quick search someone mentions that could also be caused by having duplicate MAC addresses in the MAC table as well as duplicate IP's FWIW

mazzi, 2024-06-29:
I've read that the MAC address can be forced on netplan.
Thanks for the replies @nuketech9 ! Good luck.

neestzjov, 2024-06-30:
It works, it makes it display all SSIDs around now.

honkingantelope, 2024-07-16:
ran into the same issue on a full-upgrade'd Kali raspberry PI image, RPI4.  resetting to 63cf0b4 and remove-build-install did the trick.  Soooo frustrating!

JoViOc, 2024-07-21:
Hi, I was able to enable the tp-link using git reset in [63cf0b4](https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d). However, now i am facing another problem: it does not scan the networks. Check attached image. 
[image]

Details:
Linux 6.5.0-44-generic #44~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Jun 18 14:36:16 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

lsusb:
Bus 001 Device 004: ID 2357:011f TP-Link 802.11ac WLAN Adapter 

iwconfig:
wlx3c52a18608bd  unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

dmesg:
[    4.108400] WARNING: CPU: 18 PID: 539 at /home/XXXX/rtl8812au/core/rtw_odm.c:110 rtw_odm_init_ic_type+0x5e/0x70 [88XXau]
[    4.108496] Modules linked in: intel_rapl_msr intel_rapl_common snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi snd_hda_intel edac_mce_amd snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec kvm_amd snd_hda_core snd_hwdep binfmt_misc kvm snd_pcm irqbypass crct10dif_pclmul polyval_clmulni polyval_generic ghash_clmulni_intel sha256_ssse3 snd_seq_midi sha1_ssse3 snd_seq_midi_event 88XXau(OE+) aesni_intel snd_rawmidi crypto_simd cryptd nls_iso8859_1 snd_seq rapl snd_seq_device cfg80211 input_leds snd_timer joydev wmi_bmof snd k10temp ccp soundcore mac_hid sch_fq_codel msr parport_pc ppdev lp parport drm efi_pstore ip_tables x_tables autofs4 uas usb_storage hid_generic usbhid hid nvme crc32_pclmul igb i2c_piix4 ahci i2c_algo_bit nvme_core xhci_pci libahci dca xhci_pci_renesas nvme_common wmi
[    4.108575] RIP: 0010:rtw_odm_init_ic_type+0x5e/0x70 [88XXau]
[    4.108666]  ? rtw_odm_init_ic_type+0x5e/0x70 [88XXau]
[    4.108741]  ? rtw_odm_init_ic_type+0x5e/0x70 [88XXau]
[    4.108799]  Init_ODM_ComInfo+0x38/0x5a0 [88XXau]
[    4.108867]  rtl8812_init_dm_priv+0x50/0xb0 [88XXau]
[    4.108917]  rtw_hal_dm_init+0x22/0x50 [88XXau]
[    4.108970]  rtw_init_drv_sw+0x25d/0x290 [88XXau]
[    4.109023]  rtw_usb_primary_adapter_init+0x14c/0xa50 [88XXau]
[    4.109072]  rtw_usb_primary_adapter_init+0x41c/0xa50 [88XXau]
[    4.109157]  ? __pfx_init_module+0x10/0x10 [88XXau]
[    4.109193]  init_module+0x55/0xff0 [88XXau]
[    4.109772] usbcore: registered new interface driver rtl88XXau
[    4.114004] rtl88XXau 1-2.1:1.0 wlx3c52a18608bd: renamed from wlan0
[   29.759274]  odm_get_tracking_table+0xf4/0x590 [88XXau]
[   29.759366]  odm_txpowertracking_callback_thermal_meter+0x3b7/0x740 [88XXau]
[   29.759441]  odm_txpowertracking_check+0x5d/0xe0 [88XXau]
[   29.759512]  phydm_rf_watchdog+0xe/0x20 [88XXau]
[   29.759582]  halrf_watchdog+0xe/0x20 [88XXau]
[   29.759654]  phydm_watchdog+0x8c/0xf0 [88XXau]
[   29.759740]  rtw_phydm_watchdog+0x1d8/0x230 [88XXau]
[   29.759846]  rtl8812_HalDmWatchDog+0x56/0x70 [88XXau]
[   29.759943]  rtw_hal_dm_watchdog+0x21/0x30 [88XXau]
[   29.760051]  rtw_drvextra_cmd_hdl+0x16d/0x3e0 [88XXau]
[   29.760115]  rtw_cmd_thread+0x344/0x530 [88XXau]
[   29.760169]  ? __pfx_rtw_drvextra_cmd_hdl+0x10/0x10 [88XXau]
[   29.760227]  ? __pfx_rtw_cmd_thread+0x10/0x10 [88XXau]
[   29.760321] Modules linked in: xt_conntrack nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack_netlink nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xfrm_user xfrm_algo xt_addrtype nft_compat br_netfilter bridge stp llc nf_tables libcrc32c nfnetlink overlay intel_rapl_msr intel_rapl_common snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi snd_hda_intel edac_mce_amd snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec kvm_amd snd_hda_core snd_hwdep binfmt_misc kvm snd_pcm irqbypass crct10dif_pclmul polyval_clmulni polyval_generic ghash_clmulni_intel sha256_ssse3 snd_seq_midi sha1_ssse3 snd_seq_midi_event 88XXau(OE) aesni_intel snd_rawmidi crypto_simd cryptd nls_iso8859_1 snd_seq rapl snd_seq_device cfg80211 input_leds snd_timer joydev wmi_bmof snd k10temp ccp soundcore mac_hid sch_fq_codel msr parport_pc ppdev lp parport drm efi_pstore ip_tables x_tables autofs4 uas usb_storage hid_generic usbhid hid nvme crc32_pclmul igb i2c_piix4 ahci i2c_algo_bit nvme_core xhci_pci libahci dca xhci_pci_renesas

JoViOc, 2024-07-21:
I solved using the description from [achille54](https://github.com/achille54)

neelabalan, 2024-08-11:
Doesn't detect 5 GHz network for me in both

```
TP-Link Archer T9UH v1 [Realtek RTL8814AU]
TP-Link Archer T4U v2 [Realtek RTL8812AU]
```

This is after checking out [this commit](https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d) and running `sudo make`, `sudo make install`

Inklare, 2024-09-07:
> I was also having this issue. I originally installed the most recent version using dkms, but it didn't work and I tried a few times with no success. But I got it working.
>
> Ill share what I did to [...]

BROOO THANK YOU VRY MUCH!!!

maniban, 2024-09-08:
Hi All,

I have tried the following as per my history snippet i.e.

  574  git clone https://github.com/aircrack-ng/rtl8812au.git
  575  cd rt*
  578  sudo apt-get update -y
  579  sudo apt-get install bc mokutil build-essential libelf-dev linux-headers-`uname -r` 
  580  sudo apt-get upgrade
  **581  git checkout 63cf0b4**
  582  ls -ltr
  583  cat README.md 
  584  sudo make dkms_install
  585  sudo service NetworkManager restart

But still the same isn't working for me. Could anyone please help as installed kubuntu 24.04 with 6.8.0-41-generic.

my dmesg logs attached [dmesg.log](https://github.com/user-attachments/files/16920765/dmesg.log)

Please help me with this  

Bus 001 Device 002: ID 2357:011e TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano]

neestzjov, 2024-09-08:
> Hi All,
>
> I have tried the following as per my history snippet i.e.
>
> 574 git clone https://github.com/aircrack-ng/rtl8812au.git 575 cd rt* 578 sudo apt-get update -y 579 sudo apt-get install bc [...]
```

git clone...... (from the readme)

git checkout 63cf0b4

sudo make dkms_install

Plug in/or replug device.. and I clicked on the network manager and it showed wireless networks!
```

Try this

CursedOddity, 2024-09-14:
None of this worked for me either...

I spent over $100 on a AWUS036ACH after my old one died. used it for 2 days then stupidly updated. Now i cant get it to work. I have scoured the internet and used every step by step possible. 

It just DOES NOT WORK for Raspberry Pi 5 and though i haven't spent much time on the VM i didn't get it to work there either. 

I feel sick knowing i just spend that much money of a useless adapter.. UGHHH

matthew11, 2024-09-15:
> It just DOES NOT WORK for Raspberry Pi 5 and though i haven't spent much time on the VM i didn't get it to work there either.

Pi 5 is an ARM64-based computer, you likely need to do some additional steps as described in the readme.

tukkek, 2024-09-17:
Just wanted to add my report that the latest version fails to find networks with my `TP-Link Nano USB AC600 Archer T2U` but using a fresh clone and switching to https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d has it working. Other than that, this is already the best, fastest and most stable of the 3 drivers I've used so far so big thanks to the developers and hopefully the latest issues can be fixed soon.

My kernel is: `6.10.6-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.10.6-1 (2024-08-19) x86_64 GNU/Linux`.

For both latest and detached versions, I get a nasty trace message during boot that wasn't there before and I haven't updated any system packages either. I won't share it since I'm using the older version now and everything is fine.

arxo, 2024-09-27:
I also ran into the same issue on Linux Mint 22, using commit https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d solved it for me as well. I hope this regression gets fixed in the latest version soon.

SteveTran159, 2024-10-02:
Thank you all so much. I also got stuck in installing the driver for TP-Link Archer T2U Nano (Realtek rtl8811au) in Xubuntu 22.04. I could do this in live usb smoothly, but I got an error when installing Xu on my old hard drive. 
ERROR (dkms apport): binary package for 8812au: 5.6.4.2_35491.20191025 not found
I surfed a lot and tried many ways like using old version (5.3.4), using rtl8812au-dkms (from ppa:kelebek333/kablosuz) but all failed, still got the same error or successfully installed but couldn't connect to wifi. After all, I tried the way in this page and it worked with the git code 63cf0b4 and two commands sudo make, sudo make install. The driver was not shown in dkms but it worked after I rebooted. (I got error when running sudo make dkms_install, the installing process showed some error and DKMS showed the package installed with a warning 'built different from installed')

NoDuckyAnyMore, 2024-10-04:
> I was also having this issue. I originally installed the most recent version using dkms, but it didn't work and I tried a few times with no success. But I got it working.
>
> Ill share what I did to [...]

Holy god!!! Bro, you are a true hero!!  Work for me!!!!

SteveTran159, 2024-10-12:
One more thing I'd like to share is, I also got the error of lacking package pahole and 'Skipping BTF generation xxx. due to unavailability of vmlinux' while compiling (run sudo make). I solved this after reading an article on askubuntu. The compiling process need the vmlinux file in /usr/lib/modules/`uname-r`/build but it's not there. The compiling process succeeded after I installed the package pahole and created a symlink for the vmlinux file (located in /sys/kernel/btf) in the above folder.
Besides Xubuntu 22.04, changing git code to 63cf0b4 is also a must in Linux Lite 6.6. In live usb Linux Lite 6.6, I forgot changing the git code but the installation process still worked, so I skipped this step in Linux Lite (official installation on hard drive). I saw the compiling process had some warnings, not run smoothly. I still could install the driver but couldn't connect to the internet wifi at home. Then I had to remove the driver by `sudo make uninstall`. In the second installation, I changed the git code to 63cf0b4 and it worked!

D-Nayte, 2024-11-01:
> git checkout 63cf0b4

Same here. Did it like you and just works fine now.

but after the git checkout i also did

- 32-bit ARM architecture (RPi 1/2/3/ & 0/Zero):
```
$ sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
$ sed -i 's/CONFIG_PLATFORM_ARM_RPI = n/CONFIG_PLATFORM_ARM_RPI = y/g' Makefile
```
Or, for ARM64 (RPI 3B+, 4B and Zero2) you will need to run:
```
$ sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
$ sed -i 's/CONFIG_PLATFORM_ARM64_RPI = n/CONFIG_PLATFORM_ARM64_RPI = y/g' Makefile
```

AND

ARM32
```
$ export ARCH=arm
$ sed -i 's/^MAKE="/MAKE="ARCH=arm\ /' dkms.conf
```

Or, for ARM64 run the following before re-building:
```
$ export ARCH=arm64
$ sed -i 's/^MAKE="/MAKE="ARCH=arm64\ /' dkms.conf
```

like mentioned in the Readme.

Rastafouille, 2024-12-04:
hello,
i try to install on ubuntu 24.04 kernel 6.8.0-1018-gcp.
Dkms installation of 63cf0b4 repo works fine but still no active wifi.....

mqcmd196, 2025-01-22:
For someone using Jetson, 

This solution also worked with JetPack r35.4.1 in Boston Dynamics Core I/O (a.k.a. Jetson Xavier NX). Thanks.

```
spot@core-io:~/wifi/rtl8812au$ uname -a
Linux core-io 5.10.120-coreio #1 SMP PREEMPT Wed Jan 10 18:55:27 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux
```

woodstoc242, 2025-04-20:
> BROOO THANK YOU VRY MUCH!!!

I am so grateful for this fix too! I am using ASUS USB-AC56 now on Linux Mint 20.3. With older Mint versions I used the driver manager with which the device was working, but after reboot it was not operative anymore. Due to the comment in the aircrack readme I was pointed to this forum entry and I was able to make it work.
Thank you very much for this! It worked exactly as described here! It would be great, if this could be fixed in the most recent version as well, since I suffered from this bug now for quite some time (and I wonder how many more still do).
Thanks a lot for all who have been involved in providing this information!