aircrack-ng/rtl8812au #1162: Alfa AWUS036EAC Monitor Mode issues

jjbyrnes29, 2024-06-10:
Hello.
I am successfully able to put this card into monitor mode, however packet/frame injection both do not work on this card when they had previously.

Tested Machines:
Android 14 (via inclusion as a kernel driver)
Fedora 40 (tested as sanity check via make install and make dkms_install)

Steps to reproduce:
Plug in adapter
Reboot device
$ sudo ip link set ifname down
$ sudo iw dev ifname set type monitor
$ sudo ip link set ifname up
$ sudo hcxdumptool -i ifname
$ sudo airodump-ng ifname
Both fail with blank screens in relation to detecting APs/Clients

$ sudo airmon-ng check kill
$ sudo airmon-ng start ifname
Yields the same result when testing hcxdumptool and airodump-ng.

Redditor247, 2024-06-24:
Bumping this, same issue with this driver installed. Any luck troubleshooting?

roelfrenkema, 2024-07-16:
Another bump.