aircrack-ng/rtl8812au #1163: New kernel issue

SeiV-K, 2024-06-12:
Hi,
owner of AWUS036ACS here. 

Just updated to the latest kernel 6.8 and attempted to install the driver for my adapter but it seems to be broken.

gary9872, 2024-06-12:
Did you upgrade your compiler to GCC-14?

SeiV-K, 2024-06-12:
> Did you upgrade your compiler to GCC-14?
> 
> I don't think it's a kernel issue, but rather a compiling issue. GCC14 probably broke it.

might be. However also using the "all-in-one" apt install realtek-rtl88xxau-dkms also fails at this stage
[image]
[image]

lab37, 2024-06-13:
kernel 6.8   issue
[image]

lab37, 2024-06-13:
gcc 13.2.0-15  
```
[  712.442095] 88XXau: loading out-of-tree module taints kernel.
[  712.442145] 88XXau: module verification failed: signature and/or required key missing - tainting kernel
[  712.556108] ------------[ cut here ]------------
[  712.556115] WARNING: CPU: 0 PID: 14198 at /var/lib/dkms/realtek-rtl88xxau/5.6.4.2~git20240510.b44d288/build/core/rtw_odm.c:62 rtw_odm_init_ic_type+0x44/0x70 [88XXau]
[  712.556198] Modules linked in: 88XXau(OE+) cfg80211 snd_seq_dummy snd_hrtimer snd_seq snd_seq_device rfkill qrtr snd_intel8x0 snd_ac97_codec ac97_bus snd_pcm vboxsf intel_rapl_msr intel_rapl_common snd_timer snd vboxguest joydev pcspkr soundcore ac evdev sg serio_raw sunrpc binfmt_misc efi_pstore loop dm_mod configfs nfnetlink ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic hid_generic usbhid hid sd_mod t10_pi sr_mod cdrom crc64_rocksoft_generic crc64_rocksoft crc_t10dif crct10dif_generic crc32_pclmul crct10dif_pclmul crc64 ata_generic crct10dif_common crc32c_intel ahci vmwgfx ata_piix libahci drm_ttm_helper ttm ghash_clmulni_intel drm_kms_helper ohci_pci ehci_pci sha512_ssse3 ohci_hcd libata ehci_hcd sha512_generic sha256_ssse3 scsi_mod usbcore drm sha1_ssse3 video e1000 psmouse i2c_piix4 usb_common wmi scsi_common button aesni_intel crypto_simd cryptd
[  712.556341] CPU: 0 PID: 14198 Comm: (udev-worker) Tainted: G           OE      6.8.11-amd64 #1  Kali 6.8.11-1kali2
[  712.556346] Hardware name: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[  712.556349] RIP: 0010:rtw_odm_init_ic_type+0x44/0x70 [88XXau]
[  712.556485] Code: 3e 00 00 80 fa 0e 76 17 8b 15 a4 d1 1a 00 85 d2 74 18 be 04 00 00 00 48 89 c7 e9 67 af 07 00 8b 14 95 70 2a 06 c1 85 d2 75 e8 <0f> 0b be 04 00 00 00 48 89 c7 e9 4d af 07 00 66 66 2e 0f 1f 84 00
[  712.556488] RSP: 0018:ffffb081c4b9f9d8 EFLAGS: 00010246
[  712.556493] RAX: ffffb081c0f9aed0 RBX: ffffb081c0048000 RCX: 0000000000000002
[  712.556496] RDX: 0000000000000000 RSI: ffffffffc107301b RDI: ffffb081c0048000
[  712.556499] RBP: ffff8f135b06f000 R08: 00
```

bilindows7, 2024-06-13:
"morrownr/8812au" Drivers fix Monitor mode and Deauth problems on Kali Linux 2024
Commands:

sudo apt update && sudo apt upgrade

sudo apt install -y linux-headers-$(uname -r) build-essential bc dkms git libelf-dev rfkill iw
mkdir -p ~/src

cd ~/src

git clone [https://github.com/morrownr/8812au-20...

sudo ./install-driver.sh

---

gary9872, 2024-06-13:
If you've installed the headers as blindows7 has suggested and it still doesn't work then try the manual install using the following (

CFLAGS="$CFLAGS $(INCLUDE)-D_GNU_SOURCE -Wno-error=return-mismatch -Wno-error=implicit-function-declaration -Wno-error=int-conversion -Wno-error=incompatible-pointer-types" make

(the above ALL one line)

sudo make dkms_install

Providing the dkms build logs with be helpful too.

SeiV-K, 2024-06-13:
> install Alfa AWUS036ACH (RTL8812AU) drivers on Kali Linux 2024.2 "morrownr/8812au" Drivers fix Monitor mode and Deauth problems on Kali Linux 2024 Commands:
> 
> sudo apt update && sudo apt upgrade
> 
> sudo apt install -y linux-headers-$(uname -r) build-essential bc dkms git libelf-dev rfkill iw mkdir -p ~/src
> 
> cd ~/src
> 
> git clone [https://github.com/morrownr/8812au-20...
> 
> sudo ./install-driver.sh

the setup works fine and the mod is installed, but even with the driver loaded, the adapter is not recognized at all:
[image]

SeiV-K, 2024-06-14:
> If you've installed the headers as blindows7 has suggested and it still doesn't work then try the manual install using the following:
>
> CFLAGS="$CFLAGS $(INCLUDE)-D_GNU_SOURCE [...]

compilation is successful, dkms_install is successful, but the adapter is not operational : neither on managed nor on monitor modes. It is shown on the adapters' list but it can't do anything

SeiV-K, 2024-06-14:
[image]

tried everything from https://docs.alfa.com.tw/Support/Linux/RTL8811AU/

The device appears to be loaded all good, BUT it is not operational...

[image]

gary9872, 2024-06-14:
What does your dkms status say?

bilindows7, 2024-06-14:
dkms status             

rtl8812au/5.13.6-23, 6.8.11-amd64, x86_64: installed

bilindows7, 2024-06-14:
### Troubleshooting Steps

1. Verify Driver Installation:
   Ensure the driver is correctly installed and recognized by the system.

   ```bash
   sudo dkms status
   ```

   You should see something like:
   ```
   8812au, 5.6.4.2_35491.20191025, 6.8.11-amd64, x86_64: installed
   ```

2. **Check Kernel Modules:**
   Verify that the correct kernel module is loaded.

   ```bash
   lsmod | grep 8812au
   ```

   If the module is not listed, try loading it manually:

   ```bash
   sudo modprobe 8812au
   ```

3. **Check dmesg Logs:**
   Look at the kernel log to see if there are any errors related to the adapter.

   ```bash
   dmesg | grep 8812au
   ```

   Look for any errors or messages that might indicate what the problem is.

4. **Network Manager Configuration:**
   Ensure NetworkManager is not conflicting with the adapter. Sometimes, NetworkManager might need to be restarted or configured to recognize the new adapter.

   ```bash
   sudo systemctl restart NetworkManager
   ```

5. **Monitor Mode Setup:**
   To set up monitor mode, you can use the following commands:

   ```bash
   sudo ip link set wlan0 down
   sudo iw dev wlan0 set type monitor
   sudo ip link set wlan0 up
   ```

   Replace `wlan0` with your actual wireless interface name, which can be found using:

   ```bash
   iwconfig
   ```

6. **Managed Mode Setup:**
   To set up managed mode, ensure the adapter is up and scanning for networks:

   ```bash
   sudo ip link set wlan0 down
   sudo iw dev wlan0 set type managed
   sudo ip link set wlan0 up
   sudo iwlist wlan0 scan
   ```

7. **Blacklist Conflicting Drivers:**
   Sometimes, other wireless drivers might conflict with the 8812au driver. Blacklist these drivers if necessary.

   Create or edit the blacklist file:

   ```bash
   sudo nano /etc/modprobe.d/blacklist.conf
   ```

   Add the following lines:

   ```bash
   blacklist rtl8xxxu
   blacklist r8188eu
   ```

   Save and close the file, then update the initial ramdisk:

   ```bash
   sudo update-initramfs -u
   ```

8. **Reboot the System:**
   After performing these steps, reboot the system to ensure all changes take effect.

   ```bash
   sudo reboot
   ```

### Advanced Debugging

If the adapter still does not work, consider these additional steps:

- **Update the Kernel:** Sometimes, updating the kernel to the latest version can resolve compatibility issues.
  
- **Use a Different Driver Repository:** Some users have success with alternative repositories for the same chipset. For example:

  ```bash
  git clone https://github.com/gnab/rtl8812au.git
  cd rtl8812au
  sudo make clean
  sudo make
  sudo make dkms_install
  ```

If you've followed all these steps and the adapter is still not functioning, please provide any error messages or logs from `dmesg` and `iwconfig`, and I'll help you further troubleshoot the issue.

bilindows7, 2024-06-14:
To uninstall all Wi-Fi drivers, including the drivers installed through DKMS, and then reinstall the correct driver for your ALFA AWUS036ACU (AC1200), follow these steps:

### Uninstall Existing Wi-Fi Drivers

1. **List DKMS Modules:**
   Check the installed DKMS modules to identify which Wi-Fi drivers are installed.

   ```bash
   sudo dkms status
   ```

2. **Remove DKMS Modules:**
   For each Wi-Fi driver listed (e.g., 8812au, 8821au), remove it. Replace `module_name` and `module_version` with the appropriate values from the `dkms status` output.

   ```bash
   sudo dkms remove module_name/module_version --all
   ```

   Example for 8812au:
   ```bash
   sudo dkms remove 8812au/5.6.4.2_35491.20191025 --all
   ```

   Example for 8821au:
   ```bash
   sudo dkms remove 8821au/5.6.4.2_35491.20191025 --all
   ```

3. **Remove Module Files:**
   Remove any residual module files from `/usr/src`.

   ```bash
   sudo rm -rf /usr/src/8812au-*
   sudo rm -rf /usr/src/8821au-*
   ```

4. **Blacklist Conflicting Drivers:**
   Ensure conflicting drivers are blacklisted.

   ```bash
   sudo nano /etc/modprobe.d/blacklist.conf
   ```

   Add the following lines to blacklist other drivers that might interfere:

   ```bash
   blacklist rtl8xxxu
   blacklist r8188eu
   ```

   Save and close the file, then update the initial ramdisk:

   ```bash
   sudo update-initramfs -u
   ```

5. **Reboot the System:**
   Reboot the system to apply changes.

   ```bash
   sudo reboot
   ```

### Install the Correct Driver for ALFA AWUS036ACU (AC1200)

1. **Prepare the System:**
   - Ensure your system is up-to-date.
   ```bash
   sudo apt-get update
   sudo apt-get upgrade
   ```

2. **Install Dependencies:**
   - Install necessary packages for building the driver.
   ```bash
   sudo apt-get install build-essential dkms git
   ```

3. **Download the Correct Driver:**
   - Clone the driver repository from GitHub for the RTL8821AU chipset.
   ```bash
   git clone https://github.com/morrownr/8821au-20210708
   cd 8821au-20210708
   ```

4. **Build and Install the Driver:**
   - Compile and install the driver using DKMS.
   ```bash
   sudo ./install-driver.sh
   ```

5. **Reboot the System:**
   - Reboot your system to load the new driver.
   ```bash
   sudo reboot
   ```

### Verification Steps

1. **Verify Driver Installation:**
   Ensure the driver is correctly installed and recognized by the system.

   ```bash
   sudo dkms status
   ```

   You should see something like:
   ```
   8821au, 5.6.4.2_35491.20191025, 6.8.11-amd64, x86_64: installed
   ```

2. **Check Kernel Modules:**
   Verify that the correct kernel module is loaded.

   ```bash
   lsmod | grep 8821au
   ```

   If the module is not listed, try loading it manually:

   ```bash
   sudo modprobe 8821au
   ```

3. **Check dmesg Logs:**
   Look at the kernel log to see if there are any errors related to the adapter.

   ```bash
   dmesg | grep 8821au
   ```

   Look for any errors or messages that might indicate what the problem is.

4. **Network Manager Configuration:**
   Ensure NetworkManager is not conflicting with the adapter. Sometimes, NetworkManager might need to be restarted or configured to recognize the new adapter.

   ```bash
   sudo systemctl restart NetworkManager
   ```

5. **Monitor Mode Setup:**
   To set up monitor mode, you can use the following commands:

   ```bash
   sudo ip link set wlan0 down
   sudo iw dev wlan0 set type monitor
   sudo ip link set wlan0 up
   ```

   Replace `wlan0` with your actual wireless interface name, which can be found using:

   ```bash
   iwconfig
   ```

6. **Managed Mode Setup:**
   To set up managed mode, ensure the adapter is up and scanning for networks:

   ```bash
   sudo ip link set wlan0 down
   sudo iw dev wlan0 set type managed
   sudo ip link set wlan0 up
   sudo iwlist wlan0 scan
   ```

7. **Reboot the System:**
   After performing these steps, reboot the system to ensure all changes take effect.

   ```bash
   sudo reboot
   ```

By following these updated steps with the correct driver, your ALFA AWUS036ACU (AC1200) adapter should be operational in both managed and monitor modes. If you encounter further issues, please provide any error messages or logs from `dmesg` and `iwconfig`, and I'll assist you accordingly.

bilindows7, 2024-06-14:
Based on the output from `dkms status`, it looks like you have the `rtl8812au` module installed with version `5.13.6-23`. Let's proceed with the uninstallation and then reinstall the correct driver for the RTL8821AU chipset.

### Uninstall the RTL8812AU Driver

1. **Remove the RTL8812AU DKMS Module:**
   ```bash
   sudo dkms remove rtl8812au/5.13.6-23 --all
   ```

2. **Remove Module Files:**
   ```bash
   sudo rm -rf /usr/src/rtl8812au-5.13.6-23
   ```

3. **Unload Kernel Modules:**
   Ensure the driver module is not loaded. If it is loaded, unload it.

   ```bash
   lsmod | grep 8812au
   sudo rmmod 8812au
   ```

4. **Blacklist Conflicting Drivers:**
   Edit or create the blacklist configuration file to ensure conflicting drivers are blacklisted.

   ```bash
   sudo nano /etc/modprobe.d/blacklist.conf
   ```

   Add the following lines if they are not already present:

   ```bash
   blacklist rtl8xxxu
   blacklist r8188eu
   ```

   Save and close the file, then update the initial ramdisk:

   ```bash
   sudo update-initramfs -u
   ```

5. **Reboot the System:**
   Reboot the system to ensure all changes take effect.

   ```bash
   sudo reboot
   ```

### Install the Correct Driver for ALFA AWUS036ACU (AC1200)

1. **Prepare the System:**
   - Ensure your system is up-to-date.
   ```bash
   sudo apt-get update
   sudo apt-get upgrade
   ```

2. **Install Dependencies:**
   - Install necessary packages for building the driver.
   ```bash
   sudo apt-get install build-essential dkms git
   ```

3. **Download the Correct Driver:**
   - Clone the driver repository from GitHub for the RTL8821AU chipset.
   ```bash
   git clone https://github.com/morrownr/8821au-20210708
   cd 8821au-20210708
   ```

4. **Build and Install the Driver:**
   - Compile and install the driver using DKMS.
   ```bash
   sudo ./install-driver.sh
   ```

5. **Reboot the System:**
   - Reboot your system to load the new driver.
   ```bash
   sudo reboot
   ```

### Verification Steps

1. **Verify Driver Installation:**
   Ensure the driver is correctly installed and recognized by the system.

   ```bash
   sudo dkms status
   ```

   You should see something like:
   ```
   8821au, 5.6.4.2_35491.20191025, 6.8.11-amd64, x86_64: installed
   ```

2. **Check Kernel Modules:**
   Verify that the correct kernel module is loaded.

   ```bash
   lsmod | grep 8821au
   ```

   If the module is not listed, try loading it manually:

   ```bash
   sudo modprobe 8821au
   ```

3. **Check dmesg Logs:**
   Look at the kernel log to see if there are any errors related to the adapter.

   ```bash
   dmesg | grep 8821au
   ```

4. **Network Manager Configuration:**
   Ensure NetworkManager is not conflicting with the adapter. Sometimes, NetworkManager might need to be restarted or configured to recognize the new adapter.

   ```bash
   sudo systemctl restart NetworkManager
   ```

5. **Monitor Mode Setup:**
   To set up monitor mode, you can use the following commands:

   ```bash
   sudo ip link set wlan0 down
   sudo iw dev wlan0 set type monitor
   sudo ip link set wlan0 up
   ```

   Replace `wlan0` with your actual wireless interface name, which can be found using:

   ```bash
   iwconfig
   ```

6. **Managed Mode Setup:**
   To set up managed mode, ensure the adapter is up and scanning for networks:

   ```bash
   sudo ip link set wlan0 down
   sudo iw dev wlan0 set type managed
   sudo ip link set wlan0 up
   sudo iwlist wlan0 scan
   ```

7. **Reboot the System:**
   After performing these steps, reboot the system to ensure all changes take effect.

   ```bash
   sudo reboot
   ```

By following these updated steps, your ALFA AWUS036ACU (AC1200) adapter should be operational in both managed and monitor modes. If you encounter further issues, please provide any error messages or logs from `dmesg` and `iwconfig`, and I'll assist you accordingly.

SeiV-K, 2024-06-15:
@bilindows7 thank you very much for the detailed support! I appreciate it!

The problem ended up being the usage of a compatible, yet not functional for this specific device, driver. 

After installing 8821au everything is as smooth as you describe:
[image]

I am closing this issue with a huge thanks to all guys for the support! Great community!

closed 2024-06-15 by SeiV-K

ka3boosh, 2024-07-03:
hi i have the same issue i have 
`Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter`
dkms status
`broadcom-sta/6.30.223.271, 6.8.11-amd64, x86_64: installed
realtek-rtl8814au/5.8.5.1~git20240527.d8208c8, 6.8.11-amd64, x86_64: installed
realtek-rtl88xxau/5.6.4.2~git20240510.b44d288, 6.8.11-amd64, x86_64: built
rtl8812au/5.13.6-23, 6.8.11-amd64, x86_64: installed
rtl8821au/5.12.5.2, 6.8.11-amd64, x86_64: installed`

lsmod | grep 8812au
`8812au               2613248  0
cfg80211             1355776  2 8812au,88XXau
usbcore               409600  7 ehci_pci,usbhid,ehci_hcd,8812au,btusb,88XXau,uhci_hcd
`
and i keep getting this error when i try to install fresh drivers
`┌──(zee㉿kali)-[~/Downloads/rtl8812au-5.6.4.2]
└─$ make && make install
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.8.11-amd64/build M=/home/zee/Downloads/rtl8812au-5.6.4.2  modules
make[1]: Entering directory '/usr/src/linux-headers-6.8.11-amd64'
make[1]: Leaving directory '/usr/src/linux-headers-6.8.11-amd64'
---------------------------------------------------------------------------
Visit https://github.com/aircrack-ng/rtl8812au for support/reporting issues
or check for newer versions (branches) of these drivers.                   
---------------------------------------------------------------------------
install -p -m 644 88XXau.ko  /lib/modules/6.8.11-amd64/kernel/drivers/net/wireless/
install: cannot remove '/lib/modules/6.8.11-amd64/kernel/drivers/net/wireless/88XXau.ko': Permission denied
make: *** [Makefile:1739: install] Error 1
`
the result is like this monitor mod might show but not working 
`──(zee㉿kali)-[~]
└─$ iwconfig
lo        no wireless extensions.

eth0      no wireless extensions.

wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
`
`sudo airmon-ng start wlan0
[sudo] password for zee: 
lo        no wireless extensions.

eth0      no wireless extensions.

wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

Killing these processes:

    PID Name
   1135 wpa_supplicant

PHY     Interface       Driver          Chipset

phy0    wlan0           8812au          Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
                (monitor mode enabled)`

`┌──(zee㉿kali)-[~]
└─$ sudo wifite  
   .               .    
 .´  ·  .     .  ·  `.  wifite2 2.7.0
 :  :  :  (¯)  :  :  :  a wireless auditor by derv82
 `.  ·  ` /¯\ ´  ·  .´  maintained by kimocoder
   `     /¯¯¯\     ´    https://github.com/kimocoder/wifite2

    Interface   PHY   Driver              Chipset                                                 
-----------------------------------------------------------------------
 1. wlan0       phy0  8812au              Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter

 [+] Enabling monitor mode on wlan0... failed

 [!] Error: Cannot find any interfaces in monitor mode
                                                                                                  
 [!] Full stack trace below                                                                       
                                                                                                  
 [!]    Traceback (most recent call last):                                                        
 [!]    File "/usr/lib/python3/dist-packages/wifite/__main__.py", line 104, in entry_point
 [!]        wifite.start()
 [!]    File "/usr/lib/python3/dist-packages/wifite/__main__.py", line 57, in start               
 [!]        Configuration.get_monitor_mode_interface()
 [!]    File "/usr/lib/python3/dist-packages/wifite/config.py", line 229, in get_monitor_mode_interface
 [!]        cls.interface = Airmon.ask()
 [!]                        ^^^^^^^^^^^^                                                          
 [!]    File "/usr/lib/python3/dist-packages/wifite/tools/airmon.py", line 333, in ask            
 [!]        selected.interface = Airmon.start(selected)
 [!]                             ^^^^^^^^^^^^^^^^^^^^^^                                           
 [!]    File "/usr/lib/python3/dist-packages/wifite/tools/airmon.py", line 213, in start          
 [!]        raise Exception('Cannot find any interfaces in monitor mode')
 [!]  Exception: Cannot find any interfaces in monitor mode                                       
                                                                                                  
 [!] Exiting `

there is an issue i cant fix it please help me my WIFI card was working before i update kali linux now i cant get it to work no matter what i do 
im using kali linux 2024.1  and 2024.2  not working at both and i keep getting the same erros

ka3boosh, 2024-07-16:
any help ??? the steps you offered is not working ive done it over 20 times need to be updated

Clawzman, 2024-07-16:
> To uninstall all Wi-Fi drivers, including the drivers installed through DKMS, and then reinstall the correct driver for your ALFA AWUS036ACU (AC1200), follow these steps:
> 
> ### Uninstall Existing Wi-Fi Drivers
> 1. **List DKMS Modules:**
>    Check the installed DKMS modules to identify which Wi-Fi drivers are installed.
>    ```shell
>    sudo dkms status
>    ```
> 2. **Remove DKMS Modules:**
>    For each Wi-Fi driver listed (e.g., 8812au, 8821au), remove it. Replace `module_name` and `module_version` with the appropriate values from the `dkms status` output.
>    ```shell
>    sudo dkms remove module_name/module_version --all
>    ```
>    
>    
>        
>          
>        
>    
>          
>        
>    
>        
>      
>    Example for 8812au:
>    ```shell
>    sudo dkms remove 8812au/5.6.4.2_35491.20191025 --all
>    ```
>    
>    
>        
>          
>        
>    
>          
>        
>    
>        
>      
>    Example for 8821au:
>    ```shell
>    sudo dkms remove 8821au/5.6.4.2_35491.20191025 --all
>    ```
> 3. **Remove Module Files:**
>    Remove any residual module files from `/usr/src`.
>    ```shell
>    sudo rm -rf /usr/src/8812au-*
>    sudo rm -rf /usr/src/8821au-*
>    ```
> 4. **Blacklist Conflicting Drivers:**
>    Ensure conflicting drivers are blacklisted.
>    ```shell
>    sudo nano /etc/modprobe.d/blacklist.conf
>    ```
>    
>    
>        
>          
>        
>    
>          
>        
>    
>        
>      
>    Add the following lines to blacklist other drivers that might interfere:
>    ```shell
>    blacklist rtl8xxxu
>    blacklist r8188eu
>    ```
>    
>    
>        
>          
>        
>    
>          
>        
>    
>        
>      
>    Save and close the file, then update the initial ramdisk:
>    ```shell
>    sudo update-initramfs -u
>    ```
> 5. **Reboot the System:**
>    Reboot the system to apply changes.
>    ```shell
>    sudo reboot
>    ```
> 
> ### Install the Correct Driver for ALFA AWUS036ACU (AC1200)
> 1. **Prepare the System:**
>    
>    * Ensure your system is up-to-date.
>    
>    ```shell
>    sudo apt-get update
>    sudo apt-get upgrade
>    ```
> 2. **Install Dependencies:**
>    
>    * Install necessary packages for building the driver.
>    
>    ```shell
>    sudo apt-get install build-essential dkms git
>    ```
> 3. **Download the Correct Driver:**
>    
>    * Clone the driver repository from GitHub for the RTL8821AU chipset.
>    
>    ```shell
>    git clone https://github.com/morrownr/8821au-20210708
>    cd 8821au-20210708
>    ```
> 4. **Build and Install the Driver:**
>    
>    * Compile and install the driver using DKMS.
>    
>    ```shell
>    sudo ./install-driver.sh
>    ```
> 5. **Reboot the System:**
>    
>    * Reboot your system to load the new driver.
>    
>    ```shell
>    sudo reboot
>    ```
> 
> ### Verification Steps
> 1. **Verify Driver Installation:**
>    Ensure the driver is correctly installed and recognized by the system.
>    ```shell
>    sudo dkms status
>    ```
>    
>    
>        
>          
>        
>    
>          
>        
>    
>        
>      
>    You should see something like:
>    ```
>    8821au, 5.6.4.2_35491.20191025, 6.8.11-amd64, x86_64: installed
>    ```
> 2. **Check Kernel Modules:**
>    Verify that the correct kernel module is loaded.
>    ```shell
>    lsmod | grep 8821au
>    ```
>    
>    
>        
>          
>        
>    
>          
>        
>    
>        
>      
>    If the module is not listed, try loading it manually:
>    ```shell
>    sudo modprobe 8821au
>    ```
> 3. **Check dmesg Logs:**
>    Look at the kernel log to see if there are any errors related to the adapter.
>    ```shell
>    dmesg | grep 8821au
>    ```
>    
>    
>        
>          
>        
>    
>          
>        
>    
>        
>      
>    Look for any errors or messages that might indicate what the problem is.
> 4. **Network Manager Configuration:**
>    Ensure NetworkManager is not conflicting with the adapter. Sometimes, NetworkManager might need to be restarted or configured to recognize the new adapter.
>    ```shell
>    sudo systemctl restart NetworkManager
>    ```
> 5. **Monitor Mode Setup:**
>    To set up monitor mode, you can use the following commands:
>    ```shell
>    sudo ip link set wlan0 down
>    sudo iw dev wlan0 set type monitor
>    sudo ip link set wlan0 up
>    ```
>    
>    
>        
>          
>        
>    
>          
>        
>    
>        
>      
>    Replace `wlan0` with your actual wireless interface name, which can be found using:
>    ```shell
>    iwconfig
>    ```
> 6. **Managed Mode Setup:**
>    To set up managed mode, ensure the adapter is up and scanning for networks:
>    ```shell
>    sudo ip link set wlan0 down
>    sudo iw dev wlan0 set type managed
>    sudo ip link set wlan0 up
>    sudo iwlist wlan0 scan
>    ```
> 7. **Reboot the System:**
>    After performing these steps, reboot the system to ensure all changes take effect.
>    ```shell
>    sudo reboot
>    ```
> 
> By following these updated steps with the correct driver, your ALFA AWUS036ACU (AC1200) adapter should be operational in both managed and monitor modes. If you encounter further issues, please provide any error messages or logs from `dmesg` and `iwconfig`, and I'll assist you accordingly.

I want to thank so much because had exactly same issue and tried everything. With no luck. 
Your tutorial worked perfectly! Cheers!
And thanks to the thread starter.

ka3boosh, 2024-07-17:
here is my result i did every step 

`┌──(root㉿kali)-[~]
└─# lsusb               
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 004: ID 0e0f:0008 VMware, Inc. Virtual Bluetooth Adapter
                                                                                             
┌──(root㉿kali)-[~]
└─# dkms status
rtl8821au/5.12.5.2, 6.8.11-amd64, x86_64: installed
                                                                                             
┌──(root㉿kali)-[~]
└─# lsmod | grep 8821au
                                                                                             
┌──(root㉿kali)-[~]
└─# sudo modprobe 8821au
                                                                                             
┌──(root㉿kali)-[~]
└─# dmesg | grep 8821au
[  216.240498] 8821au: loading out-of-tree module taints kernel.
[  216.240523] 8821au: module verification failed: signature and/or required key missing - tainting kernel
[  216.372761] usbcore: registered new interface driver rtl8821au
                                                                                             
┌──(root㉿kali)-[~]
└─# sudo systemctl restart NetworkManager
                                                                                                         
┌──(root㉿kali)-[~]
└─# sudo ip link set wlan0 down
Cannot find device "wlan0"
`

fraveroza4, 2024-07-18:
> here is the fix https://www.youtube.com/watch?v=Yr-8RmoNi70&t=40s
> 
> i made a video

I've spent a few days on this, and never refreshed this page.  Once I finally posted, I saw your update.

WATCH ka3boosh's VIDEO!!!!!!!!!

It worked perfectly!!!  Thank you for taking the time for posting your fix and your video.  I would thumbs-up your https://www.youtube.com/watch?v=Yr-8RmoNi70&t=40s, but I never sign in there.

ka3boosh, 2024-07-18:
> I've spent a few days on this, and never refreshed this page. Once I finally posted, I saw your update.
>
> WATCH ka3boosh's VIDEO!!!!!!!!!
>
> It worked perfectly!!! Thank you for taking the time for [...]

I'm glad it fix the problem for u my video kinda suck but it do the job lol

Adamkadaban, 2026-09-03:
For anyone on **kernel 6.12.x** with the **AWUS036ACS** (RTL8811AU, USB ID `0bda:0811`):

The latest `main` of [morrownr/8821au-20210708](https://github.com/morrownr/8821au-20210708) includes kernel 7.0+ patches that break the build on 6.12. To fix, checkout commit `cf03ca2` (the 6.12.101 fix) before the 7.0 changes:

```bash
git clone https://github.com/morrownr/8821au-20210708.git
cd 8821au-20210708
git checkout cf03ca2
sudo ./install-driver.sh
```

This builds and works on `6.12.107+deb13-amd64`.