aircrack-ng/rtl8812au #1165: Error with latest version of Kali

Aggelos11, 2024-06-14:
Hi! I just upgraded my Kali to 6.8 . I tried both the GitHub drivers and the apt realtek-eel88xxau-dkms. Even though they are successfully installed the adaptor even though it is recognised by the machine it is not operational at all. He cannot see/connect to nearby WiFi networks in neither monitor nor managed mode … please help !
[image]

Aggelos11, 2024-06-14:
Ok so I found a solution to anyone who still struggles… so the commands to install it and make it work are :
sudo apt-get update
sudo apt get upgrade -y
sudo apt-get install bc mokutil build-essential libelf-dev linux-headers-`uname -r` 
sudo git clone …( GitHub link )
cd rtl*

sudo make && sudo make install 

sudo git checkout 63cf0b4
sudo make && sudo make install

SeiV-K, 2024-06-15:
glad that you found the solution. feel free to close the issue if it no longer exists :)

closed 2024-06-15 by Aggelos11

cr7pt0pl4gu3, 2024-06-20:
@SeiV-K @Aggelos11 I don't think it's a good idea to close this issue. I just wasted 6 hours trying to understand why my Alfa is not working, and apparently it's because of the driver version, the latest one is clearly broken for my card

jaykieth73, 2024-06-22:
@Aggelos11 Fantastic! Worked like a charm after spending countless hours trying to fix the broken update. Thank you very much for commenting the fix.

 For anyone who may not know, the install line for should be typed as: 

`sudo apt-get install bc mokutil build-essential libelf-dev linux-headers-$(uname -r)`

focabigd, 2024-08-03:
guys,i've tried what u've just said,but it still isn't working for me,same problem

Aggelos11, 2024-08-03:
> guys,i've tried what u've just said,but it still isn't working for me,same problem

Which adaptor do you use ?

focabigd, 2024-08-04:
awus036ach

Aggelos11, 2024-08-04:
Hm you must be doing something wrong then . Read the instructions I wrote again

focabigd, 2024-08-04:
i just found the solution,use an older driver

Aggelos11, 2024-08-04:
This is exactly what gut checkout does my friend

focabigd, 2024-08-04:
oh,i'm new into linux