aircrack-ng/rtl8812au #1169: RTL8812AU not working after kernel upgrade

DestructIT, 2024-06-24:
After upgrading machine I cant get the RTL8812AU adapter to work anymore.
The adapter is visible but not seeing any networks.

wlan0 RTL8821AE (built-in wifi)
wlan1 RTL8812AU (USB)

**uname -av**
Linux nix 6.8.11-amd64 #1 SMP PREEMPT_DYNAMIC Kali 6.8.11-1kali2 (2024-05-30) x86_64 GNU/Linux

**lsusb**
Bus 003 Device 004: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter

**lsb_release -a**
No LSB modules are available.
Distributor ID: Kali
Description:    Kali GNU/Linux Rolling
Release:        2024.2
Codename:       kali-rolling

**iwconfig wlan1**   
wlan1     unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

dkms status
8812au/5.6.4.2_35491.20191025, 6.8.11-amd64, x86_64: installed

**dmesg | grep "usb 3-1"**
[  148.639812] usb 3-1: new high-speed USB device number 4 using xhci_hcd
[  148.788256] usb 3-1: New USB device found, idVendor=0bda, idProduct=8812, bcdDevice= 0.00
[  148.788277] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  148.788284] usb 3-1: Product: 802.11n NIC
[  148.788289] usb 3-1: Manufacturer: Realtek
[  148.788293] usb 3-1: SerialNumber: 123456
[  149.116615] 88XXau: loading out-of-tree module taints kernel.
[  149.116690] 88XXau: module verification failed: signature and/or required key missing - tainting kernel
[  149.370524] usb 3-1: 88XXau 00:c0:ca:b2:7d:e8 hw_info[d7]
[  149.372226] usbcore: registered new interface driver rtl88XXau

**lshw -c network**
  *-network                 
       description: Wireless interface
       product: RTL8821AE 802.11ac PCIe Wireless Network Adapter
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0000:02:00.0
       logical name: wlan0 <-------------- BUILT-IN RTL8821AE (Works fine)
       version: 00
       serial: 16:a3:c6:ba:09:f3
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress bus_master cap_list ethernet physical wireless
       configuration: broadcast=yes driver=rtl8821ae driverversion=6.8.11-amd64 firmware=N/A latency=0 link=no multicast=yes wireless=IEEE 802.11
       resources: irq:30 ioport:e000(size=256) memory:f7d00000-f7d03fff
  *-network DISABLED
       description: Wireless interface
       physical id: 6
       bus info: usb@3:1
       logical name: wlan1 <------------------- PROBLEM DEVICE RTL8812AU
       serial: 00:c0:ca:b2:7d:e8
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtl88XXau driverversion=6.8.11-amd64 multicast=yes wireless=unassociated

any solution?

Redditor247, 2024-06-26:
Currently experiencing the same issue on Kali. Attempted to roll back the kernel as a solution but didn't work, due to linux header files not being pulled directly.

I've tried some other drivers for this, but they do not support monitor mode. They'll detect networks perfectly fine as a managed interface but beyond that won't work.

michaelvglr, 2024-06-27:
Hi you two, had the same issue, was able to fix it using instructions from a few issues below ("USB Wifi TP-Link AC600 cannot work with newest version").

amnuhs, 2024-06-29:
me too, cannot find network connection and no led blinking -  alfa awus036ach

DestructIT, 2024-06-30:
> Hi you two, had the same issue, was able to fix it using instructions from a few issues below ("USB Wifi TP-Link AC600 cannot work with newest version").

I tried that as well, didn't work for me.

I also noticed that the adapter wont show up with **lsusb** after a reboot, I need to remove and connect it to usb again to make it vissible.

It still doesnt work, I just bought a new one (Alfa AWUS036AXML with a Mediatek MT7921AUN in it)

Ademir18, 2024-07-01:
The same thing happens to me, it doesn't work, it doesn't capture networks and it doesn't inject packets either, in my case I have Windows 11 64bit, virtual box 7 and kali linux 2024.2, AWUS036ACH antenna, I'm doing something wrong

chinawrj, 2024-07-02:
Use older version will work, such as 63cf0b4584aa8878b0fe8ab38017f31c319bde3d
[image]

I suspect that the following commit introduces the issue, but I still have no time to verify it.
[image]

LU5GOON, 2024-07-02:
> Use older version will work, such as [63cf0b4](https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d) <img alt="image" width="877" src="https://private-user-images.githubusercontent.com/1029397/344849610-b05d0ec2-1140-4a0b-ad5b-908b35aa45b1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk4OTcwMTMsIm5iZiI6MTcxOTg5NjcxMywicGF0aCI6Ii8xMDI5Mzk3LzM0NDg0OTYxMC1iMDVkMGVjMi0xMTQwLTRhMGItYWQ1Yi05MDhiMzVhYTQ1YjEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcwMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MDJUMDUwNTEzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZTZkZTdjMzZkMmUwNzExNjM0MTZlZGU1OTkyMWZjZjIzNzg3ZTdkNzhiOTY5ZDI3ZDZhMjBhMzkwYjJjNDhjNiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.vM52i03y-CX2AdStFLKDFDyw0Tj1hKBsfo70KXZSm0U">
> 
> I suspect that the following commit introduces the issue, but I still have no time to verify it. <img alt="image" width="802" src="https://private-user-images.githubusercontent.com/1029397/344849765-611e3b65-ebd4-4a9c-9a27-0c36e0682b96.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk4OTcwMTMsIm5iZiI6MTcxOTg5NjcxMywicGF0aCI6Ii8xMDI5Mzk3LzM0NDg0OTc2NS02MTFlM2I2NS1lYmQ0LTRhOWMtOWEyNy0wYzM2ZTA2ODJiOTYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcwMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MDJUMDUwNTEzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZTlhYmYwN2FkM2M0OTJkYzA2M2QwNjNhMTdjMzE2NDNkOTVjZTI3NDVhNDdlZjc0ZDVlMTZkMmVmNjJkODBhZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.T-LbJBN7cshUVWJn2C1lKE_W25qm_r3Qfu8UJvwJtHY">

Apologies for the nooby question, I am not experienced that much with GitHub
How can i achieve this edit to go back to an old working earlier version?

Thank you.

chinawrj, 2024-07-02:
> How can i achieve this edit to go back to an old working earlier version?

Download ZIP with this link: https://github.com/aircrack-ng/rtl8812au/tree/b44d288f423ede0fc7cdbf92d07a7772cd727de4

chinawrj, 2024-07-02:
I revert the patch which introduces the issue. You guys may have a try https://github.com/chinawrj/rtl8812au

LU5GOON, 2024-07-02:
Hey @DestructIT & @chinawrj 

Thank you for all the comms on this post. Very helpful to read through as i had the exact same issue and bought a newer alfa card because of it also (model ending in AXML on MediaTek chipset)

@chinawrj  Thank you so much! I can confirm your branched repo has worked for me and now my RTL8812AU is now working perfectly as it was.

Thanks so much for your workaround here, glad to be up and running again finally.

[image]

yusrahsoeker, 2024-07-02:
> me too, cannot find network connection and no led blinking - alfa awus036ach

I managed to get my Alfa AWUS036ACH working by reverting back to [63cf0b4](https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d) and by using steps from [here](https://github.com/aircrack-ng/rtl8812au/issues/1159#top). 

For some background, I am using Virtual Box and am running Kali Linux 2024.2. And these are my USB settings.
[image].

I had all headers up to date already so I didn't include those steps below. This is what finally worked for me: 

1. `dkms status` to find the name of modules installed
2. removed the driver and modules using `sudo dkms remove <module name>`
3. deleted the rtl8812au folder and its contents using `rm -r <folder name>`
4. `reboot now`
5. `git clone https://github.com/aircrack-ng/rtl8812au` to redownload the files
6. `cd rt*` to navigate to the folder where the files have been downloaded
7. `git checkout 63cf0b4` 
8. `git status `
9. `sudo make`
10. `sudo make install`
11. `systemctl restart NetworkManager`
12. plug in device

LU5GOON, 2024-07-02:
> > me too, cannot find network connection and no led blinking - alfa awus036ach
> 
> I managed to get my Alfa AWUS036ACH working by reverting back to [63cf0b4](https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d) and by using steps from [here](https://github.com/aircrack-ng/rtl8812au/issues/1159#top).
> 
> For some background, I am using Virtual Box running Kali Linux 2024.2, and these are my USB settings. ![USB settings](https://private-user-images.githubusercontent.com/40632654/344988787-a9d5b15e-d85e-4ea6-b64f-cee27cf5a20d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk5MTcyNzAsIm5iZiI6MTcxOTkxNjk3MCwicGF0aCI6Ii80MDYzMjY1NC8zNDQ5ODg3ODctYTlkNWIxNWUtZDg1ZS00ZWE2LWI2NGYtY2VlMjdjZjVhMjBkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzAyVDEwNDI1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI2ZDZiNDFiZjk2YTMwMmU2MDE4MmIxN2QwMzVlN2ZjYmM1MWZkNGQ4YzFlMDVkYjdlNDIyOTQ0MmI5MmViZDEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.nPWRIGNih3zRXrb6aWg933ZVqvQ05qMJko7H80DiKnk).
> 
> I had all headers up to date already so I didn't include those steps below. This is what finally worked for me:
> 
> 1. `dkms status` to find the name of modules installed
> 2. removed the driver and modules using `sudo dkms remove <module name>`
> 3. deleted the rtl8812au folder and its contents using `rm -r <folder name>`
> 4. `reboot now`
> 5. `git clone https://github.com/aircrack-ng/rtl8812au` to redownload the files
> 6. `cd rt*` to navigate to the folder where the files have been downloaded
> 7. `git checkout 63cf0b4`
> 8. `git status `
> 9. `sudo make`
> 10. `sudo make install`
> 11. `systemctl restart NetworkManager`
> 12. plug in device

Nice man. I too got my AWUS036ACH working by basic installation via the branched repo here. Steps are same as the original install readme. I didn't do any of your steps above but nice to know it works that way.  -> https://github.com/chinawrj/rtl8812au

It was a matter of time until workarounds were found.

DestructIT, 2024-07-02:
> I managed to get my Alfa AWUS036ACH working by reverting back to [63cf0b4](https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d) and by using steps from [...]

Thank you all, I used this solution and that seems to work fine.
I also bought the AWUS036AXML with the MT7921AUN chipset and that works also

Ademir18, 2024-07-03:
> I managed to get my Alfa AWUS036ACH working by reverting back to [63cf0b4](https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d) and by using steps from [...]

Thank you very much, I tried this method, and the magic was done, it is working, thank you so much. I'm going to test it

LU5GOON, 2024-07-03:
> Thank you all, I used this solution and that seems to work fine. I also bought the AWUS036AXML with the MT7921AUN chipset and that works also

Hey man, you are most welcome. 

Looks like we both did the same thing, i too bought the AWUS036AXML as the MediaTek chipset drivers are actually built-in to the Kali Linux kernel. You would of noticed it's plug and play without driver installations.

A great move and a good choice.

Glad all is working for you now.

DestructIT, 2024-07-03:
> Hey man, you are most welcome.
>
> Looks like we both did the same thing, i too bought the AWUS036AXML as the MediaTek chipset drivers are actually built-in to the Kali Linux kernel. You would of [...]

AWUS036AXML also supports vip and works fine with airgeddon as well

Agentmagnet, 2024-07-07:
I am using an Edimax card -  and the above method to "checkout" worked for me. Still I am seeing some delay in the packet capture process while running Airodump-ng. 

But atleast things have started working now.

apras78, 2024-08-13:
The steps listed by DestructIT worked for me. Thank you!

ret2src, 2025-01-31:
> > me too, cannot find network connection and no led blinking - alfa awus036ach
> 
> I managed to get my Alfa AWUS036ACH working by reverting back to [63cf0b4](https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d) and by using steps from [here](https://github.com/aircrack-ng/rtl8812au/issues/1159#top).
> 
> For some background, I am using Virtual Box running Kali Linux 2024.2, and these are my USB settings. [image].
> 
> I had all headers up to date already so I didn't include those steps below. This is what finally worked for me:
> 
>     1. `dkms status` to find the name of modules installed
> 
>     2. removed the driver and modules using `sudo dkms remove <module name>`
> 
>     3. deleted the rtl8812au folder and its contents using `rm -r <folder name>`
> 
>     4. `reboot now`
> 
>     5. `git clone https://github.com/aircrack-ng/rtl8812au` to redownload the files
> 
>     6. `cd rt*` to navigate to the folder where the files have been downloaded
> 
>     7. `git checkout 63cf0b4`
> 
>     8. `git status `
> 
>     9. `sudo make`
> 
>     10. `sudo make install`
> 
>     11. `systemctl restart NetworkManager`
> 
>     12. plug in device

I can confirm that this works on Ubuntu 24.04 with RTL8812AU.

If you run into problems don't forget to run both `sudo make uninstall` **AND** `sudo make dkms_remove`.

For me, the problem was that I tried different drivers using both the normal as well as the DKMS installation methods.
Hence, the DKMS-installed driver was seemingly still active and prevented the downgraded driver from working.

Finally, don't forget to reboot and (potentially) reconnect your adapter.

hydrathorn8, 2025-03-21:
> I revert the patch which introduces the issue. You guys may have a try https://github.com/chinawrj/rtl8812au

Hey @chinawrj same problem here and you really did us all a solid with this. Thanks man!