aircrack-ng/rtl8812au #1172: airodum-ng (and monitor mode ?) not working with latest commit

solsticedhiver, 2024-07-03:
I am using the *rtl8812au* **git** driver for a while on **rpi0 2 w** with usb key (0bda:0811), on **archlinux arm (armv7h)**. It was fine.

In fact, I am manually building the driver at each kernel upgrade.
This time, I run `git pull` before compiling the driver, and once I installed it, then put the card in monitor mode, I couldn't see a thing in *airodump-ng*.

No AP was detected, and the channel sweeping was very slow. May be it was working but only veeerry slloow?

I just run `git checkout 63cf0b4584aa8878b0fe8ab38017f31c319bde3d`, and rebuilt the driver.
This time, I got no problem when running *airodumg-ng*!!?

danbinns-ss, 2024-07-04:
Thanks for posting this, was pulling my hair out. Can confirm that going back to this commit allowed the driver to work correctly and start detecting APs.

nishitaggarwal, 2024-07-11:
can you help me solve the issue.?
 i didn't quite understood the solution to the problem

danbinns-ss, 2024-07-11:
@nishitaggarwal - The problem is the current commit doesn't appear to work (for some hardware at least).

Going back to an older version of the repo is necessary until the issue is fixed.

I fixed it for myself by:

1. Completely removing all traces of the driver from my system.
2. Cloning the repo.
3. Running `git checkout 63cf0b4584aa8878b0fe8ab38017f31c319bde3d`
4. Installing as normal.

nishitaggarwal, 2024-07-11:
> @nishitaggarwal - The problem is the current commit doesn't appear to work (for some hardware at least).
>
> Going back to an older version of the repo is necessary until the issue is fixed.
>
> I fixed [...]

Thanks buddy this worked

danbinns-ss, 2024-07-11:
Comment from author on the breaking commit:
https://github.com/aircrack-ng/rtl8812au/commit/c7f8f6e3633bd91a9f359eb3406beb972e58cd7b#commitcomment-143176241

Mentions this as an alternative, though it's WIP:
https://github.com/lwfinger/rtw88

MelvinDeg, 2024-07-27:
> I am using the _rtl8812au_ **git** driver for a while on **rpi0 2 w** with usb key (0bda:0811), on **archlinux arm (armv7h)**. It was fine.
>
> In fact, I am manually building the driver at each kernel [...]
Had the same problem. Your solution helped. Thank you.

egilas, 2024-09-02:
Had same problems with Ubuntu 22.04, 5.15.0-119-generic #129-Ubuntu, i.e. airodump shows no BSSIDs. The `63cf0b4584aa8878b0fe8ab38017f31c319bde3d` commit works.