aircrack-ng/rtl8812au #1173: Monitor mode not working in awsu036ach

ROoTCo4e, 2024-07-03:
when i type this command sudo iw dev wlan0 set type monitor                   
it replies > command failed: Operation not supported (-95)

i have also try many other ways but i could not able to make it working 
note: i have install this driver successfully without any error 
and i can use it as a reciver 
but cant perform monitor mode and packet injection 

plzzz help me

Heyholiday067, 2024-07-04:
There is a bug in recent update.
Use my old fork and it work (i update it before the commit brought the bug)
https://github.com/Heyholiday067/rtl8812au

ROoTCo4e, 2024-07-04:
> There is a bug in recent update. Use my old fork and it work (i update it before the commit brought the bug) https://github.com/Heyholiday067/rtl8812au

still not working brother .... please help me ^._.^

Heyholiday067, 2024-07-05:
Run `uname -a` to show the current kernel version!
I remember the kernel > 6.9 will meet some bug but i still dont figure it out.So i downgrade my kali to 6.3.0 and everything works ! (6.8.x might work,you can test various kernel)

kazundo, 2024-07-05:
> https://github.com/Heyholiday067/rtl8812au
is work me with the new kernel kali thanks

gabrielhhz135, 2024-07-05:
> > https://github.com/Heyholiday067/rtl8812au
> > is work me with the new kernel kali thanks

How you install new kernel, i also instal new kernel 6.8 but also not working

Heyholiday067, 2024-07-05:
Well...i felt confused now.
But you can follow this steps

1.run `lsmod` to check the driver is loaded or not
2. run `iwconfig` to check whether the usb adapter is identified correctly
3.run `airmon-ng check kill` to kill the processes that will interfere the interface
4.run `aireplay-ng -9 wlan0` to test the adapter can inject info into packet
5.use dmesg to output the kernel log and paste here.

gabrielhhz135, 2024-07-05:
Monitor mode working, but when i type command like airodump-ng wlan0 , it is not working and mdk4 and 3 not working

gabrielhhz135, 2024-07-05:
> Well...i felt confused now. But you can follow this steps
>
> 1.run `lsmod` to check the driver is loaded or not 2. run `iwconfig` to check whether the usb adapter is identified correctly 3.run [...]

And how i can downgrade kali linux kernel and donrt update it but udate OS, vmware also not working

Heyholiday067, 2024-07-08:
> And how i can downgrade kali linux kernel and donrt update it but udate OS, vmware also not working

It is always not suggested to use vmware or virtualbox.Use true pc instead.

Heyholiday067, 2024-07-08:
> And how i can downgrade kali linux kernel and donrt update it but udate OS, vmware also not working

By the way,you didnt provide any error log or kernel log , how people can help you ?

demigod-lab, 2024-07-08:
[+] option: kill conflicting processes enabled

    Interface   PHY   Driver              Chipset                               
-----------------------------------------------------------------------
 1. wlan0mon    phy0  8812au              Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter

 [+] Enabling monitor mode on wlan0mon... failed

 [!] Error: Cannot find any interfaces in monitor mode

 [!] Full stack trace below

 [!]    Traceback (most recent call last):
 [!]    File "/usr/lib/python3/dist-packages/wifite/__main__.py", line 104, in entry_point
 [!]        wifite.start()
 [!]    File "/usr/lib/python3/dist-packages/wifite/__main__.py", line 57, in start
 [!]        Configuration.get_monitor_mode_interface()
 [!]    File "/usr/lib/python3/dist-packages/wifite/config.py", line 229, in get_monitor_mode_interface
 [!]        cls.interface = Airmon.ask()
 [!]                        ^^^^^^^^^^^^
 [!]    File "/usr/lib/python3/dist-packages/wifite/tools/airmon.py", line 333, in ask
 [!]        selected.interface = Airmon.start(selected)
 [!]                             ^^^^^^^^^^^^^^^^^^^^^^
 [!]    File "/usr/lib/python3/dist-packages/wifite/tools/airmon.py", line 213, in start
 [!]        raise Exception('Cannot find any interfaces in monitor mode')
 [!]  Exception: Cannot find any interfaces in monitor mode

having trouble in monitor mode... can anyone help me... thank you

kazundo, 2024-07-09:
> Monitor mode working, but when i type command like airodump-ng wlan0 , it is not working and mdk4 and 3 not working

me i work for me on vmware fusion 13 pro  before install the driver[  ](https://github.com/Heyholiday067/rtl8812au)  
remove all driver rtl88xx ;;;

ka3boosh, 2024-07-11:
i cant get it to work at VMware workstation I'm using kali 2024.2 I try everything i seen here or on other forums cant get it to work i always end up with error

mariusmoo, 2024-07-16:
I can currently reproduce the same problem that Heyholiday067 mentions.

Hardware: Raspberry Pi 4

`uname -a`: Linux rpi4 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

* `sudo make dkms_remove; git checkout b44d288; sudo make dkms_install; sudo reboot`
   * Monitor mode can be enabled but never reports any packets
* `sudo make dkms_remove; git checkout 63cf0b4; sudo make dkms_install; sudo reboot`
   * Monitor mode works fine

atommo999, 2024-07-17:
I've got the same issue- tried Heyholiday067's fork but still the same problem. It can see the adapter when doing an iwconfig, but if you set it to monitor mode and do an airodump-ng with it, no networks are detected.

It is also unusable in normal 'managed' mode either (essentially 'lights on, but no one home').

I am running the arm64 Kali release, fully up to date.

ka3boosh, 2024-07-17:
I fix the issue on kali 2024.2
After tow weeks of downloading bunch of useless updates and different kali linux iso files and vmware-image the fix was 3MB file and one command the issue fixed I'll upload a video soon how to fix it

atommo999, 2024-07-20:
I've ended up buying a different adapter instead now, an AWUS036ACS.

If anyone wants more info on the issue with the ACH though, I can do more tests with it if it helps sort the problem. Just let me know what commands I should try.

ka3boosh, 2024-07-20:
> I've ended up buying a different adapter instead now, an AWUS036ACS.
> 
> If anyone wants more info on the issue with the ACH though, I can do more tests with it if it helps sort the problem. Just let me know what commands I should try.

If u wait just a little bit more, here is the fix i uploaded a video with the right driver, it was an easy fix

https://youtu.be/Yr-8RmoNi70?si=-VGorwjSUVloLPLm

atommo999, 2024-07-20:
> If u wait just a little bit more, here is the fix i uploaded a video with the right driver, it was an easy fix
>
> https://youtu.be/Yr-8RmoNi70?si=-VGorwjSUVloLPLm

Thanks for your efforts- I have just tried (using the Plasma desktop environment with my Kali) and since doing those steps, I cannot load into the desktop environment.

Straight after rebooting it was much slower to load the login screen, and after logging in it is just stuck with a cursor on a black screen.

I can only imagine the driver has caused some sort of fundamental instability.

It is hard to say whether the problem is related to using Plasma rather than Xfce, or because I am running this on a physical machine rather than virtual.

Others may be able to clarify this with their own tests.

CursedOddity, 2024-09-18:
> There is a bug in recent update. Use my old fork and it work (i update it before the commit brought the bug) https://github.com/Heyholiday067/rtl8812au

Dude.... THANK YOU. 

Im not sure if its from upgrading to 6.10.9 or your fork but i finally have my AWUS036ACH working again on kali 2024.3 with kernal 6.10.9 with your fork. Everything works again. THANK YOU!

I was trying to get this to work for over a month.