aircrack-ng/rtl8812au #1190: Has anyone got the AWUS036ACH working on Raspberry Pi 5?

CursedOddity, 2024-09-14:
I stupidly bought the mentioned card and a new Pi 5 only to find out that the card doesn't work with the new kernels. I have got it to install but monitor mode and injection doesn't work. I got the Pi5 to run Kismet and a few other IDS.

Would be extremely grateful to finally get this to work. I don't really want to go out and buy a new card, as i just spent over $100 on this thing. 

I have tried different kernels, as well as different release versions and i am completely out of ideas. 

Thanks.

gizmochefmogwai, 2024-09-21:
Same thing here.

Kali on Pi 5, tried different kernels from 6.1.64-v8+ to 6.6.45-v8+

Impossible to make or to install by dkms, I create a new issue.

hadian90, 2024-10-09:
Manage to install driver, but it's not scanning any wifi networks. 6.6.51-v8. No error while install by dkms

Mair, 2024-11-20:
@hadian90 Thanks
>git checkout 63cf0b4
```
make
sudo install 
```
worked a charm

aguro90, 2024-11-25:
The key was to do
`git clone -b v5.6.4.2 https://github.com/aircrack-ng/rtl8812au.git`
then
`sudo git checkout 63cf0b4`
then 
`make`
then  
`make install`

Working on Raspberry 5 with NVMe M2 hat with latest raspiOS with arm64(6.6.51+rpt-rpi-2712). The adapter must be connected with self powered usb hub in order to avoid faults related to voltage.

tratzum, 2024-12-22:
Does it absolutely need external power as well. I got the driver installed and in monitor mode but nothing happens. I have tried so many pi os, kali os, for my pi 4's pi zero2w. I have no clue what to do. I have a different alfa usb adapter coming but not for over a week. :(

chiliwax, 2025-05-20:
I did a documentation to use it on raspberry (I'm on dietPi but should work for any distrib)
https://brash-william-6b5.notion.site/Driver-Installation-for-AlfaNetwork-AWUS036ACH-RTW8812AU-on-Raspberry-Pi-1f8662cf2d88808e950dd0162d46a7c8

I'm using https://github.com/lwfinger/rtw88 instead of https://github.com/aircrack-ng/rtl8812au.git as it's outdated

I did the doc from memory (kind of) so don't hesitate if you're still stuck