aircrack-ng/rtl8812au #1192: Impossible to compile due to error on rtl8812au/Module.symvers

gizmochefmogwai, 2024-09-22:
System : Raspberry 5, with Kali, Kernels 6.1.64-v8+ to 6.6.45-v8+

When make, getting the following error : 
`  CC [M]  /home/kali/rtl8812au/hal/phydm/halrf/rtl8814a/halrf_iqk_8814a.o
  CC [M]  /home/kali/rtl8812au/hal/phydm/rtl8814a/phydm_regconfig8814a.o
  CC [M]  /home/kali/rtl8812au/hal/phydm/halrf/rtl8814a/halrf_8814a_ce.o
  CC [M]  /home/kali/rtl8812au/hal/phydm/rtl8814a/phydm_rtl8814a.o
  CC [M]  /home/kali/rtl8812au/hal/phydm/txbf/haltxbf8814a.o
  CC [M]  /home/kali/rtl8812au/platform/platform_ops.o
  CC [M]  /home/kali/rtl8812au/core/rtw_mp.o
  LD [M]  /home/kali/rtl8812au/88XXau.o
  MODPOST /home/kali/rtl8812au/Module.symvers
WARNING: Module.symvers is missing.
         Modules may not have dependencies or modversions.
         You may get many unresolved symbol warnings.
WARNING: modpost: "skb_pull" [/home/kali/rtl8812au/88XXau.ko] undefined!
WARNING: modpost: "rtnl_is_locked" [/home/kali/rtl8812au/88XXau.ko] undefined!
WARNING: modpost: "usb_free_coherent" [/home/kali/rtl8812au/88XXau.ko] undefined!
WARNING: modpost: "__arch_copy_to_user" [/home/kali/rtl8812au/88XXau.ko] undefined!
WARNING: modpost: "cancel_work_sync" [/home/kali/rtl8812au/88XXau.ko] undefined!
WARNING: modpost: "memcmp" [/home/kali/rtl8812au/88XXau.ko] undefined!
WARNING: modpost: "cfg80211_sinfo_alloc_tid_stats" [/home/kali/rtl8812au/88XXau.ko] undefined!
WARNING: modpost: "vfree" [/home/kali/rtl8812au/88XXau.ko] undefined!
WARNING: modpost: "wiphy_register" [/home/kali/rtl8812au/88XXau.ko] undefined!
WARNING: modpost: "cfg80211_remain_on_channel_expired" [/home/kali/rtl8812au/88XXau.ko] undefined!
WARNING: modpost: suppressed 182 unresolved symbol warnings because there were too many)
  CC [M]  /home/kali/rtl8812au/88XXau.mod.o
  LD [M]  /home/kali/rtl8812au/88XXau.ko
make[1]: Leaving directory '/usr/src/kernel'
---------------------------------------------------------------------------
Visit https://github.com/aircrack-ng/rtl8812au for support/reporting issues
or check for newer versions (branches) of these drivers.
---------------------------------------------------------------------------`

Commands were : 
`apt-get install bc mokutil build-essential libelf-dev

# Following replace apt-get install linux-headers-* following this documentation : https://www.kali.org/docs/arm/raspberry-pi-5/
cd /usr/src/kernel
sudo git clean -fdx && sudo make bcm2711_defconfig && sudo make modules_prepare

sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
sed -i 's/CONFIG_PLATFORM_ARM64_RPI = n/CONFIG_PLATFORM_ARM64_RPI = y/g' Makefile
export ARCH=arm64
sed -i 's/^MAKE="/MAKE="ARCH=arm64\ /' dkms.conf`

Any idea what happens ?

MaxJoseph4, 2024-10-02:
Hi, could you find a solution? I have the same adapter and I can't compile the drivers.

tjnull, 2025-01-14:
Hi there if you are still having an issue with this, I found a possible solution. 

In the dkms.conf file you need to include this "KBUILD_MODPOST_WARN=1"

```bash
┌──(kali㉿rasberrypi)-[/usr/src/rtl8812au]
└─$ cat dkms.conf        
PACKAGE_NAME="realtek-rtl88xxau"
PACKAGE_VERSION="5.6.4.2~20230501"
CLEAN="'make' clean"
BUILT_MODULE_NAME[0]=88XXau
PROCS_NUM=`nproc`
[ $PROCS_NUM -gt 16 ] && PROCS_NUM=16
DEST_MODULE_LOCATION[0]="/updates"
MAKE="'make' -j$PROCS_NUM KVER=${kernelver} KSRC=/lib/modules/${kernelver}/build KBUILD_MODPOST_WARN=1"
AUTOINSTALL="yes"
```

This will suppress the warnings and it will build the module for you. Once the module has finished the setup reboot your PI with Kali Linux on it and you should see the adapter: 

```bash
┌──(kali㉿rasberrypi)-[~]
└─$ iwconfig
lo        no wireless extensions.

eth0      no wireless extensions.

wlan0     IEEE 802.11bgn  ESSID:"warberry"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:2.422 GHz  Access Point: 8C:3B:AD:ED:C0:35   
          Bit Rate:72.2 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=89/100  Signal level=62/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

wlan1     IEEE 802.11  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
 ```