aircrack-ng/rtl8812au #1199: [Bug] "array-index-out-of-bounds in rtl8812au/core/rtw_wlan_util.c" when reloading 88XXau

sardChen, 2024-10-29:
When executing fuzzing test for and reload rtl8812au, I found three array-index-out-of-bounds bugs in dmesg logs:

[  684.674062] usb 1-11.4: USB disconnect, device number 8
[  686.127497] usb 1-11.4: new high-speed USB device number 9 using xhci_hcd
[  686.204600] usb 1-11.4: New USB device found, idVendor=0bda, idProduct=8812, bcdDevice= 0.00
[  686.204623] usb 1-11.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  686.204634] usb 1-11.4: Product: 802.11n NIC
[  686.204642] usb 1-11.4: Manufacturer: Realtek
[  686.204649] usb 1-11.4: SerialNumber: 123456
[  686.250241] 88XXau: loading out-of-tree module taints kernel.
[  686.250299] 88XXau: module verification failed: signature and/or required key missing - tainting kernel
[  686.494483] usb 1-11.4: 88XXau 00:c0:ca:b4:93:54 hw_info[d7]
[  686.499443] usbcore: registered new interface driver rtl88XXau
[  686.549446] rtl88XXau 1-11.4:1.0 wlx00c0cab49354: renamed from wlan0
[  691.487880] ================================================================================
[  691.487911] UBSAN: array-index-out-of-bounds in /home/sandy/workplace/wifiTool/driver/rtl8812au/core/rtw_wlan_util.c:1905:48
[  691.487960] index 1 is out of range for type 'u8 [1]'
[  691.487973] CPU: 4 PID: 0 Comm: swapper/4 Tainted: G           OE      6.6.58 #1
[  691.487987] Hardware name: Gigabyte Technology Co., Ltd. B660M GAMING AC DDR4/B660M GAMING AC DDR4, BIOS F4 01/17/2022
[  691.487995] Call Trace:
[  691.488001]  <IRQ>
[  691.488008]  dump_stack_lvl+0x48/0x70
[  691.488033]  dump_stack+0x10/0x20
[  691.488046]  __ubsan_handle_out_of_bounds+0xa2/0x100
[  691.488059]  ? read_profile+0x321/0x660
[  691.488072]  HT_caps_handler+0x1d1/0xa90 [88XXau]
[  691.488541]  ? __pfx_HT_caps_handler+0x10/0x10 [88XXau]
[  691.488946]  ? __asan_memcpy+0x4e/0x80
[  691.488971]  OnAssocRsp+0x577/0x650 [88XXau]
[  691.489283]  DoReserved+0x14b/0x1d0 [88XXau]
[  691.489574]  ? DoReserved+0x30/0x1d0 [88XXau]
[  691.489855]  ? _raw_spin_lock_bh+0x86/0xf0
[  691.489873]  mgt_dispatcher+0x39f/0x4b0 [88XXau]
[  691.490157]  ? rtw_get_stainfo+0x30c/0x360 [88XXau]
[  691.490538]  ? __pfx_mgt_dispatcher+0x10/0x10 [88XXau]
[  691.490824]  ? recvframe_chk_defrag+0x15c/0x280 [88XXau]
[  691.491185]  validate_recv_mgnt_frame+0x9a4/0xd50 [88XXau]
[  691.491530]  ? GetHalDefVar8812A+0xcf9/0xd00 [88XXau]
[  691.491963]  ? GetHalDefVar8812AUsb+0xe/0x110 [88XXau]
[  691.492368]  validate_recv_frame+0x548/0x670 [88XXau]
[  691.492722]  ? __pfx_validate_recv_frame+0x10/0x10 [88XXau]
[  691.493055]  ? rx_query_phy_status+0x92b/0x9a0 [88XXau]
[  691.493384]  recv_func_prehandle+0x85/0xe0 [88XXau]
[  691.493707]  recv_func+0x56/0x340 [88XXau]
[  691.494026]  rtw_recv_entry+0x3b/0x140 [88XXau]
[  691.494337]  pre_recv_entry+0xf0/0x230 [88XXau]
[  691.494650]  recvbuf2recvframe+0x20e/0x590 [88XXau]
[  691.495070]  usb_recv_tasklet+0x12b/0x230 [88XXau]
[  691.495517]  tasklet_action_common.constprop.0+0x275/0x670
[  691.495536]  tasklet_action+0x22/0x30
[  691.495549]  handle_softirqs+0x192/0x5d0
[  691.495565]  __irq_exit_rcu+0x15c/0x1b0
[  691.495578]  irq_exit_rcu+0xe/0x20
[  691.495591]  common_interrupt+0xa4/0xb0
[  691.495602]  </IRQ>
[  691.495607]  <TASK>
[  691.495613]  asm_common_interrupt+0x27/0x40
[  691.495623] RIP: 0010:cpuidle_enter_state+0x1df/0x520
[  691.495634] Code: 01 73 dd 48 83 c4 28 44 89 f0 5b 41 5c 41 5d 41 5e 41 5f 5d 31 d2 31 c9 31 f6 31 ff 45 31 c0 c3 cc cc cc cc fb 0f 1f 44 00 00 <45> 85 f6 0f 89 06 ff ff ff 48 c7 43 18 00 00 00 00 49 83 fd 09 0f
[  691.495645] RSP: 0018:ffff888100d6fd30 EFLAGS: 00000246
[  691.495660] RAX: 0000000000000000 RBX: ffff88885c24ffe0 RCX: 0000000000000000
[  691.495669] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
[  691.495676] RBP: ffff888100d6fd80 R08: 0000000000000000 R09: 0000000000000000
[  691.495682] R10: 0000000000000000 R11: 0000000000000000 R12: ffffffffa5a627c0
[  691.495689] R13: 0000000000000004 R14: 0000000000000004 R15: 000000a0ffe2bf2d
[  691.495705]  cpuidle_enter+0x4f/0xb0
[  691.495719]  call_cpuidle+0x47/0xd0
[  691.495732]  do_idle+0x372/0x460
[  691.495747]  ? __pfx_do_idle+0x10/0x10
[  691.495764]  cpu_startup_entry+0x58/0x70
[... 163 lines cut ...]
[  699.926265]  __irq_exit_rcu+0x15c/0x1b0
[  699.926268]  irq_exit_rcu+0xe/0x20
[  699.926270]  common_interrupt+0xa4/0xb0
[  699.926272]  </IRQ>
[  699.926273]  <TASK>
[  699.926274]  asm_common_interrupt+0x27/0x40
[  699.926276] RIP: 0010:cpuidle_enter_state+0x1df/0x520
[  699.926279] Code: 01 73 dd 48 83 c4 28 44 89 f0 5b 41 5c 41 5d 41 5e 41 5f 5d 31 d2 31 c9 31 f6 31 ff 45 31 c0 c3 cc cc cc cc fb 0f 1f 44 00 00 <45> 85 f6 0f 89 06 ff ff ff 48 c7 43 18 00 00 00 00 49 83 fd 09 0f
[  699.926281] RSP: 0018:ffff888100d6fd30 EFLAGS: 00000246
[  699.926283] RAX: 0000000000000000 RBX: ffff88885c24ffe0 RCX: 0000000000000000
[  699.926285] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
[  699.926287] RBP: ffff888100d6fd80 R08: 0000000000000000 R09: 0000000000000000
[  699.926288] R10: 0000000000000000 R11: 0000000000000000 R12: ffffffffa5a627c0
[  699.926289] R13: 0000000000000004 R14: 0000000000000004 R15: 000000a2f685635a
[  699.926291]  ? __pfx_menu_select+0x10/0x10
[  699.926295]  cpuidle_enter+0x4f/0xb0
[  699.926297]  call_cpuidle+0x47/0xd0
[  699.926300]  do_idle+0x372/0x460
[  699.926303]  ? __pfx_do_idle+0x10/0x10
[  699.926306]  cpu_startup_entry+0x58/0x70
[  699.926308]  start_secondary+0x220/0x2b0
[  699.926311]  ? __pfx_start_secondary+0x10/0x10
[  699.926313]  secondary_startup_64_no_verify+0x18f/0x19b
[  699.926317]  </TASK>
[  699.926319] ================================================================================

IamCOD3X, 2024-11-04:
Do share the driver informaiton like which branch you are using so that someone can take a look on the code.

sardChen, 2024-12-06:
> Do share the driver informaiton like which branch you are using so that someone can take a look on the code.

Sorry for the delay response. The branch we test is the latest version, v5.6.4.2. The related code is shown blew:

[image]