aircrack-ng/rtl8812au #1201: CFI Kernel Panic on Kernel v5.4.242

IamCOD3X, 2024-11-02:
Getting kernel panic from CFI with this driver when plugging in a Wi-Fi adapter in device and trying to turn on monitor mode.

Keep in mind **we can't ignore these config flags** as it's a samsung device and changing any of them causing boot failure
```
CONFIG_LTO=y
CONFIG_ARCH_SUPPORTS_LTO_CLANG=y
CONFIG_ARCH_SUPPORTS_THINLTO=y
CONFIG_THINLTO=y
# CONFIG_LTO_NONE is not set
CONFIG_LTO_CLANG=y
CONFIG_CFI_CLANG=y
CONFIG_CFI_CLANG_SHADOW=y
# CONFIG_CFI_PERMISSIVE is not set
```
last_kmsg :-
```
[  265.997560] [4: RTW_CMD_THREAD:17046] [phydm_cck_pd_th]warning
[  265.998708] [4: RTW_CMD_THREAD:17046] **Kernel panic - not syncing: CFI failure (target: 0x0)**
[  265.998732] [4: RTW_CMD_THREAD:17046] CPU: 4 PID: 17046 Comm: RTW_CMD_THREAD Tainted: G S      WC        5.4.242-ViP3R🐍-v2.0-NETHUNTER-RELOADED-OneUI-SPECIAL #1
[  265.998741] [4: RTW_CMD_THREAD:17046] Hardware name: Samsung O1S board based on EXYNOS2100 (DT)
[  265.998748] [4: RTW_CMD_THREAD:17046] Call trace:
[  265.998766] [4: RTW_CMD_THREAD:17046]  dump_backtrace.cfi_jt+0x0/0x4
[  265.998781] [4: RTW_CMD_THREAD:17046]  show_stack+0x18/0x24
[  265.998795] [4: RTW_CMD_THREAD:17046]  dump_stack+0xd0/0x10c
[  265.998806] [4: RTW_CMD_THREAD:17046]  panic+0x194/0x3f4
[  265.998820] [4: RTW_CMD_THREAD:17046]  __ubsan_handle_cfi_check_fail_abort+0x0/0x14
[  265.998830] [4: RTW_CMD_THREAD:17046]  perf_proc_update_handler+0x0/0xcc
[  265.998839] [4: RTW_CMD_THREAD:17046]  __cfi_check+0x3c4ec/0x42b3c
[  265.998975] [4: RTW_CMD_THREAD:17046]  odm_get_tracking_table+0x338/0x368 [88XXau]
[  265.999103] [4: RTW_CMD_THREAD:17046]  odm_txpowertracking_callback_thermal_meter+0x304/0x708 [88XXau]
[  265.999232] [4: RTW_CMD_THREAD:17046]  odm_txpowertracking_check+0x50/0xa8 [88XXau]
[  265.999359] [4: RTW_CMD_THREAD:17046]  phydm_watchdog+0xb8/0xe8 [88XXau]
[  265.999485] [4: RTW_CMD_THREAD:17046]  rtw_phydm_watchdog+0x19c/0x1bc [88XXau]
[  265.999620] [4: RTW_CMD_THREAD:17046]  rtl8812_HalDmWatchDog+0x44/0x54 [88XXau]
[  265.999756] [4: RTW_CMD_THREAD:17046]  rtw_hal_dm_watchdog+0x3c/0x70 [88XXau]
[  265.999895] [4: RTW_CMD_THREAD:17046]  rtw_drvextra_cmd_hdl+0xa4/0x3a4 [88XXau]
[  266.000034] [4: RTW_CMD_THREAD:17046]  rtw_cmd_thread+0x2a4/0x524 [88XXau]
[  266.000048] [4: RTW_CMD_THREAD:17046]  kthread+0x16c/0x17c
[  266.000059] [4: RTW_CMD_THREAD:17046]  ret_from_fork+0x10/0x18
[  266.000072] [4: RTW_CMD_THREAD:17046] SMP: stopping secondary CPUs
```

I know this one is similar to this issue #1036

closed 2024-11-06 by IamCOD3X

JoseMCCGit, 2024-12-06:
> Solved ✨| still seems strange as there is a unknown issue with latest commits, I build old driver and it worked out of the box.

Hi, I'm having same problem of CFI kernel panic in 5.4, tried different changes but still not working, what did you do for solve it?

IamCOD3X, 2024-12-10:
> > Solved ✨| still seems strange as there is a unknown issue with latest commits, I build old driver and it worked out of the box.
> 
> Hi, I'm having same problem of CFI kernel panic in 5.4, tried different changes but still not working, what did you do for solve it?

Already mentioned, buddy, that I used an older driver instead of the latest branch! Try searching on Open GitHub; I don’t remember whose repository I cloned.

nitrozen9, 2024-12-11:
hi, i see you are also having problems with CFI, i am working on a kernel for samsung a50 but i need to enable system v ipc in kernel for a 3d printer project, but if i try to boot a kernel with sys v ipc enabled the boot hangs because of cfi security, do you know how to disable it and use sys v ipc, thanks