aircrack-ng/rtl8812au #1203: Problems with installing drivers.

H4CKS3T, 2024-11-07:
When installing the driver, the following picture is displayed:

```
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.11.2-amd64/build M=/opt/Drivers/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-6.11.2-amd64'
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_cmd.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_security.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_debug.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_io.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_ioctl_query.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_ioctl_set.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_ieee80211.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_mlme.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_mlme_ext.o
/opt/Drivers/rtl8812au/core/rtw_mlme_ext.c: In function ‘OnAuth’:
/opt/Drivers/rtl8812au/core/rtw_mlme_ext.c:2167:41: warning: suggest braces around empty body in an ‘if’ statement [-Wempty-body]
 2167 |                                         ;/* TODO: STA re_auth within expire_to */
      |                                         ^
/opt/Drivers/rtl8812au/core/rtw_mlme_ext.c:2172:33: warning: suggest braces around empty body in an ‘if’ statement [-Wempty-body]
 2172 |                                 ; /* TODO: STA re_auth and auth timeout */
      |                                 ^
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_mi.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_wlan_util.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_vht.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_pwrctrl.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_rf.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_chplan.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_recv.o
/opt/Drivers/rtl8812au/core/rtw_recv.c: In function ‘validate_recv_frame’:
/opt/Drivers/rtl8812au/core/rtw_recv.c:2390:79: warning: suggest braces around empty body in an ‘else’ statement [-Wempty-body]
 2390 |                         DBG_COUNTER(adapter->rx_logs.core_rx_pre_data_handled);
      |                                                                               ^
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_sta_mgt.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_ap.o
/opt/Drivers/rtl8812au/core/rtw_ap.c:313:12: warning: ‘issue_aka_chk_frame’ defined but not used [-Wunused-function]
  313 | static int issue_aka_chk_frame(_adapter *adapter, struct sta_info *psta)
      |            ^~~~~~~~~~~~~~~~~~~
  CC [M]  /opt/Drivers/rtl8812au/core/mesh/rtw_mesh.o
  CC [M]  /opt/Drivers/rtl8812au/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /opt/Drivers/rtl8812au/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_xmit.o
/opt/Drivers/rtl8812au/core/rtw_xmit.c: In function ‘update_attrib’:
/opt/Drivers/rtl8812au/core/rtw_xmit.c:1376:74: warning: suggest braces around empty body in an ‘else’ statement [-Wempty-body]
 1376 |                 DBG_COUNTER(padapter->tx_logs.core_tx_upd_attrib_unknown);
      |                                                                          ^
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_p2p.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_rson.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_tdls.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_br_ext.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_iol.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_sreset.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_btcoex_wifionly.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_btcoex.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_beamforming.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_odm.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_rm.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_rm_fsm.o
  CC [M]  /opt/Drivers/rtl8812au/core/efuse/rtw_efuse.o
  CC [M]  /opt/Drivers/rtl8812au/os_dep/osdep_service.o
  CC [M]  /opt/Drivers/rtl8812au/os_dep/linux/os_intfs.o
  CC [M]  /opt/Drivers/rtl8812au/os_dep/linux/usb_intf.o
  CC [M]  /opt/Drivers/rtl8812au/os_dep/linux/usb_ops_linux.o
/opt/Drivers/rtl8812au/os_dep/linux/usb_ops_linux.c:323:12: warning: ‘usb_bulkout_zero’ defined but not used [-Wunused-function]
  323 | static u32 usb_bulkout_zero(struct intf_hdl *pintfhdl, u32 addr)
      |            ^~~~~~~~~~~~~~~~
  CC [M]  /opt/Drivers/rtl8812au/os_dep/linux/ioctl_linux.o
/opt/Drivers/rtl8812au/os_dep/linux/ioctl_linux.c:10674:12: warning: ‘rtw_tdls_get_best_ch’ defined but not used [-Wunused-function]
10674 | static int rtw_tdls_get_best_ch(struct net_device *dev,
      |            ^~~~~~~~~~~~~~~~~~~~
/opt/Drivers/rtl8812au/os_dep/linux/ioctl_linux.c:10631:12: warning: ‘rtw_tdls_getsta’ defined but not used [-Wunused-function]
[... 390 lines cut ...]
  CC [M]  /opt/Drivers/rtl8812au/hal/phydm/rtl8814a/halhwimg8814a_bb.o
  CC [M]  /opt/Drivers/rtl8812au/hal/phydm/rtl8814a/halhwimg8814a_mac.o
  CC [M]  /opt/Drivers/rtl8812au/hal/phydm/rtl8814a/halhwimg8814a_rf.o
  CC [M]  /opt/Drivers/rtl8812au/hal/phydm/halrf/rtl8814a/halrf_iqk_8814a.o
  CC [M]  /opt/Drivers/rtl8812au/hal/phydm/rtl8814a/phydm_regconfig8814a.o
  CC [M]  /opt/Drivers/rtl8812au/hal/phydm/halrf/rtl8814a/halrf_8814a_ce.o
  CC [M]  /opt/Drivers/rtl8812au/hal/phydm/rtl8814a/phydm_rtl8814a.o
  CC [M]  /opt/Drivers/rtl8812au/hal/phydm/txbf/haltxbf8814a.o
  CC [M]  /opt/Drivers/rtl8812au/platform/platform_ops.o
  CC [M]  /opt/Drivers/rtl8812au/core/rtw_mp.o
  LD [M]  /opt/Drivers/rtl8812au/88XXau.o
/opt/Drivers/rtl8812au/88XXau.o: warning: objtool: rtw_mp_cmd+0xfe: unannotated intra-function call
  MODPOST /opt/Drivers/rtl8812au/Module.symvers
  CC [M]  /opt/Drivers/rtl8812au/88XXau.mod.o
  LD [M]  /opt/Drivers/rtl8812au/88XXau.ko
  BTF [M] /opt/Drivers/rtl8812au/88XXau.ko
Skipping BTF generation for /opt/Drivers/rtl8812au/88XXau.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-6.11.2-amd64'
---------------------------------------------------------------------------
Visit https://github.com/aircrack-ng/rtl8812au for support/reporting issues
or check for newer versions (branches) of these drivers.                   
---------------------------------------------------------------------------
```

After installation and reboot, the adapter refuses to work while lying down or simply not seeing any network.

Antimorph, 2024-11-17:
I have a similar problem. The adapters are listed but greyed out. Also it seemed to have a problem with the gcc and g++ version.

iamAika, 2024-11-28:
same problem here