aircrack-ng/rtl8812au #1210: AWUS036ACH (rtl8812au) not working on kernel 6.8.0-49-generic

cesarth, 2024-11-24:
So, I downloaded the repo, and it looks like it loads okay, but nothing is working: no wifi networks are found and the led does not light.

Steps taken:
-----------------------------------------------------------------------------------
user@computer:/path $ git log -1
commit b44d288f423ede0fc7cdbf92d07a7772cd727de4 (HEAD -> v5.6.4.2, origin/v5.6.4.2, origin/HEAD)
Author: kimocoder <christian@aircrack-ng.org>
Date:   Fri May 10 06:56:54 2024 +0200

    Add logfile script

user@computer:/path $ sudo make dkms_install 
mkdir: se ha creado el directorio '/usr/src/8812au-5.6.4.2_35491.20191025'
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025
Creating symlink /var/lib/dkms/8812au/5.6.4.2_35491.20191025/source -> /usr/src/8812au-5.6.4.2_35491.20191025
dkms build -m 8812au -v 5.6.4.2_35491.20191025
Sign command: /usr/bin/kmodsign
Signing key: /var/lib/shim-signed/mok/MOK.priv
Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der

Building module:
Cleaning build area...
'make' -j4 KVER=6.8.0-49-generic KSRC=/lib/modules/6.8.0-49-generic/build.......................
Signing module /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/88XXau.ko
Cleaning build area...
dkms install -m 8812au -v 5.6.4.2_35491.20191025

88XXau.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.8.0-49-generic/updates/dkms/
depmod...
dkms status -m 8812au
8812au/5.6.4.2_35491.20191025, 6.8.0-49-generic, x86_64: installed

user@computer:/path $ lsmod | grep rtl 

user@computer:/path $  #now I plug the device.  

user@computer:/path $ dmesg -w
....
[ 1470.968511] usb 3-1: new high-speed USB device number 5 using xhci_hcd
[ 1471.094903] usb 3-1: New USB device found, idVendor=0bda, idProduct=8812, bcdDevice= 0.00
[ 1471.094916] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1471.094920] usb 3-1: Product: 802.11n NIC
[ 1471.094922] usb 3-1: Manufacturer: Realtek
[ 1471.094925] usb 3-1: SerialNumber: 123456
[ 1471.429558] usb 3-1: 88XXau 00:c0:ca:a4:71:81 hw_info[d7]
[ 1471.430705] usbcore: registered new interface driver rtl88XXau
[ 1471.443921] rtl88XXau 3-1:1.0 wlx00c0caa47181: renamed from wlan0

user@computer:/path $ lsmod | grep 88XX
88XXau               2924544  0
cfg80211             1323008  1 88XXau

user@computer:/path $  # but I can not see any network appearing in NetworkManager. (it works with another USB Wifi Dongle, so it is not NM configuration)

user@computer:/path $  # in addition, the path in the docs to control the LED is not right:

user@computer:/path $ ls /proc/net/rtl8812au/
ls: no se puede acceder a '/proc/net/rtl8812au/': No existe el archivo o el directorio

user@computer:/path $ ls /proc/net/rtl88XXau/
chplan_id_list  chplan_test  chplan_ver  country_chplan_map  drv_cfg  log_level  ver_info  wlx00c0caa47181

user@computer:/path $  # The led does nothing when I issue these commands:

user@computer:/path $  echo "0" > /proc/net/rtl88XXau/wlx00c0caa47181/led_ctrl

user@computer:/path $  echo "1" > /proc/net/rtl88XXau/wlx00c0caa47181/led_ctrl

-----------------------------------------------------------------------------------
Any hint? Do you need more info from my side?

Thank you in advance,

César

E-Savage, 2024-11-26:
I'm actually having the same issue now,  and trying to figure out how to fix it

E-Savage, 2024-11-29:
So, what I ended up doing, because I wanted to use it now, was making a live usb of kali linux 2022.2 because it has native drivers for RTL8812AU and it works beautifully. 

You can install them with DKMS.