aircrack-ng/rtl8812au #1216: Driver installed but device not listing any networks

luigi-benvenaglia, 2024-12-05:
I am using Ubuntu 24.04 on a macbook, therefore my kernel is `6.6.62-1-t2-noble`

I have installed the drivers according to the instructions and the device appears in the UI of Ubuntu as also in the output of `iwconfig` command

[image]

```
root@MyLaptop:~# iwconfig
lo        no wireless extensions.

wlxa86e84aadc65  unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

However as you can see in the screenshot it does not list any networks.

The scan command with the interface name fails as well

```
# iwlist wlxa86e84aadc65 scan
wlxa86e84aadc65 Failed to read scan data: Resource temporarily unavailable
```

What might be the issue?

legendPerceptor, 2024-12-10:
Same problem on Kali Linux. Tried many ways, and it looks like the driver's problem. Waiting for answers.

legendPerceptor, 2024-12-10:
I found one solution. Just don't use git clone. Run the following command and then reboot. Problem solved for me.

```bash
sudo apt install realtek-rtl88xxau-dkms
```

freebeego, 2024-12-11:
same problem

freebeego, 2024-12-11:
helped for me

```
git clone https://github.com/aircrack-ng/rtl8812au
cd rtl8812au/
git reset --hard 63cf0b4
make dkms_install
```

osc-fernandez, 2024-12-12:
> I found one solution. Just don't use git clone. Run the following command and then reboot. Problem solved for me.
> 
> ```shell
> sudo apt install realtek-rtl88xxau-dkms
> ```

When you say don't use clone does this mean I have uninstall the drivers I got via github?

legendPerceptor, 2024-12-13:
> > I found one solution. Just don't use git clone. Run the following command and then reboot. Problem solved for me.
> > ```shell
> > sudo apt install realtek-rtl88xxau-dkms
> > ```
> 
> When you say don't use clone does this mean I have uninstall the drivers I got via github?

The `apt install` will override your manual installation from Github so it doesn't matter if you uninstall it manually.

65usami, 2024-12-15:
I think, same problem I have been facing.
It is on Ubuntu 24.
I got following errors.
Please kindly help someone.

```
[   16.342193] ------------[ cut here ]------------
[   16.342194] UBSAN: array-index-out-of-bounds in /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/ioctl_cfg80211.c:1678:110
[   16.342203] index 24 is out of range for type 'u8 [*]'
[   16.342208] CPU: 1 PID: 610 Comm: wpa_supplicant Tainted: G           OE      6.8.0-1018-oem #18-Ubuntu
[   16.342212] Hardware name: Intel(R) Client Systems NUC11PAHi3/NUC11PABi3, BIOS PATGL357.0050.2022.1228.1726 12/28/2022
[   16.342215] Call Trace:
[   16.342217]  <TASK>
[   16.342219]  dump_stack_lvl+0x76/0xa0
[   16.342224]  dump_stack+0x10/0x20
[   16.342228]  __ubsan_handle_out_of_bounds+0xc6/0x110
[   16.342234]  cfg80211_rtw_set_default_mgmt_key+0x2ba3/0x4660 [88XXau]
[   16.342411]  cfg80211_rtw_set_default_mgmt_key+0x3032/0x4660 [88XXau]
[   16.342587]  nl80211_new_key+0x1d7/0x380 [cfg80211]
[   16.342692]  genl_family_rcv_msg_doit+0xf7/0x160
[   16.342699]  genl_family_rcv_msg+0x184/0x250
[   16.342704]  ? __pfx_nl80211_pre_doit+0x10/0x10 [cfg80211]
[   16.342800]  ? __pfx_nl80211_new_key+0x10/0x10 [cfg80211]
[   16.342907]  ? __pfx_nl80211_post_doit+0x10/0x10 [cfg80211]
[   16.343017]  genl_rcv_msg+0x4c/0xb0
[   16.343022]  ? __pfx_genl_rcv_msg+0x10/0x10
[   16.343026]  netlink_rcv_skb+0x5a/0x110
[   16.343036]  genl_rcv+0x28/0x50
[   16.343040]  netlink_unicast+0x24a/0x390
[   16.343046]  netlink_sendmsg+0x214/0x470
[   16.343054]  ____sys_sendmsg+0x38b/0x3f0
[   16.343062]  ___sys_sendmsg+0x9a/0xf0
[   16.343068]  ? __sys_setsockopt+0x79/0xe0
[   16.343075]  __sys_sendmsg+0x89/0xf0
[   16.343082]  __x64_sys_sendmsg+0x1d/0x30
[   16.343086]  x64_sys_call+0x91e/0x25a0
[   16.343091]  do_syscall_64+0x7f/0x180
[   16.343095]  ? syscall_exit_to_user_mode+0x86/0x260
[   16.343101]  ? do_syscall_64+0x8c/0x180
[   16.343105]  ? do_syscall_64+0x8c/0x180
[   16.343109]  ? __x64_sys_ioctl+0xa0/0xf0
[   16.343116]  ? syscall_exit_to_user_mode+0x86/0x260
[   16.343121]  ? do_syscall_64+0x8c/0x180
[   16.343125]  ? clear_bhb_loop+0x15/0x70
[   16.343132]  ? clear_bhb_loop+0x15/0x70
[   16.343137]  ? clear_bhb_loop+0x15/0x70
[   16.343144]  entry_SYSCALL_64_after_hwframe+0x78/0x80
[   16.343149] RIP: 0033:0x79796a72c004
[   16.343161] Code: 15 19 6e 0d 00 f7 d8 64 89 02 b8 ff ff ff ff eb bf 0f 1f 44 00 00 f3 0f 1e fa 80 3d 45 f0 0d 00 00 74 13 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 4c c3 0f 1f 00 55 48 89 e5 48 83 ec 20 89 55
[   16.343164] RSP: 002b:00007ffe973615c8 EFLAGS: 00000202 ORIG_RAX: 000000000000002e
[   16.343169] RAX: ffffffffffffffda RBX: 000062f159566850 RCX: 000079796a72c004
[   16.343172] RDX: 0000000000000000 RSI: 00007ffe97361600 RDI: 0000000000000006
[   16.343175] RBP: 00007ffe973615f0 R08: 0000000000000004 R09: 00000000000000f0
[   16.343178] R10: 00007ffe97361700 R11: 0000000000000202 R12: 000062f1595ffe00
[   16.343181] R13: 000062f159566b30 R14: 00007ffe97361600 R15: 0000000000000000
[   16.343187]  </TASK>
[   16.343196] ---[ end trace ]---

```

AM0TZ, 2024-12-16:
> I found one solution. Just don't use git clone. Run the following command and then reboot. Problem solved for me.
> 
> ```shell
> sudo apt install realtek-rtl88xxau-dkms
> ```

worked like a charm! tnx
(don't forget to reboot)

65usami, 2024-12-17:
> ```shell
> apt install realtek-rtl88xxau-dkms
> ```

This is not working my case....
The following are the logs.

```
root@mypc:~# apt install realtek-rtl88xxau-dkms
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package realtek-rtl88xxau-dkms
root@mypc:~# uname -r
6.8.0-1019-oem
root@mypc:~# dmidecode -s baseboard-product-name
NUC13ANB-M
```

I think realtek-rtl88xxau-dkms is quite an old module. That is why I can't seem to find it in Ubuntu24.

65usami, 2024-12-18:
> I believe you can install on Ubuntu by
> 
> ```shell
> sudo apt install rtl8812au-dkms
> ```

@legendPerceptor Thanks for your help.
It worked. I was able to install it.

But since I am a root user, I didn't think there was any difference between having the apt and not having it....

I was able to install it, but my USB Wifi did not work. Unfortunately.
And also there were no logs.

```
root@ctv:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 004: ID 8087:0033 Intel Corp. AX211 Bluetooth
Bus 003 Device 005: ID 0411:029b BUFFALO INC. (formerly MelCo., Inc.) 802.11ac WLAN Adapter
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

```
root@ctv:~# dmesg | grep -i 8812au
[  926.655705] 8812au: loading out-of-tree module taints kernel.
[  926.655724] 8812au: module verification failed: signature and/or required key missing - tainting kernel
[  926.687092] RTL871X: rtl8812au v4.3.8_12175.20140902
[  926.687134] usbcore: registered new interface driver rtl8812au
```

john-t-needham, 2024-12-18:
This worked for me to get my Netgear A6100 AC600 DB Wireless adapter to show "Visible networks" in Ubuntu 24.04 in the Wi-Fi section of Ubuntu's settings GUI. I believe the critical step below is the "git reset --hard 63cf0b4" step, but I listed all the steps I took anyhow. 

sudo apt-get install linux-headers-generic
sudo apt update
sudo apt upgrade
reboot
git clone https://github.com/aircrack-ng/rtl8812au
cd rtl8812au
sudo make clean
sudo make dkms_remove
reboot
git reset --hard 63cf0b4
sudo make clean
sudo make
sudo make dkms_install

When running the command "dkms status", I have somehow from trying other things also ended up with "realtek-rtl88xxau/5.6.4.2~20240509: added", but I don't think that matters, and I'm going to ignore it. The part that seems relevant is that the "dkms status" command is showing "8812au/5.6.4.2_35491.20192025" as the other line of output from the command.

65usami, 2024-12-18:
@john-t-needham 

> This worked for me to get my Netgear A6100 AC600 DB Wireless adapter to show "Visible networks" in Ubuntu 24.04 in the Wi-Fi section of Ubuntu's settings GUI. I believe the critical step below is [...]

I have also tried this way, but I get the following error Do you have any errors like this?

```
[   16.342193] ------------[ cut here ]------------
[   16.342194] UBSAN: array-index-out-of-bounds in /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/ioctl_cfg80211.c:1678:110
[   16.342203] index 24 is out of range for type 'u8 [*]'
[   16.342208] CPU: 1 PID: 610 Comm: wpa_supplicant Tainted: G           OE      6.8.0-1018-oem #18-Ubuntu
[   16.342212] Hardware name: Intel(R) Client Systems NUC11PAHi3/NUC11PABi3, BIOS PATGL357.0050.2022.1228.1726 12/28/2022
[   16.342215] Call Trace:
[   16.342217]  <TASK>
[   16.342219]  dump_stack_lvl+0x76/0xa0
```

tom-jm69, 2024-12-22:
> This worked for me to get my Netgear A6100 AC600 DB Wireless adapter to show "Visible networks" in Ubuntu 24.04 in the Wi-Fi section of Ubuntu's settings GUI. I believe the critical step below is the "git reset --hard [63cf0b4](https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d)" step, but I listed all the steps I took anyhow.
> 
> Unplug the Wi-Fi USB dongle. Restart the computer. Enter the following terminal commands: sudo apt-get install linux-headers-generic sudo apt update sudo apt upgrade
> 
> Restart the computer. git clone https://github.com/aircrack-ng/rtl8812au cd rtl8812au sudo make clean sudo make dkms_remove
> 
> Restart the computer. git reset --hard [63cf0b4](https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d) sudo make clean sudo make sudo make dkms_install
> 
> Restart the computer. Insert the USB Wi-Fi dongle back into a USB port. Open the Ubuntu Wi-Fi section of the Ubuntu's settings GUI. Select the Wi-Fi network to which you want to connect.
> 
> Note that when I run the command "dkms status", I have somehow from trying other things also ended up with "realtek-rtl88xxau/5.6.4.2~20240509: added", but I don't think that matters, and I'm going to ignore it. The part that seems relevant is that the "dkms status" command is showing "8812au/5.6.4.2_35491.20192025, 6.8.0-51-generic, x86_64: installed" as the other line of output from the command.

Fixed it for me on arch 6.12.4-arch1-1.

DanielAKAOi, 2024-12-25:
did you check if the module "8812au"  is up?

[image]

X-FLASH668, 2025-01-09:
> Fixed it for me on arch 6.12.4-arch1-1.

This also worked for me. I tried it many times before with different manuals, but this finaly works. Thanks.

sebastiaanluca, 2025-01-16:
Can confirm this works for me too (finally). Combination of Raspotify/Debian (kernel panics or DKMS crashes on Ubuntu 24), commit https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d, nmcli/NetworkManager, rebooting and plugging in dongle.

mjopensuse, 2025-01-20:
Had the same issue on OpenSuse Leap 15.6 (==Kernel 6.4, gcc 7.5). Dropping back to commit https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d worked, plus doing some local changes to get the driver to compile as follows:
iff --git a/os_dep/linux/ioctl_cfg80211.c b/os_dep/linux/ioctl_cfg80211.c
index c1f20e9..ae7e131 100644
--- a/os_dep/linux/ioctl_cfg80211.c
+++ b/os_dep/linux/ioctl_cfg80211.c
@@ -5283,7 +5283,7 @@ exit:
        return ret;
 }
 
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 7, 0))
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 4, 0))
 static int cfg80211_rtw_change_beacon(struct wiphy *wiphy, struct net_device *ndev,
                struct cfg80211_ap_update *params)
 {
diff --git a/os_dep/linux/wifi_regd.c b/os_dep/linux/wifi_regd.c
index 45c5bd1..0d964dc 100644
--- a/os_dep/linux/wifi_regd.c
+++ b/os_dep/linux/wifi_regd.c
@@ -405,7 +405,7 @@ static void _rtw_regd_init_wiphy(struct rtw_regulatory *reg, struct wiphy *wiphy
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 19, 0)) && (LINUX_VERSION_CODE < KERNEL_VERSION(6, 1, 39)) \
        || (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 3, 0)) && (LINUX_VERSION_CODE < KERNEL_VERSION(6, 3, 13)) \
        || (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 4, 0)) && (LINUX_VERSION_CODE < KERNEL_VERSION(6, 4, 4))
-wiphy->regulatory_flags |= REGULATORY_IGNORE_STALE_KICKOFF;
+wiphy->regulatory_flags |= REGULATORY_COUNTRY_IE_IGNORE;
 #endif

markruiz14, 2025-01-22:
> helped for me
> 
> ```
> git clone https://github.com/aircrack-ng/rtl8812au
> cd rtl8812au/
> git reset --hard 63cf0b4
> make dkms_install
> ```

Resetting to the above commit hash worked for me. Here's my distro and kernel info:

```
$ cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=24.04
DISTRIB_CODENAME=noble
DISTRIB_DESCRIPTION="Ubuntu 24.04.1 LTS"

$ uname -a
Linux ares 6.8.0-51-generic #52-Ubuntu SMP PREEMPT_DYNAMIC Thu Dec  5 13:09:44 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
```

65usami, 2025-01-22:
@markruiz14 

I have also tried this way, but I get the following error Do you have any errors like this?

```
[root@pc ~]# dmesg
[   16.342193] ------------[ cut here ]------------
[   16.342194] UBSAN: array-index-out-of-bounds in /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/ioctl_cfg80211.c:1678:110
[   16.342203] index 24 is out of range for type 'u8 [*]'
[   16.342208] CPU: 1 PID: 610 Comm: wpa_supplicant Tainted: G           OE      6.8.0-1018-oem #18-Ubuntu
[   16.342212] Hardware name: Intel(R) Client Systems NUC11PAHi3/NUC11PABi3, BIOS PATGL357.0050.2022.1228.1726 12/28/2022
[   16.342215] Call Trace:
[   16.342217]  <TASK>
[   16.342219]  dump_stack_lvl+0x76/0xa0
[   16.342224]  dump_stack+0x10/0x20
[   16.342228]  __ubsan_handle_out_of_bounds+0xc6/0x110
[   16.342234]  cfg80211_rtw_set_default_mgmt_key+0x2ba3/0x4660 [88XXau]
[   16.342411]  cfg80211_rtw_set_default_mgmt_key+0x3032/0x4660 [88XXau]
[   16.342587]  nl80211_new_key+0x1d7/0x380 [cfg80211]
[   16.342692]  genl_family_rcv_msg_doit+0xf7/0x160
[   16.342699]  genl_family_rcv_msg+0x184/0x250
[   16.342704]  ? __pfx_nl80211_pre_doit+0x10/0x10 [cfg80211]
[   16.342800]  ? __pfx_nl80211_new_key+0x10/0x10 [cfg80211]
[   16.342907]  ? __pfx_nl80211_post_doit+0x10/0x10 [cfg80211]
[   16.343017]  genl_rcv_msg+0x4c/0xb0
[   16.343022]  ? __pfx_genl_rcv_msg+0x10/0x10
[   16.343026]  netlink_rcv_skb+0x5a/0x110
[   16.343036]  genl_rcv+0x28/0x50
[   16.343040]  netlink_unicast+0x24a/0x390
[   16.343046]  netlink_sendmsg+0x214/0x470
[   16.343054]  ____sys_sendmsg+0x38b/0x3f0
[   16.343062]  ___sys_sendmsg+0x9a/0xf0
[   16.343068]  ? __sys_setsockopt+0x79/0xe0
[   16.343075]  __sys_sendmsg+0x89/0xf0
[   16.343082]  __x64_sys_sendmsg+0x1d/0x30
[   16.343086]  x64_sys_call+0x91e/0x25a0
[   16.343091]  do_syscall_64+0x7f/0x180
[   16.343095]  ? syscall_exit_to_user_mode+0x86/0x260
[   16.343101]  ? do_syscall_64+0x8c/0x180
[   16.343105]  ? do_syscall_64+0x8c/0x180
[   16.343109]  ? __x64_sys_ioctl+0xa0/0xf0
[   16.343116]  ? syscall_exit_to_user_mode+0x86/0x260
[   16.343121]  ? do_syscall_64+0x8c/0x180
[   16.343125]  ? clear_bhb_loop+0x15/0x70
[   16.343132]  ? clear_bhb_loop+0x15/0x70
[   16.343137]  ? clear_bhb_loop+0x15/0x70
[   16.343144]  entry_SYSCALL_64_after_hwframe+0x78/0x80
[   16.343149] RIP: 0033:0x79796a72c004
[   16.343161] Code: 15 19 6e 0d 00 f7 d8 64 89 02 b8 ff ff ff ff eb bf 0f 1f 44 00 00 f3 0f 1e fa 80 3d 45 f0 0d 00 00 74 13 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 4c c3 0f 1f 00 55 48 89 e5 48 83 ec 20 89 55
[   16.343164] RSP: 002b:00007ffe973615c8 EFLAGS: 00000202 ORIG_RAX: 000000000000002e
[   16.343169] RAX: ffffffffffffffda RBX: 000062f159566850 RCX: 000079796a72c004
[   16.343172] RDX: 0000000000000000 RSI: 00007ffe97361600 RDI: 0000000000000006
[   16.343175] RBP: 00007ffe973615f0 R08: 0000000000000004 R09: 00000000000000f0
[   16.343178] R10: 00007ffe97361700 R11: 0000000000000202 R12: 000062f1595ffe00
[   16.343181] R13: 000062f159566b30 R14: 00007ffe97361600 R15: 0000000000000000
[   16.343187]  </TASK>
[   16.343196] ---[ end trace ]---
```

markruiz14, 2025-01-22:
@65usami 

Yes, I see the same error in `dmesg`, however the card limps along and is functional for me

`dmesg` output for me:
````
[Jan22 10:07] usb 1-11: new high-speed USB device number 14 using xhci_hcd
[  +0.126253] usb 1-11: New USB device found, idVendor=0bda, idProduct=8812, bcdDevice= 0.00
[  +0.000017] usb 1-11: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000005] usb 1-11: Product: 802.11n NIC
[  +0.000004] usb 1-11: Manufacturer: Realtek
[  +0.000004] usb 1-11: SerialNumber: 123456
[  +0.345907] usb 1-11: 88XXau 00:c0:ca:b0:34:72 hw_info[d7]
[  +0.001327] usbcore: registered new interface driver rtl88XXau
[  +0.008339] rtl88XXau 1-11:1.0 wlx00c0cab03472: renamed from wlan0
[  +8.067472] ------------[ cut here ]------------
[  +0.000006] UBSAN: array-index-out-of-bounds in /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_wlan_util.c:1905:48
[  +0.000030] index 1 is out of range for type 'u8 [1]'
[  +0.000002] CPU: 11 PID: 0 Comm: swapper/11 Tainted: G           OE      6.8.0-51-generic #52-Ubuntu
[  +0.000002] Hardware name: Razer Blade 15 Base Model (Early 2020) - RZ09-0328/DA550, BIOS 01.02 04/13/2020
[  +0.000001] Call Trace:
[  +0.000002]  <IRQ>
[  +0.000003]  dump_stack_lvl+0x76/0xa0
[  +0.000008]  dump_stack+0x10/0x20
[  +0.000001]  __ubsan_handle_out_of_bounds+0xc6/0x110
[  +0.000004]  HT_caps_handler+0x339/0x350 [88XXau]
[  +0.000078]  OnAssocRsp+0x222/0x290 [88XXau]
[  +0.000063]  mgt_dispatcher+0x17e/0x250 [88XXau]
[  +0.000038]  validate_recv_mgnt_frame+0x203/0x3f0 [88XXau]
[  +0.000051]  validate_recv_frame+0x146/0x1e0 [88XXau]
[  +0.000051]  recv_func+0x50/0x2e0 [88XXau]
[  +0.000051]  pre_recv_entry+0x59/0x140 [88XXau]
[  +0.000051]  recvbuf2recvframe+0x4b/0x1e0 [88XXau]
[  +0.000077]  usb_recv_tasklet+0x54/0x100 [88XXau]
[  +0.000059]  tasklet_action_common.isra.0+0xfa/0x1e0
[  +0.000003]  tasklet_action+0x22/0x30
[  +0.000002]  handle_softirqs+0xd8/0x340
[  +0.000002]  __irq_exit_rcu+0xd9/0x100
[  +0.000001]  irq_exit_rcu+0xe/0x20
[  +0.000002]  common_interrupt+0xa4/0xb0
[  +0.000002]  </IRQ>
[  +0.000001]  <TASK>
[  +0.000001]  asm_common_interrupt+0x27/0x40
[  +0.000002] RIP: 0010:cpuidle_enter_state+0xda/0x730
[  +0.000003] Code: 27 f6 fe e8 38 ee ff ff 8b 53 04 49 89 c7 0f 1f 44 00 00 31 ff e8 66 be f4 fe 80 7d d0 00 0f 85 61 02 00 00 fb 0f 1f 44 00 00 <45> 85 f6 0f 88 f7 01 00 00 4d 63 ee 49 83 fd 0a 0f 83 17 05 00 00
[  +0.000001] RSP: 0018:ffffadb200193e18 EFLAGS: 00000246
[  +0.000002] RAX: 0000000000000000 RBX: ffffcdb1ffd95100 RCX: 0000000000000000
[  +0.000001] RDX: 000000000000000b RSI: 0000000000000000 RDI: 0000000000000000
[  +0.000001] RBP: ffffadb200193e68 R08: 0000000000000000 R09: 0000000000000000
[  +0.000001] R10: 0000000000000000 R11: 0000000000000000 R12: ffffffffb64e5e80
[  +0.000001] R13: 0000000000000003 R14: 0000000000000003 R15: 00000096c4932145
[  +0.000001]  ? cpuidle_enter_state+0xca/0x730
[  +0.000002]  ? tick_nohz_stop_tick+0x110/0x210
[  +0.000003]  cpuidle_enter+0x2e/0x50
[  +0.000003]  call_cpuidle+0x23/0x60
[  +0.000002]  cpuidle_idle_call+0x11d/0x190
[  +0.000001]  do_idle+0x87/0xf0
[  +0.000001]  cpu_startup_entry+0x2a/0x30
[  +0.000001]  start_secondary+0x129/0x160
[  +0.000002]  secondary_startup_64_no_verify+0x184/0x18b
[  +0.000003]  </TASK>
[  +0.000006] ---[ end trace ]---
[  +0.000001] ------------[ cut here ]------------

... same error for Index 2 ...
```

65usami, 2025-01-23:
@markruiz14 

I tried resolving the error below in my own way, but it’s not perfect.
Could you fix it properly?
So far, it’s been working more stably than the 63cf0b4 commit.

https://github.com/tvision-jp/rtl8812au/tree/r1.0.0

chulas21, 2025-01-30:
> [@markruiz14](https://github.com/markruiz14)
>
> I tried resolving the error below in my own way, but it’s not perfect. Could you fix it properly? So far, it’s been working more stably than the [...]

Worked like a charm! Thanks u very much @65usami 🚀

65usami, 2025-01-31:
I tried resolving the error below in my own way for this [2607850607](https://github.com/aircrack-ng/rtl8812au/issues/1216#issuecomment-2607850607), but it’s not perfect.
Could someone fix it properly or do pull request to this aircrack-ng/rtl8812au?
So far, it’s been working more stably than the https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d commit.

https://github.com/tvision-jp/rtl8812au/tree/r1.0.0