aircrack-ng/rtl8812au #1217: Sharing:  TP-Link AC600 Nano on Raspbian Bookworm 32-Bit

lbussy, 2024-12-09:
I figured this out, and I wanted to share how to do it.

1.  Kernel headers are not (yet?) available on 32-bit v8.  To get past this:

  ``` bash
  echo "arm_64bit=0" | sudo tee -a /boot/firmware/config.txt > /dev/null
  ```
2. Reboot
3. Either:
  ``` bash
  sudo apt --fix-broken install -y
  sudo apt update
  sudo apt full-upgrade -y
  sudo apt autoremove --purge -y
  sudo apt clean
  ```
   or
  ``` bash
  curl -fsSL https://gist.githubusercontent.com/lbussy/23c05d8dc8c24d8d8edddf1d381f1c8b/raw/install_aliases.sh | bash
  ```

(this will pull down a .bash_aliases, run the above updates, and add some handy aliases.  [More in this Gist](https://gist.github.com/lbussy/23c05d8dc8c24d8d8edddf1d381f1c8b#bash-aliases).)

4. Reboot (if indicated)
5. Install required packages:
  ``` bash
  sudo apt-get install dkms raspberrypi-kernel-headers
  ```
6. Clone repo:
  ``` bash
  git clone -b v5.6.4.2 https://github.com/aircrack-ng/rtl8812au.git
  cd rtlrtl8812au
  ```
7. Change line in `dkms.conf`:
  ``` bash
  MAKE="ARCH=arm 'make' -j$PROCS_NUM KVER=${kernelver} KSRC=/lib/modules/${kernelver}/build"
  ```
    or
  ``` bash
  sed -i 's/^MAKE="/MAKE="ARCH=arm\ /' dkms.conf
  ```
8. Update `Makefile`:
  ``` bash
  CONFIG_PLATFORM_I386_PC = n
  CONFIG_PLATFORM_ANDROID_ARM64 = n
  CONFIG_PLATFORM_ARM_RPI = y
  CONFIG_PLATFORM_ARM64_RPI = n
  CONFIG_PLATFORM_ARM_NV_NANO = n
  ```
    or
  ``` bash
  sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
  sed -i 's/CONFIG_PLATFORM_ARM_RPI = n/CONFIG_PLATFORM_ARM_RPI = y/g' Makefile
  ```
9. Export environment:
  ``` bash
  export ARCH=arm
  ```
10. Make and install drivers:
  ``` bash
  sudo make dkms_install
  ```
11. Reboot
12. Bask in the glory of the `wlan0` being available with `ip link show`:
  ```
  $ ip link show
  1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
      link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
  2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN mode DEFAULT group default qlen 1000
      link/ether dc:a6:32:23:77:b4 brd ff:ff:ff:ff:ff:ff
  3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DORMANT group default qlen 1000
      link/ether dc:a6:32:23:77:b6 brd ff:ff:ff:ff:ff:ff
  4: wlan1: <NO-CARRIER,BROADCAST,MULTICAST,UP,LOWER_UP> mtu 2312 qdisc mq state DORMANT mode DORMANT group default qlen 1000
      link/ether b0:19:21:1c:79:ab brd ff:ff:ff:ff:ff:ff
  ```

g0730n, 2025-01-02:
Trying this right now... Following instructions on readme for Raspberry Pi 1 did not work correctly. After following those instructions was able to see the device, but unable to get it to work. We'll see if this works. I am on a fresh install of the latest Raspbian OS.

lbussy, 2025-01-02:
Did you get any errors?  Are you using 32 or 64-bit?  I don't have a Pi 1 here to try it on.

lbussy, 2025-01-02:
[This thread](https://askubuntu.com/questions/1194048/what-does-mean-dormant-mode-in-ip-link-show-wlan0) seems to indicate that the adapter is there and available and in power save mode.  Have you tried to bind a connection to it?

g0730n, 2025-01-02:
I am typing this in on my laptop which is setting next to the pi, so i may have some types but this is what it shows up as when I run "ip link show":
`3: wlan0: <NO-CARRIER, BROADCAST, MULTICAST, UP, LOWER_UP> mtu 2312 qdisc mq state DORMANT mode DORMANT group default qlen 1000`

It's been a while since I messed with network adapters in linux, how would I go about binding a connection to it? Thanks!

g0730n, 2025-01-02:
When I ran "make && make install" the first time, i saw some messages about pi64 kernel headers or something like that... I am wondering if the wrong headers got installed. This is a fresh install of rasbian OS so I may just start over.

g0730n, 2025-01-02:
When I boot the Pi up, after logging in, a screen and a half of error messages scroll up. I was poking around some more, and I think the driver I actually need is the "RTL8821au", uninstalled the other one and trying that one from here: [https://github.com/ivanovborislav/rtl8821au](https://github.com/ivanovborislav/rtl8821au)
Will let ya'll know if that was the issue.
8821 and 8812 is kinda confusing, at first glance I see it as the same.

lbussy, 2025-01-02:
I probably would have shot right past that as well.