aircrack-ng/rtl8812au #1218: Project: Add 8812au, 8821/11au and 8814au mac80211 in-kernel drivers to the Linux Mainline kernel. Status...

morrownr, 2024-12-20:
Greetings all,

This project has been underway for over one year. A tremendous amount of work has taken place. The new 8812au and 8821/11au drivers recently went into kernel 6.13. Follow up patches to add performance are scheduled to go into kernel 6.14.

If you simply want to test all 3 of these drivers, proceed to the following repo:

https://github.com/lwfinger/rtw88

The new driver for the 8814au was added today, 2024-12-20. It is ready for testing.

We need testers. Programmers are very welcome also. We especially need monitor mode testers.

See you at the repo.

@morrownr

dualfade, 2025-01-11:
@morrownr This works perfect; thanks !!

morrownr, 2025-01-11:
Glad to hear that. Enjoy.

morrownr, 2025-02-07:
FYI: The new 8814au driver has started the process of working its way into the kernel. It will happen as a series of 4 patch sets. The first set has gone in. It looks like it may be ready to go into kernel 6.15.

fariouche, 2025-02-10:
is injection working or not yet? I remember having seen an issue about it.
anyway, it's great news

morrownr, 2025-05-12:
@fariouche 

> is injection working or not yet? I remember having seen an issue about it.
anyway, it's great news

I have been seeing reports that injection is work on all 3 of the news drivers. Report any problems in the repo listed in the first message.

morrownr, 2025-05-12:
The new driver for the rtl8814au chip went into kernel 6.15. More patches that include optimizations are ready to go into kernel 6.16.

For those not yet using kernel 6.14/6.15, the repo in the first message will allow you to install the new standards compliant rtw8814au, rtl8812au and rtl8821/11au drivers on kernels as old as 5.4. We continue to need testing on the various monitor mode functions to include VLAN.